patman15/BMS_BLE-HA #287: Comunication stops to work

MatteoTSC, 2025-05-02:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I have thought about [providing a debug log](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) and/or [diagnostic data](https://www.home-assistant.io/docs/configuration/troubleshooting/#download-diagnostics).

### Describe the issue

Hello, i've started to use the integration one mont ago more or less and everything since now work fine. i'm using an ESP32 as BT bridge and a PowerQueen as battery. Since today i cant connect to battery anymore, debug information tells "assert ble_device.name is not None" "aassertion error" 
Can you help me?

Thank you very much,

Matteo

patman15, 2025-05-02:
That assertion should never trigger. Sounds like a strange BT issue. Can you please provide a debug log. It's hard to tell anything without it.
Best regards,
Patrick

MatteoTSC, 2025-05-02:
[logs.xt.txt](https://github.com/user-attachments/files/20017060/logs.xt.txt)

MatteoTSC, 2025-05-02:
meanwhile maybe i've found the cause, i'm not totally sure, but coud be a switchbot cloud, using his bt bridge the cause.

MatteoTSC, 2025-05-02:
Thank you Patrick!

patman15, 2025-05-02:
Still weird system behaviour that is reported. Looks like your battery does not have a name. No idea how this can happen. I need to dig into the source code of Home Assistant to see how this could be possible. This will take a bit of time.
Meanwhile:
- are you using a default HAOS installation?
- can you provide the information, the BT advertisement monitor shows?
- can you get debug logs from the BT proxy using `VERY_VERBOSE` level?

Thanks,
Patrick

Cellgalvano, 2025-05-02:
Hi!

Same problem for me. I have multiple PowerQueen Batteries and I'm also using the Bluetooth Proxy. With 1.15.0 I can connect and configure the batteries without a problem. When updating to 1.16.0 already configured connections seem to remain working, but new connections are not possible. A downgrade to 1.15.0 fixes the issue for me.

MatteoTSC, 2025-05-03:
Hi Patrick,
HA installation is very old, i’ve donwloaded the ova from official site IFI i remember well. 

Here logs from esp32 bt proxy (if is not enought, can you let me know how i can enable the very_verbose level logging?)

[logs_esp32-bluetooth-proxy-793ba4_logs.txt](https://github.com/user-attachments/files/20019531/logs_esp32-bluetooth-proxy-793ba4_logs.txt)

MatteoTSC, 2025-05-03:
> Hi!
>
> Same problem for me. I have multiple PowerQueen Batteries and I'm also using the Bluetooth Proxy. With 1.15.0 I can connect and configure the batteries without a problem. When updating to [...]

Thank you very much, i’ll try to do a downgrade.

patman15, 2025-05-03:
Hi @Cellgalvano 

> A downgrade to 1.15.0 fixes the issue for me.

Can you also provide me a debug log? Is it really the same error message?
I checked the code, but couldn't spot any relevant changes from 1.15 to 1.16 that would explain the problem. I'll have a deeper look next week.

patman15, 2025-05-05:
@Cellgalvano, @MatteoTSC can you please try [this branch](https://github.com/patman15/BMS_BLE-HA/tree/279-integration-fails-with-initialising-ultimatron-battery) via [manual install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#installation). Please let me know if it helps and can you send me a debug log?

Cellgalvano, 2025-05-05:
@patman15 I'm using two ESP32-C3 BT-Proxys. One of them is also running a Victron integration for my solar charger.
With that setup it auto detects the batteries with V1.15.0 and the setup works. After upgrading to V1.16.0 already setup connections still work but new connections are not shown anymore. So a reconfig of the same battery is not possible for me.

patman15, 2025-05-05:
Ok totally strange.
- Can you please send me the information the advertisement monitor shows when you use v1.16.0? Is it even there?
- I guess you have rebooted between the version changes and waited at least 90s?
- I could see in the short log that the other two batteries work without issue.

I will do another code comparison to see if I can spot something.

patman15, 2025-05-05:
> @patman15 What do you mean by advertisement monitor? The discovered devices page?

Yes, [bluetooth integration](https://my.home-assistant.io/redirect/integration/?domain=bluetooth). On your BT adapter select `configure -> advertisement monitor`.
If it is not there it is some sort of BT issue. If the device does not show up there, the integration will not find it. Strange thing is that you say it depends on the integration version, which does not make any sense to me as there is none of my code involved with regards to the BT detection.

Cellgalvano, 2025-05-05:
@patman15  I just checked the advertisement monitor (thank you for pointing this feature out) with both versions. I shows all devices regardless of the version. Only difference is that with V1.16.0 the devices are not shown on the discovered devices page and also when clicking on the add button within the integration it shows that no devices were found. Strange indeed.

Cellgalvano, 2025-05-05:
@patman15 with V1.16.0 the not connected battery is still shown in the advertisement monitor but not available via the integration.

This is the data from the advertisement monitor: 
{"name":"P-24050BNNA70-A01133","address":"C8:47:80:18:ED:BC","rssi":-68,"manufacturer_data":{"22618":"c8478018edbc"},"service_data":{},"service_uuids":["0000ffe0-0000-1000-8000-00805f9b34fb"],"source":"A0:76:4E:31:01:12","connectable":true,"time":1746442612.348774,"tx_power":null}

patman15, 2025-05-05:
@Cellgalvano I can see in some logs of you:
```
2025-05-05 11:05:20.359 DEBUG (MainThread) [bleak_retry_connector] C8:47:80:15:22:83 - C8:47:80:15:22:83: Failed to connect: No backend with an available connection slot that can reach address C8:47:80:15:22:83 was found, device_missing: False, backing off: 4.0 (attempt: 1, last rssi: None)
```
meaning that it looks like the integration is fighting over the 3 available connections slots. Did you check the available connection slots for your BT adapter? It looks like they are sometimes used up. A BT proxy can only do 3 devices, see the [big fat warning](https://esphome.io/components/bluetooth_proxy.html)!

patman15, 2025-05-05:
@MatteoTSC 
> Hi Patrick, HA installation is very old
> 
But it needs to be >2024.02, right?

> Here logs from esp32 bt proxy (if is not enought, can you let me know how i can enable the very_verbose level logging?)
> 
See the [logger](https://esphome.io/components/logger.html), please modify your configuration to have `level: VERY_VERBOSE`.

Cellgalvano, 2025-05-05:
@patman15 I know of the limitations per adapter. In my normal setup all three slots are used up by three batteries using your integration. I can add another adapter to check if that helps but still for me I do not understand why this is working flawlessly with V1.15.0 and not with V1.16.0. I switched versions multiple times now. The behavior is reproducable on my side. Any other hints from you side on where to look? Thanks!

MatteoTSC, 2025-05-05:
> [@MatteoTSC](https://github.com/MatteoTSC)
>
>
> But it needs to be >2024.02, right?
>
>
> See the [logger](https://esphome.io/components/logger.html), please modify your configuration to have `level: [...]

Hi, yes is the last version, it's an old installation but i'm keeping it updated.

MatteoTSC, 2025-05-05:
Now i'm having the issue, but now in log file i can see a very different error:

2025-05-05 13:38:18.664 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JP0KKQQ5BRNNA5JBMTC8Z1ER version=1 domain=bms_ble title=C8:47:80:19:D2:16 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=C8:47:80:19:D2:16>
2025-05-05 13:38:18.664 DEBUG (MainThread) [custom_components.bms_ble] Failed to discover device C8:47:80:19:D2:16 via Bluetooth

probably i have to try to use another esp32 as BTproxy. is there a "best practice" configuration for BT proxy using an ESP32?

MatteoTSC, 2025-05-05:
Nothing seems it doesn't wark anymore, i've try even downgrade to 1.5 wothout success. now i'm still receiving error "Failed setup, will retry: Could not find BMS (C8:47:80:19:D2:16) via Bluetooth."

MatteoTSC, 2025-05-05:
> Hi, yes is the last version, it's an old installation but i'm keeping it updated.

Here logs with verbose enabled, probably esp restart have solve my temporary issue, in fact now i'm connected to battery. I think was non related to the origina issue.... 

[logs_esp32-bluetooth-proxy-793ba4_logs.txt](https://github.com/user-attachments/files/20037760/logs_esp32-bluetooth-proxy-793ba4_logs.txt)

patman15, 2025-05-05:
> Here logs with verbose enabled, probably esp restart have solve my temporary issue, in fact now i'm connected to battery. However, I believe this was unrelated to the original problem.

@MatteoTSC Sorry, I'm a bit lost about your status. The log you posted does not show anything related to the battery, because you are connected. So did the problem disappear for you?

patman15, 2025-05-05:
@Cellgalvano 
> I do not understand why this is working flawlessly with V1.15.0 and not with V1.16.0. I switched versions multiple times now. The behavior is reproducable on my side.

I only know that if you use all slots the BT system gets very sensitive to disturbances, either on the air or timing wise. What can happen is that depending on the version of my integration the timing is a bit different and other BT integrations get priority. Interestingly, this is very reproducible and not random at all with a certain code base but might change with any other version of any BT code again.  

> Any other hints from you side on where to look? Thanks!

Try adding another BT proxy or use a [high-performance USB BT adapter](https://www.home-assistant.io/integrations/bluetooth/#known-working-high-performance-adapters). I would almost bet that this solves the issue.

Cellgalvano, 2025-05-05:
@patman15 Is there a possibility to choose which adapters slots are used by the integration? I assume that connections need to be MAC based so that batteries can not "move" between adapters, right? So if I want to share the load between two proxies it would be great to be able to select the interface to use.

MatteoTSC, 2025-05-05:
> [@MatteoTSC](https://github.com/MatteoTSC) Sorry, I'm a bit lost about your status. The log you posted does not show anything related to the battery, because you are connected. So did the problem [...]

Yes, i have had a different problem, in that case related to the bt proxy. Now seems working, i’ll monitoring if there will be new disconnection and i’ll let you know. Thank you for your support

Cellgalvano, 2025-05-08:
@patman15 I just installed V1.16.1 and tested the connections / setup with one and with two BL proxies. It now works fine in my environment just as with V1.15.0. And I do not have any problems even when using all three connections on one proxy.

patman15, 2025-05-08:
@Cellgalvano ok, thanks for the feedback so there was definitely some bug in the 1.16. I guess #294 helped.

MatteoTSC, 2025-05-08:
Sorry but unfortunately with new version i’m having trouble… i cannot start at all please can you check my logo?

[home-assistant_bms_ble_2025-05-08T20-16-02.201Z.log](https://github.com/user-attachments/files/20110148/home-assistant_bms_ble_2025-05-08T20-16-02.201Z.log)

MatteoTSC, 2025-05-08:
Now after every reboot i receive this error:

[image]

patman15, 2025-05-09:
> Sorry but unfortunately with new version i’m having trouble… i cannot start at all please can you check my logo?
> 
Do you have 2 BT proxies? Could they be disturbing each other (too close together)?
Log shows that you again have the problem that you had when you opened the issue. The device does not report a name. Also, BT is not able to connect.

MatteoTSC, 2025-05-09:
Hi,
Thank you for log checking.
I have only one esp32 bt proxy, and one hub from SwitchBot but the distance is at least 4-5 meters.
When I restart HA every Times happens, i have to reboot bt proxy, force reload of BLE battery more than one time and finally the battery appears.

patman15, 2025-05-09:
Some thoughts from my side, maybe this gives you new ideas on where to look:
- I had a look at the log you posted in https://github.com/patman15/BMS_BLE-HA/issues/287#issuecomment-2848316810 which does not show your device being seen by the proxy at all. That's strange. How did you setup the proxy? Can you post your config and reset it to a minimal one? (I see you have `interval` and `window` set which is known to cause WiFi issues, which could result in lost advertisements, which would explain why your proxy does not relay any message for 20s)
- The fact that you need to force reloading the integration is strange. There is nothing that a forced reload does different to the usual 30s interval updates, except that it closes the BT connection. But in your case the connection doesn't even get established.
- Does it only happen on reboot? How long did you wait till you considered it "non-working"? I would say everything till 90s can be still normal behavior, although it usually goes much faster.
- Did you ever delete the battery from the integration and re-add it? I'm still wondering how you can end up with a device without a name.

Can you post another proxy log in `VERY_VERBOSE` when it's not working? I would like to see if the device MAC even appears at the proxy. If it doesn't it's not an integration issue in the first place.

MatteoTSC, 2025-05-09:
Hi 
thank you very much for your support.
Now i’m abroad and I’ll come back to home next week, so to try a restart i prefer to wait when I’m back.
Meanwhile i can paste the esp32 conf

[image]
[image]

The problem appears for sure after reboot, in the past, with previus app version sometimes i seen battery disconnected without a restart, since i’e installed 1.16 version did’t happened.

No, i’ve never delete the battery.

patman15, 2025-05-09:
Please remove
```YAML
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true
```
and try again (see [this note](https://esphome.io/components/bluetooth_proxy.html#improving-reception-performance)). Probably, to be on the safe side, please also remove
```YAML
web_server:
  port: 80
```

MatteoTSC, 2025-05-10:
> Please remove
> 
>   scan_parameters:
>     interval: 1100ms
>     window: 1100ms
>     active: true
> and try again (see [this note](https://esphome.io/components/bluetooth_proxy.html#improving-reception-performance)). Probably, to be on the safe side, please also remove
> 
> web_server:
>   port: 80

Hi!
I have removed the lines and i did a restart, seems works, I’ll try better in next days.

Thank you very much!

MatteoTSC, 2025-05-15:
Hi, as update,
a first step i've removed parameter you told me, but sometimes i still had disconnection. So 2 days ago i've changed the esp32 board and i have prepared it as standard bt proxy from esphome.
even in that case i still having disconnection, as you can see in this picture:
[image]

probably i have some kind of interference on BT frequency

patman15, 2025-05-15:
Can you enable `RSSI` and `link quality ` sensor and post a >=24h chart? Are you able to provide a debug log of the time including and a bit surrounding the outages?
Since it mostly works I only see a low chance that it is related to the integration. I suspect BT interference as you say.

MatteoTSC, 2025-05-16:
Hi, 
i can confirm that i'm having trouble with bt proxy (even the new one), when the battery disconnection appears I can see:
[image]

And trying to connect to bt proxy:
[image]

so, thank you very much for your support, you can close the bug.

Have a nice day,

Matteo

closed 2025-05-16 by patman15