patman15/BMS_BLE-HA #288: ACCURAT Battery loosing connection

Guenni9, 2025-05-03:
### Checklist

- [x] I have tried the latest version of BMS_BLE-HA.
- [x] I have enabled debug logging for my installation.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [x] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

Jiabaida

### Describe the issue

Dear @patman15,
I have exactly this battery: [#88, ACCURAT 200Ah](https://github.com/patman15/BMS_BLE-HA/issues/88)
I found out, if battery discharging is over, the communication stops. If switching to discharge manually, the entities where shown fine, again.
I'm using a [Voltronic compatible charger](https://my.easunpower.com/products/easun-power-3-5kw-24v-mppt-solar-inverter-with-wifi-pv-power-5000w?variant=46962629312823) to charge with sun and discharge at night.
While entities are unavailable, the app was in spite of it able to read the BMS out.
Thank you!

### Reproduction steps

```text
1.Wait until battery discharging is over
2.Entities became unavailable
3.Discharging the battery manually brings all entities back
...
```

### Debug logs

[image]

patman15, 2025-05-03:
Hi! Can you please provide a debug log? It could be related to #273.

patman15, 2025-05-05:
@Guenni9 can you please try [this branch](https://github.com/patman15/BMS_BLE-HA/tree/279-integration-fails-with-initialising-ultimatron-battery) via [manual install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#installation). Please let me know if it helps and can you send me a debug log?

Guenni9, 2025-05-05:
Dear @patman15 , thank you!

the branch above doesn't connect no more. I will try to connect with the app, when I'm at home. Yesterday it was the same, as soon as the android app was started, the BMS connects...
I attach a log file from this morning, the most time it was connected and looses sometimes the connection...

[home-assistant_bms_ble_2025-05-05T09-45-32.892Z.log](https://github.com/user-attachments/files/20036548/home-assistant_bms_ble_2025-05-05T09-45-32.892Z.log)

Guenni9, 2025-05-05:
Ah yes, and with the above mentioned branch the attached file was logged:

[home-assistant_bms_ble_2025-05-05T12-45-33.317Z.log](https://github.com/user-attachments/files/20037120/home-assistant_bms_ble_2025-05-05T12-45-33.317Z.log)

Guenni9, 2025-05-05:
Thank you, @patman15, I've just ordered a BLE Proxy: [ESP32 POE](https://docs.m5stack.com/en/unit/poesp32), because I can't get them closer..

Guenni9, 2025-05-11:
Dear @patman15 ,

let me confirm, with 1.16.1 all issues are gone! Even connected to the RPI4 directly at -73dBm!

thank you!!

closed 2025-05-11 by Guenni9

patman15, 2025-05-11:
@Guenni9 ok, thanks a lot for that feedback, got my lesson in low-level BT issues. Sorry for messing up 1.16.0!