patman15/BMS_BLE-HA #290: Seplos BMS V3 - Data from 2nd Pack

lmuser22, 2025-05-05:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Getting the Data from BMS works fine, but would be great to get Data from Pack01 and Pack02 separately, like in Spelos BMS Studio

[image]

patman15, 2025-05-05:
Hi! This data should be available as attributes to the respective sensors. Please see the README and let me know if it works for you.

lmuser22, 2025-05-13:
> I noticed today when I disconnect the 2nd rs485 cable pack 2, I get 2 seperate bms in ble.

but then my Inverter System does not get Info about 2nd BMS via CAN

broekieman, 2025-05-13:
> > I noticed today when I disconnect the 2nd rs485 cable pack 2, I get 2 seperate bms in ble.
> 
> but then my Inverter System does not get Info about 2nd BMS via CAN

Exactly, would be nice to have both available.

lmuser22, 2025-05-14:
> > > I noticed today when I disconnect the 2nd rs485 cable pack 2, I get 2 seperate bms in ble.
> > 
> > 
> > but then my Inverter System does not get Info about 2nd BMS via CAN
> 
> Exactly, would be nice to have both available.

Data is available in the attributes of Sensors. Now i have both BMS in HA :) 

[image]

[image]

broekieman, 2025-05-14:
Please tell how to extract the cells from attribute. Don't know how.

patman15, 2025-05-14:
Please see the [FAQ](https://github.com/patman15/BMS_BLE-HA/tree/309-some-jbd-bms-start-with-a-jbd--string?tab=readme-ov-file#how-do-i-get-the-cell-voltages-as-individual-sensor-for-tracking), there is the answer to your question.

lmuser22, 2025-05-14:
> Please tell how to extract the cells from attribute. Don't know how.

My HA is in German Setting but Code for my "Helfer" is 

Cell 00
{{ state_attr('sensor.sp05b2408270330_differenzspannung', 'cell_voltages')[0]}}

Cell01
{{ state_attr('sensor.sp05b2408270330_differenzspannung', 'cell_voltages')[1]}}

....

closed 2025-06-27 by github-actions[bot]