patman15/BMS_BLE-HA #291: Neey Balancer 4th Gen

lmuser22, 2025-05-05:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Neey Balancer has a BT Interface,

patman15, 2025-05-06:
Do you have any of the information that is requested by the template available to support the integration?

patman15, 2025-05-07:
Ok, looks like there is [code](https://github.com/syssi/esphome-jk-bms/blob/main/components/heltec_balancer_ble/heltec_balancer_ble.cpp) available. I can have a look after the next release, if there is still enough interest.

SOLAR-RAIDER, 2025-05-07:
Great thank you !

patman15, 2025-05-20:
Can someone please get me the advertisement from the [advertisement monitor](https://my.home-assistant.io/redirect/bluetooth_advertisement_monitor/) via copy to clipboard?

lmuser22, 2025-05-20:
{"name":"Backup BMS","address":"38:36:11:01:95:BF","rssi":-88,"manufacturer_data":{"34952":"8888bf9501113638"},"service_data":{},"service_uuids":["0000ffe0-0000-1000-8000-00805f9b34fb","0000fee7-0000-1000-8000-00805f9b34fb"],"source":"E4:5F:01:25:20:7A","connectable":true,"time":1747769341.4591684,"tx_power":null}

patman15, 2025-05-21:
I now made an initial version that you can [manually install](https://github.com/patman15/BMS_BLE-HA#installation) from the branch linked to this ticket. I started from the JK version, but I'm not sure about the protocol specifics. Please provide me a debug log.
I might also need a capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery. If that is simple for you, would be appreciated just to check that I do not miss a lot of special behavior ...
Expected outcome: Detection should work and you should see a number of sensors with values filled, constantly updated.

SOLAR-RAIDER, 2025-06-08:
`2025-06-08 12:50:16.896 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration porscheconnect which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-06-08 12:50:16.897 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-06-08 12:50:16.899 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solis_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-06-08 12:50:16.900 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration alexa_media which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-06-08 12:50:16.901 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration bms_ble which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-06-08 12:51:25.469 WARNING (MainThread) [ring_doorbell.doorbot] Your Ring account does not have an active subscription.
2025-06-08 12:52:56.489 DEBUG (MainThread) [custom_components.bms_ble.neey_bms::SR-LV-01:1F08)] RX BLE data (start): bytearray(b'U\xaa\x11\x01\x02\x00,\x01;gB\\@\xc2I\\@PG\\@\n;\\@\xfd/\\@\x968\\@\xfd/\\@Z\x1c\\@\xf2$\\@gB\\@\xe74\\@\x8aH\\@\xce9\\@gB\\@\x16F\\@gB\\@\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04\xc0\x06>K\xef\r>X\xcf\x15>B\x85\x1e>\xd6\xed#>K\xcb(>\x9b\xb3/>\xda\x910>X\x861>\xda\xf1)>_\x87(>\xbb')
2025-06-08 12:52:56.519 DEBUG (MainThread) [custom_components.bms_ble.neey_bms::SR-LV-01:1F08)] RX BLE data (cnt.): bytearray(b"\x03#>6r\x1c>\xdb\x96\x13>\'n\x07>\xbc\xa0\x04>\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xcd:\\B\xcd:\\@\x00\xa05;\x01\x07\x0f\x06\x00\x00\x00\x00q=\xd8Aq=\xd8A\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x12\r\x1d\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x8b\xff")
2025-06-08 12:52:56.939 DEBUG (MainThread) [custom_components.bms_ble.neey_bms::SR-LV-01:1F08)] RX BLE data (start): bytearray(b'U\xaa\x11\x01\x02\x00,\x01<\xdbD\\@\xfeJ\\@\x8aH\\@\n;\\@\xfd/\\@\xce9\\@r\xfc[@\t\x05\\@\xf2$\\@\xa0C\\@s2\\@\x16F\\@\x968\\@gB\\@\xdbD\\@-A\\@\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04\xc0\x06>K\xef\r>X\xcf\x15>B\x85\x1e>\xd6\xed#>K\xcb(>\x9b\xb3/>\xda\x910>X\x861>\xda\xf1)>_\x87(>\xbb')
2025-06-08 12:52:56.969 DEBUG (MainThread) [custom_components.bms_ble.neey_bms::SR-LV-01:1F08)] RX BLE data (cnt.): bytearray(b"\x03#>6r\x1c>\xdb\x96\x13>\'n\x07>\xbc\xa0\x04>\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0c6\\B\x0c6\\@\x00\x18\x9d;\x01\x06\x0f\x06\x00\x00\x00\x00q=\xd8Aq=\xd8A\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x12\r\x1d\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00~\xff")
2025-06-08 12:52:57.630 DEBUG (MainThread) [custom_components.bms_ble.neey_bms::SR-LV-01:1F08)] RX BLE data (start): bytearray(b'U\xaa\x11\x01\x02\x00,\x01=\xa0C\\@\xc2I\\@\x8aH\\@\n;\\@\x89-\\@C<\\@\xc3.\\@Z\x1c\\@,&\\@gB\\@\xe74\\@\x8aH\\@\n;\\@\xf1?\\@PG\\@\xf1?\\@\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04\xc0\x06>K\xef\r>X\xcf\x15>B\x85\x1e>\xd6\xed#>K\xcb(>\x9b\xb3/>\xda\x910>X\x861>\xda\xf1)>_\x87(>\xbb')
2025-06-08 12:52:57.660 DEBUG (MainThread) [custom_components.bms_ble.neey_bms::SR-LV-01:1F08)] RX BLE data (cnt.): bytearray(b"\x03#>6r\x1c>\xdb\x96\x13>\'n\x07>\xbc\xa0\x04>\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xe1:\\B\xe1:\\@\x00\xa05;\x01\x07\x0f\x06\x00\x00\x00\x00q=\xd8Aq=\xd8A\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x13\r\x1d\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xee\xff")
2025-06-08 12:52:57.930 DEBUG (MainThread) [custom_components.bms_ble.neey_bms::SR-LV-01:1F08)] RX BLE data (start): bytearray(b'U\xaa\x11\x01\x02\x00,\x01>\xa0C\\@\xfeJ\\@PG\\@\n;\\@\x89-\\@\n;\\@\xc3.\\@Z\x1c\\@\x7f"\\@\xdbD\\@\xab3\\@\x8aH\\@\xce9\\@gB\\@\x8aH\\@\xa0C\\@\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04\xc0\x06>K\xef\r>X\xcf\x15>B\x85\x1e>\xd6\xed#>K\xcb(>\x9b\xb3/>\xda\x910>X\x861>\xda\xf1)>_\x87(>\xbb')
[... cut ...]
2025-06-08 12:54:48.841 DEBUG (MainThread) [custom_components.bms_ble.neey_bms::SR-LV-01:1F08)] RX BLE data (cnt.): bytearray(b"\x03#>6r\x1c>\xdb\x96\x13>\'n\x07>\xbc\xa0\x04>\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00@&\\B@&\\@\x00\x8c\x9f;\x02\x07\x0f\x06\x00\x00\x00\x00q=\xd8Aq=\xd8A\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\r\x1d\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xbd\xff")
`

patman15, 2025-06-09:
> That's working.

Great! Thanks for the feedback?

> Is it possible to change parameters?

No, the integration does not support setting parameters.

> Or adding a second Neey?

Yes, any Neey in range should be detected automatically. If not, please post the info from the advertisement monitor.

SOLAR-RAIDER, 2025-06-10:
[image]
[image]
[image]

patman15, 2025-06-29:
@SOLAR-RAIDER did it work? Do you have feedback for me? Thanks!

johanolfert, 2025-08-17:
Hi, can you please add this Model: EK-16S2EB Balancer? I have three batteries with a Neey balancer that work fine, but this one doesn't work. 

[image]