patman15/BMS_BLE-HA #301: Add PowerQueen PQ-24* detection pattern

patman15, 2025-05-09:
> @patman15 Can you add also add "PQ-24*" to the list?
> "for pattern in ("R-12*", "R-24*", "P-12*", "P-24*", "PQ-24*")" works perfectly for my second PowerQueen 24v 100ah battery.  

 _Originally posted by @MatthiasTe in [#280](https://github.com/patman15/BMS_BLE-HA/issues/280#issuecomment-2864382409)_

patman15, 2025-05-09:
@MatthiasTe could you please provide me the info from the [advertisement monitor](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#if-your-device-is-not-recognized), so I can add a regression test?

MatthiasTe, 2025-05-09:
@patman15 Please find the data from advertisment monitor below:

{"name":"PQ-24100B-A00292","address":"C8:47:80:10:B7:9A","rssi":-65,"manufacturer_data":{"22618":"c8478010b79a"},"service_data":{},"service_uuids":["0000ffe0-0000-1000-8000-00805f9b34fb"],"source":"14:2B:2F:C4:F0:7A","connectable":true,"time":1746824435.9037337,"tx_power":null}

patman15, 2025-05-11:
@MatthiasTe I added it with PR #305. Could you test it?