patman15/BMS_BLE-HA #302: Add support for Ogrphy (bat-bms)

csuich2, 2025-05-09:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

I'm trying to connect my [ogrphy 48v](https://ogrphybat.com/products/ogrphy-48v100ah-lithium-golf-cart-battery-with-200a-bms-and-mobile-app-monitor) battery and am not sure what kind of BMS it has. Based on https://github.com/patman15/BMS_BLE-HA/issues/94 I tried renaming my battery to `BP00` and the device is detected but fails to connect (see screenshot below).

I'm able to connect to the battery via the [BAT-BMS](https://play.google.com/store/apps/details?id=com.bms.grenergy) Android app.

[image]

### Additional context

Device info from advertisement logs:
```json
{
  "name": "OGR-48100-123456789",
  "address": "<mac>",
  "rssi": -47,
  "manufacturer_data": {
    "13880": "13014e9a"
  },
  "service_data": {},
  "service_uuids": [
    "0000ff00-0000-1000-8000-00805f9b34fb",
    "00000001-0000-1000-8000-00805f9b34fb"
  ],
  "source": "<looks-like-a-mac>",
  "connectable": true,
  "time": 1746823674.5856438,
  "tx_power": null
}
```

ESPHome bluetooth proxy logs:
```console
[17:01:44][I][bluetooth_proxy:282]: [0] [MAC_ADDRESS] Connecting v3 with cache
[17:01:44][D][esp32_ble_tracker:115]: connecting: 0, discovered: 1, searching: 0, disconnecting: 0
[17:01:44][D][esp32_ble_tracker:231]: Pausing scan to make connection...
[17:01:44][I][esp32_ble_client:110]: [0] [MAC_ADDRESS] 0x00 Attempting BLE connection
[17:01:44][D][esp32_ble_tracker:115]: connecting: 1, discovered: 0, searching: 0, disconnecting: 0
[17:01:44][D][esp32_ble_client:191]: [0] [MAC_ADDRESS] ESP_GATTC_CONNECT_EVT
[17:01:44][D][esp32_ble_client:191]: [0] [MAC_ADDRESS] ESP_GATTC_OPEN_EVT
[17:01:44][I][esp32_ble_client:263]: [0] [MAC_ADDRESS] Connected
[17:01:44][D][esp32_ble_tracker:115]: connecting: 0, discovered: 0, searching: 0, disconnecting: 0
[17:01:44][D][esp32_ble_tracker:282]: Starting scan...
[17:01:44][D][esp32_ble_client:296]: [0] [MAC_ADDRESS] cfg_mtu status 0, mtu 517
[17:01:44][D][esp32_ble_client:191]: [0] [MAC_ADDRESS] ESP_GATTC_REG_FOR_NOTIFY_EVT
[17:01:45][D][esp32_ble_client:191]: [0] [MAC_ADDRESS] ESP_GATTC_WRITE_DESCR_EVT
[17:01:45][D][esp32_ble_client:191]: [0] [MAC_ADDRESS] ESP_GATTC_WRITE_CHAR_EVT
[17:01:49][I][esp32_ble_client:147]: [0] [MAC_ADDRESS] Disconnecting (conn_id: 0).
[17:01:49][D][esp32_ble_tracker:115]: connecting: 0, discovered: 0, searching: 0, disconnecting: 1
[17:01:49][D][esp-idf:000][BTU_TASK]: W (610841) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13

[17:01:49][D][esp-idf:000][BTU_TASK]: W (610847) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16

[17:01:49][D][esp-idf:000][BTU_TASK]: W (610849) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x16

[17:01:49][D][esp-idf:000][BTU_TASK]: W (610861) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x16

[17:01:50][D][esp32_ble_client:191]: [0] [MAC_ADDRESS] ESP_GATTC_CLOSE_EVT
[17:01:50][D][esp32_ble_tracker:115]: connecting: 0, discovered: 0, searching: 0, disconnecting: 0
[17:01:50][D][esp-idf:000][BTU_TASK]: W (610898) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16

[17:01:56][I][bluetooth_proxy:282]: [0] [MAC_ADDRESS] Connecting v3 with cache
[17:01:56][D][esp32_ble_tracker:115]: connecting: 0, discovered: 1, searching: 0, disconnecting: 0
[17:01:56][D][esp32_ble_tracker:231]: Pausing scan to make connection...
[17:01:56][I][esp32_ble_client:110]: [0] [MAC_ADDRESS] 0x00 Attempting BLE connection
[17:01:56][D][esp32_ble_tracker:115]: connecting: 1, discovered: 0, searching: 0, disconnecting: 0
[17:01:59][D][esp32_ble_client:191]: [0] [MAC_ADDRESS] ESP_GATTC_CONNECT_EVT
[17:01:59][D][esp32_ble_client:191]: [0] [MAC_ADDRESS] ESP_GATTC_OPEN_EVT
[17:01:59][I][esp32_ble_client:263]: [0] [MAC_ADDRESS] Connected
[17:01:59][D][esp32_ble_tracker:115]: connecting: 0, discovered: 0, searching: 0, disconnecting: 0
[17:01:59][D][esp32_ble_tracker:282]: Starting scan...
[17:01:59][D][esp32_ble_client:296]: [0] [MAC_ADDRESS] cfg_mtu status 0, mtu 517
[17:01:59][D][esp32_ble_client:191]: [0] [MAC_ADDRESS] ESP_GATTC_REG_FOR_NOTIFY_EVT
[17:01:59][D][esp32_ble_client:191]: [0] [MAC_ADDRESS] ESP_GATTC_WRITE_DESCR_EVT
[17:01:59][D][esp32_ble_client:191]: [0] [MAC_ADDRESS] ESP_GATTC_WRITE_CHAR_EVT
[17:02:04][I][esp32_ble_client:147]: [0] [MAC_ADDRESS] Disconnecting (conn_id: 0).
[17:02:04][D][esp32_ble_tracker:115]: connecting: 0, discovered: 0, searching: 0, disconnecting: 1
[17:02:04][D][esp-idf:000][BTU_TASK]: W (625707) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13

[17:02:04][D][esp-idf:000][BTU_TASK]: W (625712) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16

[17:02:04][D][esp-idf:000][BTU_TASK]: W (625715) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x16

[17:02:04][D][esp-idf:000][BTU_TASK]: W (625727) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x16

[17:02:04][D][esp-idf:000][BTU_TASK]: W (625740) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16

[17:02:04][D][esp32_ble_client:191]: [0] [MAC_ADDRESS] ESP_GATTC_CLOSE_EVT
[17:02:04][D][esp32_ble_tracker:115]: connecting: 0, discovered: 0, searching: 0, disconnecting: 0
```

patman15, 2025-05-10:
> I think I was able to capture the Bluetooth HCI logs but would prefer to share them privately.

I think I need them to identify the correct BMS, unless you are sure it is a Seplos v2/v3? You can reach my via PM using Discord (HA server) or HA community. Links are in my profile.

csuich2, 2025-05-11:
Oh a whim I decided to rename the battery to `GJ-0` since it seems JBD seems like a common BMS and sure enough it worked!

patman15, 2025-05-11:
@csuich2 ok cool, thanks for the info, I can add the `OGR-*` pattern to the JBD then, so you do not have to do the modification with each release.

closed 2025-05-11 by patman15