patman15/BMS_BLE-HA #303: Braun Power recognized as JBD, communication time out

prodisz, 2025-05-09:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Please help with unknown BMS. Battery renamed by reseller as QQ7.

[https://www.alibaba.com/product-detail/Urankine-Batterie-Solaire-Lithium-200ah-Batteries_1601372093370.html](url)

[image]

Android App: [https://www.pgyer.com/xMGD](url)

Integration log:

```
2025-05-06 18:47:20.147 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JTK7VMQSTQ7555CPPV8V7QYZ version=1 domain=bms_ble title=QQ7 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=24:19:72:D9:F8:E7>
2025-05-06 18:47:20.147 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] initializing Jiabaida Smart BMS, BT address: 24:19:72:D9:F8:E7
2025-05-06 18:47:20.147 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for QQ7 (24:19:72:D9:F8:E7) as Jiabaida Smart BMS
2025-05-06 18:47:20.148 DEBUG (MainThread) [custom_components.bms_ble] QQ7: advertisement: {'name': 'QQ7', 'address': '24:19:72:D9:F8:E7', 'rssi': -70, 'manufacturer_data': {123: b'\x02\xff\xff}'}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': 'EC:94:CB:63:88:0A', 'advertisement': AdvertisementData(local_name='QQ7', manufacturer_data={123: b'\x02\xff\xff}'}, service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-70), 'device': BLEDevice(24:19:72:D9:F8:E7, QQ7), 'connectable': True, 'time': 611705.650169837, 'tx_power': None}
2025-05-06 18:47:20.148 DEBUG (MainThread) [custom_components.bms_ble] QQ7: BMS data update
2025-05-06 18:47:20.148 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] connecting BMS
2025-05-06 18:47:20.148 DEBUG (MainThread) [bleak_retry_connector] 24:19:72:D9:F8:E7 - 24:19:72:D9:F8:E7: Connection attempt: 1
2025-05-06 18:47:21.375 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] TX BLE data: dd a5 03 00 ff fd 77
2025-05-06 18:47:26.377 DEBUG (MainThread) [custom_components.bms_ble] QQ7: device communication timed out
2025-05-06 18:47:26.377 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching QQ7 data in 6.229 seconds (success: False)
2025-05-06 18:47:26.377 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS device (QQ7)
2025-05-06 18:47:26.377 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] disconnecting BMS
2025-05-06 18:47:26.499 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] disconnected from BMS
```

### Additional context

nRF Connect log:
```
QQ7 (24:19:72:D9:F8:E7)
D 18:27:49.116 [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V 18:27:49.336 Connecting to 24:19:72:D9:F8:E7...
D 18:27:49.336 gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D 18:27:49.384 [Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I 18:27:49.384 Connected to 24:19:72:D9:F8:E7
I 18:27:49.386 MTU changed to: 517
V 18:27:49.413 Discovering services...
D 18:27:49.413 gatt.discoverServices()
V 18:27:49.415 Discovering services...
D 18:27:49.415 gatt.discoverServices()
D 18:27:49.419 [Callback] Services discovered with status: 0
I 18:27:49.419 Services discovered
V 18:27:49.426 Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Resolvable Private Address Only [R] (0x2AC9)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
- Client Supported Features [R W] (0x2B29)
- Database Hash [R] (0x2B2A)
Unknown Service (0000ff00-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N] (0000ff01-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [WNR] (0000ff02-0000-1000-8000-00805f9b34fb)
D 18:27:49.426 gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D 18:27:49.435 gatt.setCharacteristicNotification(0000ff01-0000-1000-8000-00805f9b34fb, true)
D 18:27:49.437 [Callback] Services discovered with status: 0
I 18:27:49.437 Services discovered
V 18:27:49.443 Generic Access (0x1800)
[... 378 lines cut ...]
A 18:28:34.990 "(0x) 7B-08-16-00-00-00-02-00-00-00-33-00-01-00-01-00-01-00-00-00-00-00-00-00-00-7D" received
I 18:28:35.353 Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value: (0x) 7B-74-1D-4B-53-5F-42-4C-45-5F-57-49-46-49-5F-56-65-72-31-2E-30-2E-30-5F-32-30-32-34-30-33-31-33-7D
A 18:28:35.353 "(0x) 7B-74-1D-4B-53-5F-42-4C-45-5F-57-49-46-49-5F-56-65-72-31-2E-30-2E-30-5F-32-30-32-34-30-33-31-33-7D" received
I 18:28:35.617 Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value: (0x) 7B-F4-03-02-03-01-7D
A 18:28:35.617 "(0x) 7B-F4-03-02-03-01-7D" received
I 18:28:35.740 Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value: (0x) 7B-01-20-00-55-14-8C-00-F0-00-CD-00-F0-F1-FB-63-9F-75-30-75-30-00-00-00-03-00-00-00-00-00-0E-00-00-00-64-7D
A 18:28:35.740 "(0x) 7B-01-20-00-55-14-8C-00-F0-00-CD-00-F0-F1-FB-63-9F-75-30-75-30-00-00-00-03-00-00-00-00-00-0E-00-00-00-64-7D" received
I 18:28:35.741 Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value: (0x) 7B-F5-02-02-39-7D
A 18:28:35.741 "(0x) 7B-F5-02-02-39-7D" received
I 18:28:35.797 Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value: (0x) 7B-09-03-FF-FF-FF-7D
A 18:28:35.797 "(0x) 7B-09-03-FF-FF-FF-7D" received
I 18:28:36.045 Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value: (0x) 7B-02-21-10-0C-E5-0C-CF-0C-E5-0C-D3-0C-E2-0C-CF-0C-E4-0C-A6-0C-E3-0C-D0-0C-E2-0C-D5-0C-E4-0C-CF-0C-E4-0C-D5-7D
A 18:28:36.045 "(0x) 7B-02-21-10-0C-E5-0C-CF-0C-E5-0C-D3-0C-E2-0C-CF-0C-E4-0C-A6-0C-E3-0C-D0-0C-E2-0C-D5-0C-E4-0C-CF-0C-E4-0C-D5-7D" received
I 18:28:36.341 Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value: (0x) 7B-03-09-04-0B-7D-0B-73-0B-73-0B-7D-7D
A 18:28:36.341 "(0x) 7B-03-09-04-0B-7D-0B-73-0B-73-0B-7D-7D" received
I 18:28:36.702 Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value: (0x) 7B-08-16-00-00-00-02-00-00-00-33-00-01-00-01-00-01-00-00-00-00-00-00-00-00-7D
A 18:28:36.702 "(0x) 7B-08-16-00-00-00-02-00-00-00-33-00-01-00-01-00-01-00-00-00-00-00-00-00-00-7D" received
I 18:28:37.037 Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value: (0x) 7B-74-1D-4B-53-5F-42-4C-45-5F-57-49-46-49-5F-56-65-72-31-2E-30-2E-30-5F-32-30-32-34-30-33-31-33-7D
A 18:28:37.037 "(0x) 7B-74-1D-4B-53-5F-42-4C-45-5F-57-49-46-49-5F-56-65-72-31-2E-30-2E-30-5F-32-30-32-34-30-33-31-33-7D" received
I 18:28:37.298 Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value: (0x) 7B-F4-03-02-03-01-7D
A 18:28:37.298 "(0x) 7B-F4-03-02-03-01-7D" received
I 18:28:37.359 Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value: (0x) 7B-F5-02-02-39-7D
A 18:28:37.359 "(0x) 7B-F5-02-02-39-7D" received

```

patman15, 2025-05-10:
Hi! I think you are not using the latest version 1.16.1. Can you please first update before O take a detailed look and send me a debug log. The newest version has some features that might help here.
Best regards,
Patrick

prodisz, 2025-05-10:
This is log from v. 1.16.1:
```
2025-05-10 14:23:06.085 DEBUG (MainThread) [custom_components.bms_ble] Device QQ7 (24:19:72:d9:f8:e7) detected as 'Jiabaida Smart BMS'
2025-05-10 14:23:07.745 DEBUG (MainThread) [custom_components.bms_ble] user step
2025-05-10 14:23:07.746 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JTX2CCQ2TCNF40JW3KRQTGZ0 version=1 domain=bms_ble title=QQ7 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=24:19:72:D9:F8:E7>
2025-05-10 14:23:07.746 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] initializing Jiabaida Smart BMS, BT address: 24:19:72:D9:F8:E7
2025-05-10 14:23:07.746 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for QQ7 (24:19:72:D9:F8:E7) as Jiabaida Smart BMS
2025-05-10 14:23:07.746 DEBUG (MainThread) [custom_components.bms_ble] QQ7: advertisement: {'name': 'QQ7', 'address': '24:19:72:D9:F8:E7', 'rssi': -70, 'manufacturer_data': {123: b'\x02\xff\xff}'}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': 'EC:94:CB:63:88:0A', 'advertisement': AdvertisementData(local_name='QQ7', manufacturer_data={123: b'\x02\xff\xff}'}, service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-70), 'device': BLEDevice(24:19:72:D9:F8:E7, QQ7), 'connectable': True, 'time': 941453.129731205, 'tx_power': None, 'raw': b'\x02\x01\x06\x03\x02\x00\xff\x04\tQQ7\x07\xff{\x00\x02\xff\xff}'}
2025-05-10 14:23:07.746 DEBUG (MainThread) [custom_components.bms_ble] QQ7: BMS data update
2025-05-10 14:23:07.746 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] connecting BMS
2025-05-10 14:23:07.746 DEBUG (MainThread) [bleak_retry_connector] 24:19:72:D9:F8:E7 - 24:19:72:D9:F8:E7: Connection attempt: 1
2025-05-10 14:23:07.746 INFO (MainThread) [habluetooth.wrappers] 24:19:72:D9:F8:E7 - QQ7: Found 1 connection path(s), preferred order: jbd-bms-ble (EC:94:CB:63:88:0A) (RSSI=-70) (failures=0) (in_progress=0) (score=-70.0)
25-05-10 14:23:08.224 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] TX BLE data #1 (WNR): dd a5 03 00 ff fd 77
2025-05-10 14:23:08.940 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] TX BLE data #2 (WNR): dd a5 03 00 ff fd 77
2025-05-10 14:23:10.370 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] TX BLE data #3 (WNR): dd a5 03 00 ff fd 77
2025-05-10 14:23:13.228 DEBUG (MainThread) [custom_components.bms_ble] QQ7: device communication timed out
2025-05-10 14:23:13.228 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching QQ7 data in 5.482 seconds (success: False)
2025-05-10 14:23:13.228 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS device (QQ7)
2025-05-10 14:23:13.228 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] disconnecting BMS
2025-05-10 14:23:13.348 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] disconnected from BMS

2025-05-10 14:23:18.496 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JTX2CCQ2TCNF40JW3KRQTGZ0 version=1 domain=bms_ble title=QQ7 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=24:19:72:D9:F8:E7>
2025-05-10 14:23:18.496 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] initializing Jiabaida Smart BMS, BT address: 24:19:72:D9:F8:E7
2025-05-10 14:23:18.496 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for QQ7 (24:19:72:D9:F8:E7) as Jiabaida Smart BMS
2025-05-10 14:23:18.496 DEBUG (MainThread) [custom_components.bms_ble] QQ7: advertisement: {'name': 'QQ7', 'address': '24:19:72:D9:F8:E7', 'rssi': -62, 'manufacturer_data': {123: b'\x02\xff\xff}'}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': 'EC:94:CB:63:88:0A', 'advertisement': AdvertisementData(local_name='QQ7', manufacturer_data={123: b'\x02\xff\xff}'}, service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-62), 'device': BLEDevice(24:19:72:D9:F8:E7, QQ7), 'connectable': True, 'time': 941463.97503646, 'tx_power': None, 'raw': b'\x02\x01\x06\x03\x02\x00\xff\x04\tQQ7\x07\xff{\x00\x02\xff\xff}'}
2025-05-10 14:23:18.496 DEBUG (MainThread) [custom_components.bms_ble] QQ7: BMS data update
2025-05-10 14:23:18.496 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] connecting BMS
2025-05-10 14:23:18.496 DEBUG (MainThread) [bleak_retry_connector] 24:19:72:D9:F8:E7 - 24:19:72:D9:F8:E7: Connection attempt: 1
2025-05-10 14:23:18.497 INFO (MainThread) [habluetooth.wrappers] 24:19:72:D9:F8:E7 - QQ7: Found 1 connection path(s), preferred order: jbd-bms-ble (EC:94:CB:63:88:0A) (RSSI=-62) (failures=0) (in_progress=0) (score=-62.0)
2025-05-10 14:23:18.977 DEBUG (MainThread) [bleak_retry_connector] 24:19:72:D9:F8:E7 - 24:19:72:D9:F8:E7: Connected after 1 attempts
2025-05-10 14:23:19.010 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] TX BLE data #1 (WNR): dd a5 03 00 ff fd 77
2025-05-10 14:23:19.726 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] TX BLE data #2 (WNR): dd a5 03 00 ff fd 77
2025-05-10 14:23:21.156 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] TX BLE data #3 (WNR): dd a5 03 00 ff fd 77
2025-05-10 14:23:24.014 DEBUG (MainThread) [custom_components.bms_ble] QQ7: device communication timed out
2025-05-10 14:23:24.014 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching QQ7 data in 5.518 seconds (success: False)
2025-05-10 14:23:24.014 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS device (QQ7)
2025-05-10 14:23:24.014 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] disconnecting BMS
2025-05-10 14:23:24.095 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::QQ7:F8E7)] disconnected from BMS
```

patman15, 2025-05-10:
Are you able to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery? Looks like this is either not a JBD device or uses a new protocol.

prodisz, 2025-05-10:
nRF Connect log from first post was captured while using Android vendor app.

prodisz, 2025-05-10:
[btsnooz_hci.log](https://github.com/user-attachments/files/20140025/btsnooz_hci.log)

patman15, 2025-05-14:
Hi! The BT capture is for some reason limit to only a few bytes of data for each message (looks truncated), but I can infer from the nRF connect trace. Unfortunately, there the commands (sent to the BMS) are missing there. If you have additional data, it would be cool, but I will try to make a test version to see how the BMS reacts.

prodisz, 2025-05-18:
I found an old rooted phone, I hope this log will help:

[btsnoop_hci2.log](https://github.com/user-attachments/files/20273199/btsnoop_hci2.log)

patman15, 2025-05-18:
Looks pretty good, will have a detailed look soon, to guess the values, but with that information it is quite probably I can make it happen.

prodisz, 2025-05-18:
Don't ask me. I don’t really understand it myself, but you said that data looks truncated.
I read that Xiaomi phones have issues with the btsnoop log. That’s why I ended up using a old, custom rom, rooted phone.

prodisz, 2025-05-26:
The app has an option to rename the device. I found a test of a similar model on YouTube, and it was probably originally called “HSKS-25A-118”.
https://www.youtube.com/watch?v=xj51tjqN99g

patman15, 2025-05-26:
@prodisz I made a [first very basic version](https://github.com/patman15/BMS_BLE-HA/tree/303-add-braun-power-battery) that you could install via [manual install](https://github.com/patman15/BMS_BLE-HA/tree/303-add-braun-power-battery?tab=readme-ov-file#installation).
What I would need is either the [debug log + the information the app shows] or [another BT capture + the information the app shows] in order to map the values. Currently (I think) I was only able to map voltage and current charge and even there I'm not 100% sure. First is probably easier, if the code works.

patman15, 2025-05-27:
@prodisz I added a number of further sensors. Just update from the branch. What I'm still missing is the "Fault Data". So if you are able to generate some fault and send me the debug log of it, I could also add them. Please also verify that all other data makes sense to you. You can enable/disable the integration in the 3-dots menu next to the battery in settings. That way you can switch between integration and app.

patman15, 2025-06-06:
I'm not 100% sure if I have the right data, but I implemented a readout for the faults that should be `ok` when there is no alert (hopefully). I still need to fix the battery detection in order to merge it for a release. You can update the branch for testing, I will follow up later this month.

prodisz, 2025-06-12:
Problem entity is not setting up:
```
2025-06-12 12:38:01.560 DEBUG (MainThread) [custom_components.bms_ble] Unloaded config entry: 24:19:72:D9:F8:E7, ok? True!
2025-06-12 12:38:01.560 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS (QQ7)
2025-06-12 12:38:01.560 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] disconnecting BMS
2025-06-12 12:38:01.650 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] disconnected from BMS
2025-06-12 12:38:01.650 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JX520FBWPWW826K0ZGQ9PGSH version=1 domain=bms_ble title=QQ7 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=24:19:72:D9:F8:E7>
2025-06-12 12:38:01.652 DEBUG (MainThread) [custom_components.bms_ble] Failed to discover device 24:19:72:D9:F8:E7 via Bluetooth
2025-06-12 12:38:01.653 DEBUG (MainThread) [custom_components.bms_ble] Bluetooth device detected: <BluetoothServiceInfoBleak name=QQ7 address=24:19:72:D9:F8:E7 rssi=-63 manufacturer_data={123: b'\x02\xff\xff}'} service_data={} service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'] source=EC:94:CB:63:88:0A connectable=True time=79118.10948599 tx_power=None raw=b'\x02\x01\x06\x03\x02\x00\xff\x04\tQQ7\x07\xff{\x00\x02\xff\xff}'>
2025-06-12 12:38:01.654 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JX520FBWPWW826K0ZGQ9PGSH version=1 domain=bms_ble title=QQ7 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=24:19:72:D9:F8:E7>
2025-06-12 12:38:01.654 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] initializing Braun Power Smart BMS, BT address: 24:19:72:D9:F8:E7
2025-06-12 12:38:01.654 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for QQ7 (24:19:72:D9:F8:E7) as Braun Power Smart BMS
2025-06-12 12:38:01.655 DEBUG (MainThread) [custom_components.bms_ble] QQ7: advertisement: {'name': 'QQ7', 'address': '24:19:72:D9:F8:E7', 'rssi': -63, 'manufacturer_data': {123: b'\x02\xff\xff}'}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': 'EC:94:CB:63:88:0A', 'advertisement': AdvertisementData(local_name='QQ7', manufacturer_data={123: b'\x02\xff\xff}'}, service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-63), 'device': BLEDevice(24:19:72:D9:F8:E7, QQ7), 'connectable': True, 'time': 79118.10948599, 'tx_power': None, 'raw': b'\x02\x01\x06\x03\x02\x00\xff\x04\tQQ7\x07\xff{\x00\x02\xff\xff}'}
2025-06-12 12:38:01.655 DEBUG (MainThread) [custom_components.bms_ble] QQ7: BMS data update
2025-06-12 12:38:01.655 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] connecting BMS
2025-06-12 12:38:01.655 INFO (MainThread) [habluetooth.wrappers] 24:19:72:D9:F8:E7 - QQ7: Found 1 connection path(s), preferred order: jbd-bms-ble (EC:94:CB:63:88:0A) (RSSI=-63) (failures=0) (in_progress=0) (score=-63.0)
2025-06-12 12:38:01.979 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] TX BLE req #1 (WNR): 7b 01 00 7d
2025-06-12 12:38:02.154 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] RX BLE data (start): bytearray(b'{\x01 \x00N\x15%\x01"\x01\x10\x01\x18\x11\x8f\\Cu0u0\x00\x00\x00\x16\x00\x00\x00\x00\x00\r\x00\x00\x00d}')
2025-06-12 12:38:02.155 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] TX BLE req #1 (WNR.): 7b 02 00 7d
2025-06-12 12:38:02.266 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] RX BLE data (start): bytearray(b'{\x02!\x10\r8\r5\r4\r7\r5\r6\r6\rR\r7\r6\r4\r7\r5\r6\r4\r6}')
2025-06-12 12:38:02.266 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] TX BLE req #1 (WNR.): 7b 03 00 7d
2025-06-12 12:38:02.402 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] RX BLE data (start): bytearray(b'{\x03\t\x04\x0b\xc3\x0b\xb9\x0b\xb9\x0b\xb9}')
2025-06-12 12:38:02.402 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] TX BLE req #1 (WNR.): 7b 08 00 7d
2025-06-12 12:38:02.568 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] RX BLE data (start): bytearray(b'{\x08\x16\x00\x00\x00\x02\x00\x00\x01\x83\x00\x01\x00\x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00}')
2025-06-12 12:38:02.568 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] TX BLE req #1 (WNR.): 7b 09 00 7d
2025-06-12 12:38:02.671 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] RX BLE data (start): bytearray(b'{\t\x03\xff\xff\xff}')
2025-06-12 12:38:02.671 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] TX BLE req #1 (WNR.): 7b 74 00 7d
2025-06-12 12:38:02.775 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] RX BLE data (start): bytearray(b'{t\x1dKS_BLE_WIFI_Ver1.0.0_20240313}')
2025-06-12 12:38:02.776 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] TX BLE req #1 (WNR.): 7b f4 00 7d
2025-06-12 12:38:02.814 ERROR (MainThread) [homeassistant.components.google_assistant.http] Request for https://homegraph.googleapis.com/v1/devices:reportStateAndNotification failed: 404
2025-06-12 12:38:02.882 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] RX BLE data (start): bytearray(b'{\xf4\x03\x02\x03\x01}')
2025-06-12 12:38:02.882 DEBUG (MainThread) [custom_components.bms_ble.braunpwr_bms::QQ7:F8E7)] TX BLE req #1 (WNR.): 7b f5 00 7d
[... cut ...] 765421187317295098848346112(<class 'int'>
2025-06-12 12:38:03.103 INFO (MainThread) [homeassistant.components.sensor] Setting up bms_ble.sensor
2025-06-12 12:38:03.107 WARNING (Recorder) [homeassistant.components.recorder.table_managers.state_attributes] State is not JSON serializable: <state binary_sensor.qq7_problem=on; problem_code=44601492406480209765421187317295098848346112, device_class=problem, friendly_name=QQ7 Problem @ 2025-06-12T12:38:03.103165+02:00>: Integer exceeds 64-bit range
```

patman15, 2025-06-13:
Oh, there is an issue with the `problem_code` and the GUI. I'll need to fix that. Will let you know when I'm back to my development system.

prodisz, 2025-06-30:
Now the entity binary_sensor.qq7_problem is always in the on state, even when there are no active warnings or alarms.

I created a template sensor to extract the raw problem code:
```
template:
  - name: "QQ7 Problem Code"
    state: >
      {{ state_attr('binary_sensor.qq7_problem', 'problem_code') }}
```
However, the value of problem_code does not change — it remains constant:
`0x20000020d0001000100010000000000000000`
This happens even when I trigger a warning or alarm on the battery.

patman15, 2025-06-30:
Can you provide me a few minutes of a debug log, because I did not change the way I extract the value, so maybe the value itself is wrongly taken and there are some flags in that have a different meaning. The sensor is always on because, the `problem_code` is not zero. Maybe I need to take only the last 16 nibbles and cut the rest ... 🤔

patman15, 2025-07-19:
@prodisz are there any open points? I changed the detection pattern now to `HSKS-*` to not conflict with JBD. I would merge the changes now if you are ok.

prodisz, 2025-07-19:
I'm currently waiting for the second, non-rebranded battery unit to arrive (expected delivery in August), so I'll be able to check then whether the pattern matches.

As for the current version, the error code doesn't change during cell protection warnings or alarms.

patman15, 2025-07-19:
Ok, can you send me short logs when there is an alarm or an active protection? Then I can try to fix it by comparing it with other logs.

prodisz, 2025-08-20:
I finally received the second, non-branded energy storage unit: here are the logs from it.

```
2025-08-18 12:32:57.070 DEBUG (MainThread) [custom_components.bms_ble] confirm step for BL-25F-170-118
2025-08-18 12:32:58.649 DEBUG (MainThread) [custom_components.bms_ble] confirm step for BL-25F-170-118
2025-08-18 12:32:58.650 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01K2YBTJGSG4SAMB101DGBJN4B version=1 domain=bms_ble title=BL-25F-170-118 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=B4:2F:03:B2:54:62>
2025-08-18 12:32:58.650 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] initializing Jiabaida Smart BMS, BT address: B4:2F:03:B2:54:62
2025-08-18 12:32:58.650 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for BL-25F-170-118 (B4:2F:03:B2:54:62) as Jiabaida Smart BMS
2025-08-18 12:32:58.650 DEBUG (MainThread) [custom_components.bms_ble] BL-25F-170-118: advertisement: {'name': 'BL-25F-170-118', 'address': 'B4:2F:03:B2:54:62', 'rssi': -76, 'manufacturer_data': {123: b'\x02\xff\xff}'}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': 'EC:94:CB:63:88:0A', 'advertisement': AdvertisementData(local_name='BL-25F-170-118', manufacturer_data={123: b'\x02\xff\xff}'}, service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-76), 'device': BLEDevice(B4:2F:03:B2:54:62, BL-25F-170-118), 'connectable': True, 'time': 166490.532804245, 'tx_power': None, 'raw': b'\x02\x01\x06\x03\x02\x00\xff\x0f\tBL-25F-170-118\x07\xff{\x00\x02\xff\xff}'}
2025-08-18 12:32:58.650 DEBUG (MainThread) [custom_components.bms_ble] BL-25F-170-118: BMS data update
2025-08-18 12:32:58.650 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] connecting BMS
2025-08-18 12:32:59.877 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] TX BLE req #1 (WNR): dd a5 03 00 ff fd 77
2025-08-18 12:33:00.128 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] TX BLE request timed out.
2025-08-18 12:33:00.129 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] TX BLE req #2 (WNR): dd a5 03 00 ff fd 77
2025-08-18 12:33:00.629 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] TX BLE request timed out.
2025-08-18 12:33:00.630 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] TX BLE req #3 (WNR): dd a5 03 00 ff fd 77
2025-08-18 12:33:00.952 INFO (MainThread) [custom_components.xiaomi_miot.core.device.xtl.vacuum.xm2216] Device updated: {'sensor.vacuum.status': 'naładowany', 'only_info': False}
2025-08-18 12:33:01.632 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] TX BLE request timed out.
2025-08-18 12:33:01.632 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] TX BLE req #1 (!WNR): dd a5 03 00 ff fd 77
2025-08-18 12:33:01.711 WARNING (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] TX BLE request error, retrying connection (BleakError)
2025-08-18 12:33:01.712 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] disconnecting BMS
2025-08-18 12:33:01.736 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] disconnected from BMS
2025-08-18 12:33:01.736 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] connecting BMS
2025-08-18 12:33:01.736 INFO (MainThread) [habluetooth.wrappers] B4:2F:03:B2:54:62 - BL-25F-170-118: Found 1 connection path(s), preferred order: jbd-bms-ble (EC:94:CB:63:88:0A) (RSSI=-76) (failures=0) (in_progress=0) (score=-76.0)
2025-08-18 12:33:02.126 DEBUG (MainThread) [custom_components.bms_ble] BL-25F-170-118: BMS communication timed out, check signal strength (-76 dBm)
2025-08-18 12:33:02.127 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching BL-25F-170-118 data in 3.477 seconds (success: False)
2025-08-18 12:33:02.127 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS (BL-25F-170-118)
2025-08-18 12:33:02.127 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] disconnecting BMS
2025-08-18 12:33:02.153 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] disconnected from BMS
2025-08-18 12:33:07.466 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01K2YBTJGSG4SAMB101DGBJN4B version=1 domain=bms_ble title=BL-25F-170-118 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=B4:2F:03:B2:54:62>
2025-08-18 12:33:07.466 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::BL-25F-170-118:5462)] initializing Jiabaida Smart BMS, BT address: B4:2F:03:B2:54:62
2025-08-18 12:33:07.467 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for BL-25F-170-118 (B4:2F:03:B2:54:62) as Jiabaida Smart BMS
2025-08-18 12:33:07.467 DEBUG (MainThread) [custom_components.bms_ble] BL-25F-170-118: advertisement: {'name': 'BL-25F-170-118', 'address': 'B4:2F:03:B2:54:62', 'rssi': -76, 'manufacturer_data': {123: b'\x02\xff\xff}'}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': 'EC:94:CB:63:88:0A', 'advertisement': AdvertisementData(local_name='BL-25F-170-118', manufacturer_data={123: b'\x02\xff\xff}'}, service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-76), 'device': BLEDevice(B4:2F:03:B2:54:62, BL-25F-170-118), 'connectable': True, 'time [... cut ...]  300, 'cycles': 65, 'problem_code': 546, 'cell_voltages': [3.446, 3.393, 3.47, 3.516, 3.439, 3.368, 3.451, 3.492, 3.513, 3.514, 3.493, 3.516, 3.513, 3.471, 3.46, 3.441], 'temp_values': [33.0, 33.0, 32.0, 32.0], 'delta_voltage': 0.148, 'cycle_capacity': 16644.78, 'power': -209.752, 'battery_charging': False, 'runtime': 285676, 'temperature': 32.5, 'problem': True}
2025-08-20 15:43:55.343 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching QQ7 data in 1.114 seconds (success: True)
```

patman15, 2025-08-20:
@prodisz I finally managed to mock the battery and figured out the protection flags. Code adapted, I also fixed the JBD conflict. So I think we are set to merge. Can you do a final test?

prodisz, 2025-08-21:
Yes, it’s the same device, just not rebranded by the reseller. It’s supported by the same Android app. After the update it’s not detected by the add-on at all