patman15/BMS_BLE-HA #306: DALY BMS

Hene01eb, 2025-05-11:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Dobrý den, je možnost u daly bms přepsat (z haos) stav nabytí baterie v % (z nějaké hodnoty třeba na 100%), mám totiž 2 bms daly a tyto stavy % baterii se po čase rozejdou a potřebuju je sjednotit optimálně z home assistant. je to reálné?

patman15, 2025-05-11:
Please raise your question(s) in English! Thanks.

Hene01eb, 2025-05-11:
Hello, is it possible to overwrite (from haos) the battery charge status in % (from some value, for example, to 100%), because I have 2 bms daly and these % battery statuses will diverge over time and I need to optimally unify them from home assistant. is this realistic?

patman15, 2025-05-11:
You could derive the new value from the original sensor using a [template sensor](https://www.home-assistant.io/integrations/template/). If you need more help on that, I suggest to open a thread with the [HA community](https://community.home-assistant.io/c/configuration/11)

Hene01eb, 2025-05-11:
Maybe I'd rather get advice on how to do it exactly so that the % value in the BMS is overwritten.

patman15, 2025-05-11:
Sorry, I do not have experience with the BMS. The SoC value calculation is device specific so you need to ask people who have experience with that problem. Can't help here, as I do not own one.

closed 2025-05-16 by patman15