patman15/BMS_BLE-HA #309: Some JBD BMS start with a "JBD-" string

timonsku, 2025-05-11:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

I have JBD BMS, it mostly matches the typical pattern but it has an additional "JBD-" infront. F.e. JBD-SP16S020-L8S
It works fine when I added a custom matching string. Not sure if the regex engine for the BLE match string feature supports optional matches but that could otherwise maybe be added to the existing JBD matching strings.

patman15, 2025-05-12:
Hi! Thanks for the info, could you please provide me the data from the [advertisement monitor](https://github.com/patman15/BMS_BLE-HA#if-your-device-is-not-recognized) so I can add a regression test?

timonsku, 2025-05-13:
Here the output with MAC address removed

{"name":"JBD-SP16S020-L8S","address":"<removed>","rssi":-81,"manufacturer_data":{},"service_data":{},"service_uuids":["00001800-0000-1000-8000-00805f9b34fb","00001801-0000-1000-8000-00805f9b34fb","0000180a-0000-1000-8000-00805f9b34fb","0000ff00-0000-1000-8000-00805f9b34fb","00010203-0405-0607-0809-0a0b0c0d1912"],"source":"<removed>","connectable":true,"time":1747069173.930697,"tx_power":null}

patman15, 2025-05-14:
@timonsku I added the pattern to the [branch](https://github.com/patman15/BMS_BLE-HA/tree/309-some-jbd-bms-start-with-a-jbd--string). Could you test via [manual install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#installation) to prove it works?

timonsku, 2025-05-14:
Installing it manually made it not find the battery. Though adding the changes to manifest.json and jbd_bms.py to 1.6.1 downloaded via HACS worked. So far I had also modified this version just with my specific version string hardcoded but JBD_* worked fine too.
I assume something went wrong when I manually installed it from and it otherwise works fine.

patman15, 2025-05-14:
Sorry, I had a typo and missed that the tests failed. It should work now if you update once again.