patman15/BMS_BLE-HA #311: EEL DIY Battery Box with Seplos V2 BMS

Bushrider, 2025-05-12:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Hi, the bms is showing up in the bluetooth proxy monitor: 
{"name":"BP00","address":"60:6E:41:49:2D:BF","rssi":-92,"manufacturer_data":{},"service_data":{},"service_uuids":[],"source":"08:D1:F9:E2:D5:CE","connectable":true,"time":1747088832.44798,"tx_power":null}

When i'm trying to add the device i got a : "No supported devices found via Bluetooth."

I can connect through the app.
I'm disconnected from the app when i try to connect via homeassistant

### Additional context

https://www.eelbattery.com/EEL-48V-16S-Server-Rack-DIY-Unit-BOX-Built-in-Seplos-BMS-stackable-typesuit-for-230280302Ah-cells-EU-Stock-p4491159.html

https://drive.google.com/file/d/1VkqUj50UKesQP7QZY3ZWnT9a0iyax6rt/view?usp=sharing

Bushrider, 2025-05-12:
Apparently this is an "EN BMS"  same hardware as seplos V2 but not the firmware...

patman15, 2025-05-13:
Please check your BT setup first, your `RSSI=-92`. You need to get around `-75 dBm` to ensure that the system can connect without issues.

Bushrider, 2025-05-13:
> Please check your BT setup first, your `RSSI=-92`. You need to get around `-75 dBm` to ensure that the system can connect without issues.

Thanks,
Installing the BT proxy nearer makes the difference but now i'm getting `Échec de la configuration, nouvel essai: device communication failed, check signal strength (-89 dBm): 60:6E:41:49:2D:BF - 60:6E:41:49:2D:BF: Failed to connect after 9 attempt(s): Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish: Interference/range; External Bluetooth adapter w/extension may help; Extension cables reduce USB 3 port interference (BleakAbortedError)`

I'll try to install it closer this week !

Bushrider, 2025-05-13:
Everything is ok now that i installed the BT adapter closer ! 
Thanks for the great job !