patman15/BMS_BLE-HA #312: Device “CS Batteries Smart Power Lithium 300Ah” not recognized

Ton3016, 2025-05-14:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Please add this battery
{"name":"Camper LiFePo","address":"A4:C1:37:44:91:2C","rssi":-44,"manufacturer_data":{"37164":"4437c1a4"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"DC:A6:32:50:E6:E0","connectable":true,"time":1747199858.8299744,"tx_power":null}

patman15, 2025-05-14:
You could test [this branch](https://github.com/patman15/BMS_BLE-HA/tree/309-some-jbd-bms-start-with-a-jbd--string) via [manual install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#installation) and rename your device to `JBD-`...?

Ton3016, 2025-05-14:
Did a manual install and changed the name.
When integration is loaded the message “No supported devices found via Bluetooth” appears. Integration is not loaded.
Battery is advertising in Bluetooth like before with changed name.

patman15, 2025-05-14:
Sorry, I had a typo and missed that the tests failed. It should work now if you update once again.

closed 2025-05-15 by Ton3016