patman15/BMS_BLE-HA #314: Supervolt SV12V.......

Daflo13, 2025-05-14:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

New Supervolt-Batterietyp: SV12V....... integration

### Additional context

{"name":"SV12V150224041","address":"02:00:07:A0:30:F7","rssi":-39,"manufacturer_data":{"33384":"01020007a030f7"},"service_data":{},"service_uuids":[],"source":"D8:3A:DD:19:8B:65","connectable":true,"time":1747241922.5906131,"tx_power":null}

https://support.supervolt.de/attachments/token/PM3Voe6dor9a3UPMV2QoepUUl/? Name=rs485-uart-rs232+Protocol_BT.doc

patman15, 2025-05-14:
Could you please test [this branch](https://github.com/patman15/BMS_BLE-HA/tree/314-supervolt-sv12v) via [manual install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#installation)?

patman15, 2025-05-15:
There are plenty of issues visible with your BT connection. I guess you need to check your BT proxy configuration. But in principle it seems to work somehow.

Daflo13, 2025-05-15:
Here are still data

[image]

There seems to be a connection, but no data comes to me

Daflo13, 2025-05-15:
Is it possible that there are problems because there are two batteries?

patman15, 2025-05-15:
Can be, but there are also connection errors to your proxy visible in the log.

closed 2025-05-17 by patman15