patman15/BMS_BLE-HA #317: ECO-WORTHY 150Ah (plastic case): device communication failed: Characteristic ff01 was not found! (BleakCharacteristicNotFoundError)

awilliam, 2025-05-15:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Please add support for [ECO-WORTHY 150Ah LiFePO4 12.8v battery](https://www.eco-worthy.com/products/lifepo4-12v-150ah-lithium-iron-phosphate-battery-with-bluetooth-and-low-temperature-protection)

Integration currently reports:
`Failed setup, will retry: BMS communication failed: Characteristic ff01 was not found! (BleakCharacteristicNotFoundError) `

Tried updating to commit "29c4c98 Update README.md" for latest ECO-WORTHY updates, but did not help.

Advertisement monitor info:
`
{"name":"ECOBF9F","address":"ED:0B:CE:2E:BF:9F","rssi":-50,"manufacturer_data":{"3053":"ce2ebf9f"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb","00000001-0000-1000-8000-00805f9b34fb"],"source":"C4:5B:BE:38:FC:86","connectable":true,"time":1747274548.0745826,"tx_power":null}
`

Debug log:
```
2025-05-14 20:08:59.806 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JTZYK7CYX8CM3P97G0DYEKKK version=1 domain=bms_ble title=ECOBF9F state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=ED:0B:CE:2E:BF:9F>
2025-05-14 20:08:59.807 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECOBF9F:BF9F)] initializing Jiabaida Smart BMS, BT address: ED:0B:CE:2E:BF:9F
2025-05-14 20:08:59.807 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for ECOBF9F (ED:0B:CE:2E:BF:9F) as Jiabaida Smart BMS
2025-05-14 20:08:59.807 DEBUG (MainThread) [custom_components.bms_ble] ECOBF9F: advertisement: {'name': 'ECOBF9F', 'address': 'ED:0B:CE:2E:BF:9F', 'rssi': -58, 'manufacturer_data': {3053: b'\xce.\xbf\x9f'}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb', '00000001-0000-1000-8000-00805f9b34fb'], 'source': 'C4:5B:BE:38:FC:86', 'advertisement': AdvertisementData(local_name='ECOBF9F', manufacturer_data={3053: b'\xce.\xbf\x9f'}, service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb', '00000001-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-58), 'device': BLEDevice(ED:0B:CE:2E:BF:9F, ECOBF9F), 'connectable': True, 'time': 2092083.548391823, 'tx_power': None}
2025-05-14 20:08:59.808 DEBUG (MainThread) [custom_components.bms_ble] ECOBF9F: BMS data update
2025-05-14 20:08:59.808 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECOBF9F:BF9F)] connecting BMS
2025-05-14 20:08:59.808 DEBUG (MainThread) [bleak_retry_connector] ED:0B:CE:2E:BF:9F - ED:0B:CE:2E:BF:9F: Connection attempt: 1
2025-05-14 20:09:00.428 DEBUG (MainThread) [bleak_retry_connector] ED:0B:CE:2E:BF:9F - ED:0B:CE:2E:BF:9F: Connected after 1 attempts
2025-05-14 20:09:00.428 INFO (MainThread) [custom_components.bms_ble.jbd_bms::ECOBF9F:BF9F)] failed to initialize BMS connection (BleakCharacteristicNotFoundError)
2025-05-14 20:09:00.428 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECOBF9F:BF9F)] disconnecting BMS
2025-05-14 20:09:00.522 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECOBF9F:BF9F)] disconnected from BMS
2025-05-14 20:09:00.523 DEBUG (MainThread) [custom_components.bms_ble] ECOBF9F: BMS communication failed: Characteristic ff01 was not found! (BleakCharacteristicNotFoundError)
2025-05-14 20:09:00.523 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching ECOBF9F data in 0.716 seconds (success: False)
2025-05-14 20:09:00.524 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS (ECOBF9F)

```

patman15, 2025-05-15:
That reads like an issue with your BT setup. If you have another BT adapter at hand, please try that. If you are using ESPproxy, please minimise it and post it here if it does not work.
With the mentioned commit I would expect it to work, but the error message indicates an underlying Bluetooth issue.

awilliam, 2025-05-16:
I disabled all my esp32 BT proxies, the battery is now ~6ft from my HAOS box with Intel BT that works well with Airthings and Mopeka sensors.  I do somehow get some more service UUIDs now:

`{"name":"ECOBF9F","address":"ED:0B:CE:2E:BF:9F","rssi":-72,"manufacturer_data":{"3053":"ce2ebf9f"},"service_data":{},"service_uuids":["00000001-0000-1000-8000-00805f9b34fb","00001800-0000-1000-8000-00805f9b34fb","00001801-0000-1000-8000-00805f9b34fb","0000ff00-0000-1000-8000-00805f9b34fb"],"source":"48:51:C5:E8:B3:FD","connectable":true,"time":1747368049.9432573,"tx_power":null}`

But ultimately the integration gets the same failure:
`Failed setup, will retry: BMS communication failed: Characteristic ff01 was not found! (BleakCharacteristicNotFoundError)`

```
2025-05-15 22:05:17.459 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JTZYK7CYX8CM3P97G0DYEKKK version=1 domain=bms_ble title=ECOBF9F state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=ED:0B:CE:2E:BF:9F>
2025-05-15 22:05:17.459 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECOBF9F:BF9F)] initializing Jiabaida Smart BMS, BT address: ED:0B:CE:2E:BF:9F
2025-05-15 22:05:17.459 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for ECOBF9F (ED:0B:CE:2E:BF:9F) as Jiabaida Smart BMS
2025-05-15 22:05:17.459 DEBUG (MainThread) [custom_components.bms_ble] ECOBF9F: advertisement: {'name': 'ECOBF9F', 'address': 'ED:0B:CE:2E:BF:9F', 'rssi': -75, 'manufacturer_data': {3053: b'\xce.\xbf\x9f'}, 'service_data': {}, 'service_uuids': ['00000001-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000ff00-0000-1000-8000-00805f9b34fb'], 'source': '48:51:C5:E8:B3:FD', 'advertisement': AdvertisementData(local_name='ECOBF9F', manufacturer_data={3053: b'\xce.\xbf\x9f'}, service_uuids=['00000001-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000ff00-0000-1000-8000-00805f9b34fb'], rssi=-75), 'device': BLEDevice(ED:0B:CE:2E:BF:9F, ECOBF9F), 'connectable': True, 'time': 2185461.09359643, 'tx_power': None}
2025-05-15 22:05:17.459 DEBUG (MainThread) [custom_components.bms_ble] ECOBF9F: BMS data update
2025-05-15 22:05:17.459 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECOBF9F:BF9F)] connecting BMS
2025-05-15 22:05:17.460 DEBUG (MainThread) [bleak_retry_connector] ED:0B:CE:2E:BF:9F - ED:0B:CE:2E:BF:9F: Connection attempt: 1
2025-05-15 22:05:18.931 DEBUG (MainThread) [bleak_retry_connector] ED:0B:CE:2E:BF:9F - ED:0B:CE:2E:BF:9F: Connected after 1 attempts
2025-05-15 22:05:18.932 INFO (MainThread) [custom_components.bms_ble.jbd_bms::ECOBF9F:BF9F)] failed to initialize BMS connection (BleakCharacteristicNotFoundError)
2025-05-15 22:05:18.932 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECOBF9F:BF9F)] disconnecting BMS
2025-05-15 22:05:21.272 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECOBF9F:BF9F)] disconnected from BMS
2025-05-15 22:05:21.273 DEBUG (MainThread) [custom_components.bms_ble] ECOBF9F: BMS communication failed: Characteristic ff01 was not found! (BleakCharacteristicNotFoundError)
2025-05-15 22:05:21.273 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching ECOBF9F data in 3.814 seconds (success: False)
2025-05-15 22:05:21.273 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS (ECOBF9F)
```

BLE phone scanner shows:

[image]

mmeudy, 2025-05-16:
I have the same Problem with the same Battery

patman15, 2025-05-16:
Hmm, strange. Probably it's a different BMS but just looks like one the integration knows. 🤔 
Can someone run the [service explorer script](https://github.com/hbldh/bleak/blob/develop/examples/service_explorer.py) and/or (even better) see if you are able to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery?
I need to identify the used protocol to identify the BMS.

brad2388, 2025-05-16:
I have the same problem. Im unable to connect to this battery with the overkill app also. Just the ecoworthy app works.

awilliam, 2025-05-16:
Bluetooth HCI snoop log
[btsnooz_hci.log.zip](https://github.com/user-attachments/files/20259402/btsnooz_hci.log.zip)

patman15, 2025-05-17:
Hi! I'm sorry, I merged the detection for this type of batteries lacking feedback from the original requester in #282. Unfortunately, I reviewed it again and this type of batteries use classic Bluetooth not BLE. So I can't support them for technical reasons. I need to fix it by removing the detection again. Sorry, for raising wrong hopes.

closed 2025-05-17 by patman15