patman15/BMS_BLE-HA #319: add support for CHINS 48V golf cart battery (JBD BMS)

ebuckhalt, 2025-05-16:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

I have a CHINS 48V battery with a 200A JBD BMS,  it had an identifier like J-* I added that to the pattern matching in the plugins/jbd_bms.py  file and it worked just fine.  If you need screenshots or more info I can get it.  This battery used the BAT-BMS app by default, but I used Overkill Solar as well and that is where I found the identifier needed.

### Additional context

Here is the advertisment info. 

{"name":"GJ-cart","address":"10:A5:62:66:XX:XX","rssi":-84,"manufacturer_data":{},"service_data":{},"service_uuids":["00001800-0000-1000-8000-00805f9b34fb","00001801-0000-1000-8000-00805f9b34fb","0000180a-0000-1000-8000-00805f9b34fb","0000ff00-0000-1000-8000-00805f9b34fb","00010203-0405-0607-0809-0a0b0c0d1912"],"source":"8C:68:8B:20:XX:XX","connectable":true,"time":1747357933.3500855,"tx_power":null}

patman15, 2025-05-16:
Unfortunately, JBD does not manage to provide reasonable manufacturer ID or data to make a proper identification without plenty of name patterns. I think your changes just worked by accident. HA does not accept patterns with wildcards in the first 3 letters. But since your device is anyway called `GJ-` there is no issue. Probably, your low `RSSI` was the actual issue?

ebuckhalt, 2025-05-16:
Good to know, so it sounds like the solution was using the factory 
recommended app to change the "name" to something that matched in that 
list, in this case I went from golf-cart to GJ-Cart. Thank you for the 
information.  I will cancel/remove this request if possible.

-eric

Links:
------
[1] 
https://github.com/patman15/BMS_BLE-HA/issues/319#issuecomment-2887419181
[2] 
https://github.com/

closed 2025-05-16 by ebuckhalt