patman15/BMS_BLE-HA #326: DCHouse Smart Battery

Cnecne59, 2025-05-21:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

{"name":"DCHOUSE 01_4F9F","address":"E2:E0:5A:72:A7:45","rssi":-68,"manufacturer_data":{"57570":"5a72a745"},"service_data":{},"service_uuids":["0000fff0-0000-1000-8000-00805f9b34fb"],"source":"90:15:06:FA:74:F6","connectable":true,"time":1747478143.5196407,"tx_power":null}

### Additional context

https://www.dchousepower.com/products/12v-100ah-3-0-portable-deep-cycle-lifepo4-lithium-battery

Thanks in advance!

patman15, 2025-05-22:
@Cnecne59 please do a [manual install](https://github.com/patman15/BMS_BLE-HA#installation) using [this branch](https://github.com/patman15/BMS_BLE-HA/tree/326-dchouse-smart-battery). A debug log would be nice to verify that everything is ok. Let me know if it works for you!

Cnecne59, 2025-05-22:
Ok, i will install the branch!
But the battery is broken and i am waiting for a new one.
So it can take some days before I can do the install 
Regards
Christoph

patman15, 2025-05-29:
Since it won't hurt others, I will merge this for the upcoming (beta) release. Ticket will get closed, just ping me or raise a new ticket in case you experience issues when your battery is back.
Have fun, Patrick

Cnecne59, 2025-05-29:
I've tried the version 1.18.1, but the same response that the bat is not found. In the advertisment-monitor there is an entry with the name "DCHOUSE 01_D2BF".
The device-information:
{"name":"DCHOUSE 01_D2BF","address":"E2:E7:79:8F:4C:66","rssi":-55,"manufacturer_data":{"59362":"798f4c66"},"service_data":{},"service_uuids":["0000fff0-0000-1000-8000-00805f9b34fb"],"source":"90:15:06:FA:74:F6","connectable":true,"time":1748525643.8518314,"tx_power":null}

I see that the manufacturer_data is different to my first post!

patman15, 2025-05-29:
@Cnecne59 always when you think "what can go wrong" ...
Could you please try [this branch](https://github.com/patman15/BMS_BLE-HA/tree/fix-ecoworthy-dchouse-pattern) before I do another beta?

Cnecne59, 2025-05-29:
I thought, that i entered the screenshots from HA, and i will add the debug log:

[home-assistant_bms_ble_2025-05-29T19-14-35.711Z.log](https://github.com/user-attachments/files/20510391/home-assistant_bms_ble_2025-05-29T19-14-35.711Z.log)
it's taken while discharging with 6A!

patman15, 2025-05-30:
@shaf can you please confirm/test whether the current is also off by a factor of 10 for your battery (`ECO-WORTHY 0B_5AD4`) that I used to implement this protocol in #286? Please discharge the battery with a known current and provide a debug log plus screenshot if you are unsure. Thanks!

patman15, 2025-06-01:
@Cnecne59, @shaf [this branch](https://github.com/patman15/BMS_BLE-HA/tree/326-dchouse-fix-current-readout) should solve the wrong current readout. Can someone confirm?

shaf, 2025-06-01:
> [@Cnecne59](https://github.com/Cnecne59), [@shaf](https://github.com/shaf) [this branch](https://github.com/patman15/BMS_BLE-HA/tree/326-dchouse-fix-current-readout) should solve the wrong current readout. Can someone confirm?

I'm remote at the moment but have applied the branch for testing

I can't see a change ? Both the current and wattage look off but then again my battery is floating off the solar at the moment so I will only be able to tell after the sun goes down

[image]

patman15, 2025-06-01:
Btw. this is the change I did: https://github.com/patman15/BMS_BLE-HA/commit/af2bd1c77ccd802fc6a0c7acd7ff29d5a1d14ff9

Cnecne59, 2025-06-01:
@shaf Yes, i have restarted my esp32 Proxy and yes, the battery is not visible to the dchouse app. Even not through the bluetooth app on my iphone. Now i try to discharge the bat until cutoff from bms and then charging. Perhaps i may help!

shaf, 2025-06-02:
Hi @patman15 

So the levels look in line with the victron smart solar stats IE the magnitude seems to now be correct

However one thing I noticed before is that there no negative current registered

Thanks
Shaf

[image]

patman15, 2025-06-02:
Can you send me a debug log of a timespan where you would expect negative current? Actually I have a test to check if negative currents work which passes, so I can't see an obvious reason. Maybe the negative current was too low to be reported by the battery?

shaf, 2025-06-02:
> Can you send me a debug log of a timespan where you would expect negative current? Actually, I have a test to check if negative currents work, which passes, so I can't see an obvious reason. Maybe the negative current was too low to be reported by the battery?

OK I did some experimentation

I disconnected the PV input so that the battery is discharging

On normal idle I use between 0.05 and 0.2A but this doesn't register even in the eco worthy app showing 0A

[image]

[image]

Next test a higher discharge rate by turning some lights on

[image]

[image]

So that looks promising , you're right my nominal background current draw is too low that the eco worthy BMS doesn't register it

patman15, 2025-06-02:
@shaf Thanks a lot for testing and the feedback, I will merge soon for the next beta release and close the ticket.

closed 2025-06-02 by patman15