patman15/BMS_BLE-HA #328: Weize Battery MFG-DGJBD Model-DP04S007L4S120A

msva17, 2025-05-22:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

This battery does not get discovered and I believe it should be based on the supported BMS types.  Here is the copy paste of the 2 batteries that I feel should be discovered.

{"name":"Camper Right Battery","address":"A5:C2:37:3F:9F:E8","rssi":-74,"manufacturer_data":{"40936":"3f37c2a5"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"CC:DB:A7:3E:06:86","connectable":true,"time":1747926174.5557504,"tx_power":null}

{"name":"Camper Left Battery","address":"A5:C2:37:3F:AD:00","rssi":-75,"manufacturer_data":{"44288":"3f37c2a5"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"CC:DB:A7:3E:06:86","connectable":true,"time":1747926170.569716,"tx_power":null}

### Additional context

The IOS app is XiaoXiangElectric

patman15, 2025-05-22:
Since it looks like this battery uses a JBD BMS, please use the current version from [main](https://github.com/patman15/BMS_BLE-HA) via [manual install](https://github.com/patman15/BMS_BLE-HA#installation) and rename your BMSs to `JBD-` and then whatever you want.

msva17, 2025-05-22:
>   patman15  left a comment   (patman15/BMS_BLE-HA#328) (https://github.com/patman15/BMS_BLE-HA/issues/328#issuecomment-2901651864)

Awesome I’ll give that a go thanks much

closed 2025-05-28 by msva17