patman15/BMS_BLE-HA #329: Vatrer 460Ah LiFePo battery with JBD 250A BMS

Sciencefreak, 2025-05-22:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

Looks like naming scheme of Vatrer is not included yet

{"name":"DWCK00309J-029","address":"A4:C1:37:34:BF:FC","rssi":-46,"manufacturer_data":{"49148":"3437c1a4"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"E4:5F:01:23:3D:05","connectable":true,"time":1747946269.7573605,"tx_power":null}

patman15, 2025-05-24:
@Sciencefreak can you  please do a [manual install](https://github.com/patman15/BMS_BLE-HA#installation) using [this branch](https://github.com/patman15/BMS_BLE-HA/tree/329-vatrer-460ah-lifepo-battery-with-jbd-250a-bms). A debug log would be nice to verify that everything is ok. Let me know if it works for you!

Sciencefreak, 2025-05-24:
@patman15 thx for the quick update. Connection took quite long, but everything seems to work now. Values also agree very well with the ones in the smartphone app beside the fact, that app shows charge state in Ah instead of kWh (I assume it is just a transformation done by the app, I prefer the kWh given by your integration)

Debug file also looks fine with respect to your integration (still struggling with integration of victron BLE stuff)

[home-assistant_bms_ble_2025-05-24T18-14-59.953Z.log](https://github.com/user-attachments/files/20427599/home-assistant_bms_ble_2025-05-24T18-14-59.953Z.log)

closed 2025-05-29 by patman15