patman15/BMS_BLE-HA #331: JDB AP21S001-v11GB Version 4.4 not detected

robbesutr, 2025-05-23:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

 hi all,

I use the HA ble proxy on a esp32. I have only 1 other device connected to the ble proxy. which is the JDB DP24S002 v1.0 this one works perfect with BLE battery management. But the AP21S001-v11GB isn't detected. I made sure that no mobile phone or other device had ble on. RSSI of the BMS is -56 (esp is very close) many thanks for your advice. 

details from the adv monitor:
------------------------------------------------------
Adres: A4:C1:37:11:8C:B4
Naam: P09924280150
Bron: A8:42:E3:AE:A9:2A

**Advertentie data**

**Fabrikant data**
36020	31 31 33 37 63 31 61 34

**Service data**

**Service UUID's**
0000ff00-0000-1000-8000-00805f9b34fb
------------------------------------------------------

patman15, 2025-05-24:
@robbesutr for JBD unfortunately the name must match the patterns listed in the README. So you can either rename your device ` AP21S001...` or use the current version from [main](https://github.com/patman15/BMS_BLE-HA) via [manual install](https://github.com/patman15/BMS_BLE-HA#installation) and rename your BMSs to `JBD-` and then whatever you want. Let me know if that works for you.

ropper01, 2025-05-25:
renaming to JBD-xxx worked  ! many thanks Patrick