patman15/BMS_BLE-HA #337: Epoch GC2 Detected, but unable to add

oldjohngalt, 2025-05-27:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

I have an two [Epoch GC](https://www.epochbatteries.com/products/48v-60ah-gc2-golf-cart-lifepo4-lithium-battery) batteries in my golf cart and a bluetooth BMS adapter.  I have a BLE Proxy device located close to the golf cart and the integration detects the EpochGC2, but is unable to add it.  The debug log is below.

Would a capture of the iOS traffic be helpful?

`2025-05-26 23:07:24.143 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JTGER9NQNXK2580G9QBR667Q version=1 domain=bms_ble title=EpochGC2-0184
 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=30:55:44:38:C1:6C>
2025-05-26 23:07:24.144 DEBUG (MainThread) [custom_components.bms_ble.ective_bms::EpochGC2-0184:C16C)] initializing Ective Smart BMS, BT address: 30:55:44:38:C1:6C
2025-05-26 23:07:24.144 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for EpochGC2-0184 (30:55:44:38:C1:6C) as Ective Smart BMS
2025-05-26 23:07:24.144 DEBUG (MainThread) [custom_components.bms_ble] EpochGC2-0184: advertisement: {'name': 'EpochGC2-0184', 'address': '30:55:44:38:C1:6C', 'rssi': -83, 'manufacturer_data': {65535: b'0UD8\xc1l'}, 'service_data': {'00003333-0000-1000-8000-00805f9b34fb': b'456'}, 'service_uuids': ['0000ffe0-0000-1000-8000-00805f9b34fb'], 'source': 'A0:A3:B3:79:4E:76', 'advertisement': AdvertisementData(local_name='EpochGC2-0184', manufacturer_data={65535: b'0UD8\xc1l'}, service_data={'00003333-0000-1000-8000-00805f9b34fb': b'456'}, service_uuids=['0000ffe0-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-83), 'device': BLEDevice(30:55:44:38:C1:6C, EpochGC2-0184), 'connectable': True, 'time': 1927299.503065396, 'tx_power': None, 'raw': b'\x02\x01\x05\x03\x02\xe0\xff\t\xff\xff\xff0UD8\xc1l\x0e\tEpochGC2-0184\x06\x1633456'}
2025-05-26 23:07:24.144 DEBUG (MainThread) [custom_components.bms_ble] EpochGC2-0184: BMS data update
2025-05-26 23:07:24.144 DEBUG (MainThread) [custom_components.bms_ble.ective_bms::EpochGC2-0184:C16C)] connecting BMS
2025-05-26 23:07:24.144 DEBUG (MainThread) [bleak_retry_connector] 30:55:44:38:C1:6C - 30:55:44:38:C1:6C: Connection attempt: 1
2025-05-26 23:07:32.997 DEBUG (MainThread) [bleak_retry_connector] 30:55:44:38:C1:6C - 30:55:44:38:C1:6C: Failed to connect: Error ESP_GATT_CONN_TIMEOUT while connecting: Connection failed due to timeout, device_missing: False, backing off: 0.5 (attempt: 1, last rssi: None)
2025-05-26 23:07:33.499 DEBUG (MainThread) [bleak_retry_connector] 30:55:44:38:C1:6C - 30:55:44:38:C1:6C: Connection attempt: 2
2025-05-26 23:07:35.134 DEBUG (MainThread) [bleak_retry_connector] 30:55:44:38:C1:6C - 30:55:44:38:C1:6C: Connected after 2 attempts
2025-05-26 23:07:40.161 DEBUG (MainThread) [custom_components.bms_ble] EpochGC2-0184: BMS communication timed out, check signal strength (-83 dBm)
2025-05-26 23:07:40.161 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching EpochGC2-0184 data in 16.017 seconds (success: False)
2025-05-26 23:07:40.162 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS (EpochGC2-0184)
2025-05-26 23:07:40.162 DEBUG (MainThread) [custom_components.bms_ble.ective_bms::EpochGC2-0184:C16C)] disconnecting BMS
2025-05-26 23:07:40.227 DEBUG (MainThread) [custom_components.bms_ble.ective_bms::EpochGC2-0184:C16C)] disconnected from BMS
2025-05-26 23:07:40.227 DEBUG (MainThread) [custom_components.bms_ble.ective_bms::EpochGC2-0184:C16C)] disconnected from BMS
2025-05-26 23:07:45.275 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JTGER9NQNXK2580G9QBR667Q version=1 domain=bms_ble title=EpochGC2-0184
 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=30:55:44:38:C1:6C>
2025-05-26 23:07:45.275 DEBUG (MainThread) [custom_components.bms_ble.ective_bms::EpochGC2-0184:C16C)] initializing Ective Smart BMS, BT address: 30:55:44:38:C1:6C
2025-05-26 23:07:45.275 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for EpochGC2-0184 (30:55:44:38:C1:6C) as Ective Smart BMS
2025-05-26 23:07:45.275 DEBUG (MainThread) [custom_components.bms_ble] EpochGC2-0184: advertisement: {'name': 'EpochGC2-0184', 'address': '30:55:44:38:C1:6C', 'rssi': -82, 'manufacturer_data': {65535: b'0UD8\xc1l'}, 'service_data': {'00003333-0000-1000-8000-00805f9b34fb': b'456'}, 'service_uuids': ['0000ffe0-0000-1000-8000-00805f9b34fb'], 'source': 'A0:A3:B3:79:4E:76', 'advertisement': AdvertisementData(local_name='EpochGC2-0184', manufacturer_data={65535: b'0UD8\xc1l'}, service_data={'00003333-0000-1000-8000-00805f9b34fb': b'456'}, service_uuids=['0000ffe0-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-82), 'device': BLEDevice(30:55:44:38:C1:6C, EpochGC2-0184), 'connectable': True, 'time': 1927320.576192446, 'tx_power': None, 'raw': b'\x02\x01\x05\x03\x02\xe0\xff\t\xff\xff\xff0UD8\xc1l\x0e\tEpochGC2-0184\x06\x1633456'}
2025-05-26 23:07:45.275 DEBUG (MainThread) [custom_components.bms_ble] EpochGC2-0184: BMS data update
2025-05-26 23:07:45.275 DEBUG (MainThread) [custom_components.bms_ble.ective_bms::EpochGC2-0184:C16C)] connecting BMS
2025-05-26 23:07:45.275 DEBUG (MainThread) [bleak_retry_connector] 30:55:44:38:C1:6C - 30:55:44:38:C1:6C: Connection attempt: 1`

### Additional context

Here is a [link](https://www.epochbatteries.com/products/bluetooth-module-for-gc2-motive-batteries) to the Bluetooth module for the batteries.

patman15, 2025-05-27:
> Would a capture of the iOS traffic be helpful?

Definitely, but first before you take that effort, I would try to improve your `RSSI` value. It's pretty low and there are a number of errors that the connection failed. If that does not help, I need the traffic log to check the protocol.

patman15, 2025-05-29:
Unfortunately, this is again a new protocol that is not yet implemented. It seems that it is not too complicated but still some work. I'd like to finish 3 others first and would then come back if you are still interested?

jkeeney-apreohealth, 2025-07-09:
@patman15 - Thanks!  I removed/deleted the old integration and I think I followed all of the steps correctly.  However, when I try to add the integration I get an error that says "No supported devices found via Bluetooth."

Does this mean the integration can't find my Bluetooth Proxies or that a support BMS system hasn't been detected?  I am still traveling until Sunday so I won't be able to move a proxy closer to the golf cart until then.  Then again maybe somebody stole my golf cart and that is why it is out of range.

patman15, 2025-07-09:
Regarding range, you can check the [advertisement monitor](https://my.home-assistant.io/redirect/bluetooth_advertisement_monitor/) and check the `RSSI` value. In addition, I had a small issue in the detection. Please update the code and try again.