patman15/BMS_BLE-HA #342: Sensors unavailable after time

BuurmanG, 2025-05-29:
### Checklist

- [x] I have tried the latest version of BMS_BLE-HA.
- [x] I have enabled debug logging for my installation.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [x] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

PowerQueen (Redodo)

### Describe the issue

De integration stops updating. Sensors become unavailable. Reloading the integration does not work. After restarting home assistant the integration works fine.

### Reproduction steps

```text
1.
2.
3.
...
```

### Debug logs

Do not have logs

patman15, 2025-05-29:
Without a debug log I cannot solve this issue. Please follow [these steps](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#in-case-you-have-troubles-youd-like-to-have-help-with) and wait till the problem occurs. In case it is already there, just skip the restart of HA and create a log of ~2 minutes.

BuurmanG, 2025-05-29:
Meanwhile I have improved the bluetooth signalstrength, downloaded the latest release 1.18.0-beta 1, started debug logging. Now awaiting for results...

sircambridge, 2025-05-29:
I am using a UGREEN USB bluetooth dongle, and i am running home assistant in a docker container on my synology NAS DS220+ (intel celeron). its shared with the container with dbus.

maybe the issue is with the blueZ running on my synology, however all my other bluetooth devices are working fine with HA.

even after the disconnect, I can see the BMS show up in the bluetooth advertisements in the bluetooth integration has having a good signal RSSI -52.

when it gets into this stuck state, what can I do to help get more diagnostic information? should I look in the bluetoothd logs - i need to figure out where they are stored on my synology

sircambridge, 2025-05-29:
could this be relevant? 

`2025-05-29 22:41:04.560 DEBUG (MainThread) [custom_components.bms_ble] ECO-LFP48100-3U-8EC7: BMS data update
2025-05-29 22:41:04.560 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECO-LFP48100-3U-8EC7:8EC7)] connecting BMS
2025-05-29 22:41:06.272 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECO-LFP48100-3U-8EC7:8EC7)] RX BLE data (start): bytearray(b'\x01Q\x00\x00\xff\xff\x00.\x14\xc3\xff\xd1\x00P\x00d\x0b\xdf\x0c\xfc')
2025-05-29 22:41:06.272 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECO-LFP48100-3U-8EC7:8EC7)] incorrect frame end (length: 20).
2025-05-29 22:41:06.272 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECO-LFP48100-3U-8EC7:8EC7)] RX BLE data (cnt.): bytearray(b'\x0c\xf9\x0b\xe1\x0b\xd0\x00\x03\x03\x18\x03\xe8\x00\x01\x00\x01\x00\x17\x00\x00')
2025-05-29 22:41:06.272 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECO-LFP48100-3U-8EC7:8EC7)] incorrect frame end (length: 40).
2025-05-29 22:41:06.273 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECO-LFP48100-3U-8EC7:8EC7)] RX BLE data (cnt.): bytearray(b'\x00\x00\x01\x10\x01\x06\x01\x03\x01\x02\x00\x00\x00\x004\x9d')
2025-05-29 22:41:06.274 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECO-LFP48100-3U-8EC7:8EC7)] incorrect frame end (length: 56).
2025-05-29 22:41:06.722 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::ECO-LFP48100-3U-8EC7:8EC7)] disconnected from BMS
2025-05-29 22:41:06.722 INFO (MainThread) [custom_components.bms_ble.jbd_bms::ECO-LFP48100-3U-8EC7:8EC7)] failed to initialize BMS connection (BleakDBusError)
2025-05-29 22:41:06.723 DEBUG (MainThread) [custom_components.bms_ble] ECO-LFP48100-3U-8EC7: BMS communication failed, check signal strength (-79 dBm): [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Unlikely Error) (BleakDBusError)
2025-05-29 22:41:06.723 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching ECO-LFP48100-3U-8EC7 data in 2.163 seconds (success: False)
2025-05-29 22:41:36.560 DEBUG (MainThread) [custom_components.bms_ble] ECO-LFP48100-3U-8EC7: BMS data update`

BuurmanG, 2025-05-30:
Update on Bluetooth Signal Quality and Battery Level Impact

I’ve observed a significant degradation in Bluetooth signal quality when the battery level drops below 10%. Specifically, the signal strength can fall to approximately -90 dBm, which is considered too low for stable communication and can result in intermittent connectivity or complete disconnection.

After recharging the battery and reloading both the Bluetooth and BMS integrations, the signal stabilizes and returns to acceptable levels. This behavior suggests a correlation between low battery voltage and reduced Bluetooth module performance, possibly due to power-saving mechanisms or insufficient voltage supply to the communication module.

patman15, 2025-05-30:
For both of you, in case you need my help still (for the moment it does not sound like a bug in the integration),

- use version 1.18.0-beta.1 (or higher in future)
- enable `RSSI` and `link_quality` sensor to get an overview of 24hrs
- provide any observation together with a log showing the minute before the first time `Finished fetching <BMS> data in x seconds (success: False)` (with success: false) appears and the minute thereafter.

The integration should automatically recover from any situation where
- the Bluetooth module of HA is not stuck or degraded, e.g. Raspberry PI, Realtek chipsets, bad connection to proxy
- the BMS is not in a rogue state

i.e. reloading the integration should never make a difference (given that you wait for the recovery to kick in, worst case ~1h).

BuurmanG, 2025-05-31:
[image]
See result of switching battery off at 10% capacity: the link quality remains intact, and so does the interface with the integration, without interruptions.

closed 2025-05-31 by BuurmanG

mareklab, 2025-08-06:
I've noticed the same with 100 Balance MBS by Dale. After a few hours, sensors become unavailable, and it starts working again after a full restart of HA. Batteries loaded to 100%.