patman15/BMS_BLE-HA #347: ECO-WORTHY 12V 150AH Bluetooth, Built-in 120A BMS

SubBass100, 2025-05-31:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue%20label%3A%22enhancement%22).

### Please describe the enhancement you would like to have.

I have this battery https://www.amazon.com/dp/B0DBLP99QJ?ref_=ppx_hzsearch_conn_dt_b_fed_asin_title_5&th=1 I am also running esphome with ESP32-c3 UltraMini (works with other BT devices though I have one dedicated to only this battery). Sorry I couldn't find much more info though if you show me what to look for, I can help get any other information you need from this battery. Thanks!

patman15, 2025-05-31:
Can you please at least provide the advertisement information, as described in the [README](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#if-your-device-is-not-recognized)?

SubBass100, 2025-05-31:
No problem, this is the device advertisement info:
{"name":"ECO22E2","address":"85:7C:AE:31:22:E2","rssi":-84,"manufacturer_data":{"31877":"ae3122e2"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb","00000001-0000-1000-8000-00805f9b34fb"],"source":"7C:2C:67:D2:F6:7A","connectable":true,"time":1748699100.765188,"tx_power":null}

patman15, 2025-06-01:
Hi @SubBass100, I just added a note to the README:
> ECO-WORTHY batteries that show up as ECOxxxx use classic Bluetooth and do not support Bluetooth Low Energy (BLE). Thus, they unfortunately cannot be integrated. 

So sorry, this one cannot be supported by this integration (and I think it's a general issue with Home Assistant).

closed 2025-06-01 by patman15

SubBass100, 2025-06-01:
Thank you for checking.