patman15/BMS_BLE-HA #354: Wattcycle BT battery is not recognized

Armalite-2, 2025-06-04:
Hello
I received my Wattcycle 100ah BT Mini battery today. The reason I bought it was because of its size and the ability to integrate it into my home automation system.

Unfortunately, the integration isn't working.
I've attached the pictures.

Maybe there's a solution.

It times out.

[image]
[image]

patman15, 2025-06-04:
Did you manually install the main branch?
Currently, released version of the integration do not support it. Will come with 1.18.0-beta.2

Armalite-2, 2025-06-04:
I installed the BMS_BLE-HA as usual via HACs. I already have two Redodo batteries running.

HA immediately detected the Wattcycle battery, but the connection repeatedly breaks with a timeout.

I didn't do anything else.

closed 2025-06-05 by patman15

Armalite-2, 2025-06-05:
Hello, first of all, thank you very much for your effort. I installed the beta version. The Wattcycle battery was immediately detected and integrated. However, it keeps going offline. Deactivating and activating it brings it back online. Is there a way to make it permanently online?

[image]
[image]
[image]
[image]
[image]

patman15, 2025-06-05:
@Armalite-2 can you please enable `RSSI` and `link quality` sensor and provide a trace of that? Debug log would help as well. I suppose that you are struggling with the 3 device limit of the proxy? Or do you have multiple proxies?

Armalite-2, 2025-06-05:
Hello

The battery is located 1m from an Intel NUC.

https://github.com/user-attachments/assets/4413e0e4-b66c-4483-9dc3-4d2e05400e7a
[image]
[image]

patman15, 2025-06-05:
As mentioned, I need the items listed in the README. I believe you the issues, but the screenshots don't give me enough information to help you further.

patman15, 2025-06-05:
Ok, thanks. I will have a look later. There seems to be a protocol issue. Just not sure why the tests don't catch it, so I need to manually verify the sequence.
Debug log is helpful 👍