patman15/BMS_BLE-HA #358: FLYBAT LifePO4 nor properly detected

dweber69, 2025-06-06:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue%20label%3A%22enhancement%22).

### Please describe the enhancement you would like to have.

My FLYBAT LifePO4 batteries adverties themselves as "libatt..." and I hoped they would properly work with the existing E&J Technology BMS but unfortunately this is not the case.

Here the advertisement data from HomeAssistant

`{"name":"libattU12E9","address":"84:28:08:9D:12:E9","rssi":-62,"manufacturer_data":{"21320":"8428089d12e96400"},"service_data":{},"service_uuids":["00001800-0000-1000-8000-00805f9b34fb","00001801-0000-1000-8000-00805f9b34fb","6e400001-b5a3-f393-e0a9-e50e24dcca9e"],"source":"2C:CF:67:07:92:31","connectable":true,"time":1749210859.1335516,"tx_power":null}`

`{"name":"libattU14DB","address":"84:28:08:9D:14:DB","rssi":-62,"manufacturer_data":{"21320":"8428089d14db6400"},"service_data":{},"service_uuids":["00001800-0000-1000-8000-00805f9b34fb","00001801-0000-1000-8000-00805f9b34fb","6e400001-b5a3-f393-e0a9-e50e24dcca9e"],"source":"2C:CF:67:07:92:31","connectable":true,"time":1749210873.1502213,"tx_power":null}`

I have also attached the debug log

[home-assistant_bms_ble_2025-06-06T11-44-15.720Z.log](https://github.com/user-attachments/files/20627736/home-assistant_bms_ble_2025-06-06T11-44-15.720Z.log)

### Additional context

Unfortunately I have no protocol information. The company did not want to provide it.

patman15, 2025-06-06:
Hi @dweber69, I checked your log, the battery is correctly recognized, but you it constantly disappears from Bluetooth. Since your `RSSI` value looks ok, and the integration reports an CRC error, I guess you have an issue with the BT setup. Please see the [note about the internal RPi adapter](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#troubleshooting) that I think you are using.

patman15, 2025-06-09:
Hmm, interesting, seems that your BMS does not calculate any checksum, it's always 0xFB, independent of the data. At least in your last log.

dweber69, 2025-06-16:
I tried to use the iOS nrfConnect app to connect to the BMS and extract a few data exchanges but it does not work at all. I have enabled notifications on the UUID which indicated „Notify“ but I never get any data when sending the query command. Strange that it worked in some cases with you plugin bug not with nrfConnect. The BMS native app on the same phone always works flawlessly so its not the ble adapter or rssi. Strange. Must be something simple stupid I am overlooking. Any hint maybe?

patman15, 2025-06-29:
@dweber69 could please you test the [linked branch](https://github.com/patman15/BMS_BLE-HA/tree/358-flybat-lifepo4-nor-properly-detected) in this ticket via [manual install](https://github.com/patman15/BMS_BLE-HA#installation)?

dweber69, 2025-06-30:
I can do that in about 2 weeks. The vehicle is currently on the road with someone else.

dweber69, 2025-07-18:
@patman15 unfortunately it didn't fully work. Seems as if the CRC check is disabled for one message but not the other and therefore the initialization failed. The debug log is attached.

[home-assistant_bms_ble_2025-07-18T08-06-53.800Z.log](https://github.com/user-attachments/files/21312868/home-assistant_bms_ble_2025-07-18T08-06-53.800Z.log)

patman15, 2025-07-21:
@dweber69 ok thanks for the feedback, then I will merge the current state for the next release.