patman15/BMS_BLE-HA #360: LiTime 51.2V 30A Golf Batteries failing setup

snarfattack, 2025-06-09:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I have thought about [providing a debug log](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) and/or [diagnostic data](https://www.home-assistant.io/docs/configuration/troubleshooting/#download-diagnostics).

### Describe the issue

Installed 1.18.0-beta.3, batteries are detected and I've added them, but I'm getting the following error now:

Failed setup, will retry: BMS communication failed: Characteristic 6e400003-b5a3-f393-e0a9-e50e24dcca9e was not found! (BleakCharacteristicNotFoundError)

patman15, 2025-06-09:
Unfortunately, this sounds like LiTime uses a different BMS for the larger batteries.
Can you  please see if you are able to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery?
Heads-up: I'm currently off my debug/development system, so updates will take a while.
Best regards,
Patrick

snarfattack, 2025-06-09:
Sent an invite to the repo with the captured traffic between one of the batteries and the app.

patman15, 2025-06-11:
@snarfattack I downloaded your capture. I will have a look when I return.

patman15, 2025-07-04:
@snarfattack I had a look at the recorded traffic, but it does not really contain use-able information. Can you try again a bit longer and click some things in the app? Also the information from the [service explorer script](https://github.com/hbldh/bleak/blob/develop/examples/service_explorer.py) could give me more hints.

snarfattack, 2025-07-10:
Just uploaded a new capture, I think this one actually captured what is needed. You should see it connect to 3 different batteries.

patman15, 2025-07-10:
@snarfattack thanks for the capture! It enabled me to identify the BMS. Can you please try the [branch linked to this ticket](https://github.com/patman15/BMS_BLE-HA/tree/360-litime-512v-30a-golf-batteries-failing-setup) via [manual install](https://github.com/patman15/BMS_BLE-HA/tree/360-litime-512v-30a-golf-batteries-failing-setup?tab=readme-ov-file#installation)?

closed 2025-07-12 by patman15