patman15/BMS_BLE-HA #361: Another JBD pattern

robwolff3, 2025-06-09:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue%20label%3Aenhancement%20OR%20label%3A%22new%20device%22).
- [x] I have checked out the latest <a href="https://github.com/patman15/BMS_BLE-HA/releases">[image]</a>

### Please describe the enhancement you would like to have.

My JBD BMS doesnt seem to be detected. Another one to add to the integration?

{"name":"xiaoxiang BMS","address":"A4:C1:37:20:2C:31","rssi":-43,"manufacturer_data":{"11313":"2037c1a4"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"2C:CF:67:AF:72:66","connectable":true,"time":1749505718.6813028,"tx_power":null}

Let me know if the explorer script or the bluetooth traffic is needed.

patman15, 2025-06-10:
Hi! Thanks, I'll add it to the next release. Seems there are uncountable patterns for this BMS but no useful manufacturer ID. :-\

patman15, 2025-06-29:
@robwolff3 can you check the [linked branch](https://github.com/patman15/BMS_BLE-HA/tree/361-another-jbd-pattern)?

closed 2025-07-02 by patman15