patman15/BMS_BLE-HA #362: Strange behavior of the integration when reading from JBD BMS

4mr, 2025-06-10:
### Checklist

- [x] I have tried the latest version of BMS_BLE-HA.
- [x] I have enabled debug logging for my installation.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [x] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

JBD BMS, model DP04S007 (4 cell model)

### Describe the issue

I noticed some unusual behavior: approximately once per hour, the same BLE request to read data from the BMS gets sent twice in a row:
> TX BLE data #1 (WNR): dd a5 03 00 ff fd 77 
> TX BLE data #2 (WNR): dd a5 03 00 ff fd 77

In other words, the same packet is transmitted twice almost simultaneously. As a result, the BMS sends back multiple responses, which seems to confuse the parser. This causes incorrect data to be displayed — for example, cells #5 to #16 suddenly appear, even though they shouldn't.

I've attached logs showing both the incorrect data and an example of normal behavior.

### Reproduction steps

```text
1. Can't reproduce
```

### Debug logs

Incorrect data

> 2025-06-10 07:16:41.340 DEBUG (MainThread) [custom_components.bms_ble] DP04S007L4S150A: BMS data update
> 2025-06-10 07:16:41.340 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] BMS already connected
> 2025-06-10 07:16:41.341 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] TX BLE data #1 (WNR): dd a5 03 00 ff fd 77
> 2025-06-10 07:16:41.693 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (start): bytearray(b'\xdd\x03\x00"\x057\x00\x00\xb5\x94\xf5P\x00\x011N\x00\x00\x00\x00')
> 2025-06-10 07:16:41.936 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (cnt.): bytearray(b'\x00\x00f]\x03\x04\x01\x0bq\x00\x00\x00\xc3P\xb5\x94\x00\x00\xf8\xf1')
> 2025-06-10 07:16:42.060 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] TX BLE data #2 (WNR): dd a5 03 00 ff fd 77
> 2025-06-10 07:16:42.180 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (cnt.): bytearray(b'w')
> 2025-06-10 07:16:42.181 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] TX BLE data #1 (WNR): dd a5 04 00 ff fc 77
> 2025-06-10 07:16:42.229 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (start): bytearray(b'\xdd\x03\x00"\x057\x00\x00\xb5\x94\xf5P\x00\x011N\x00\x00\x00\x00')
> 2025-06-10 07:16:42.278 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (cnt.): bytearray(b'\x00\x00f]\x03\x04\x01\x0bq\x00\x00\x00\xc3P\xb5\x94\x00\x00\xf8\xf1')
> 2025-06-10 07:16:42.280 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (cnt.): bytearray(b'w')
> 2025-06-10 07:16:42.282 DEBUG (MainThread) [custom_components.bms_ble] DP04S007L4S150A: BMS data sample {'temp_sensors': 1, 'voltage': 13.35, 'current': 0.0, 'battery_level': 93, 'cycle_charge': 464.84, 'cycles': 1, 'problem_code': 0, 'temp#0': 19.8, 'cell#0': 1.335, 'cell#1': 0.0, 'cell#2': 46.484, 'cell#3': 62.8, 'cell#4': 0.001, 'cell#5': 12.622, 'cell#6': 0.0, 'cell#7': 0.0, 'cell#8': 0.0, 'cell#9':26.205, 'cell#10': 0.772, 'cell#11': 0.267, 'cell#12': 28.928, 'cell#13': 0.0, 'cell#14': 50.0, 'cell#15': 46.484, 'cell#16': 0.0, 'delta_voltage': 62.8, 'cycle_capacity': 6205.614, 'power': 0.0, 'battery_charging': False, 'temperature': 19.8, 'problem': True}
> 2025-06-10 07:16:42.282 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching DP04S007L4S150A data in 0.942 seconds (success: True)
> 2025-06-10 07:16:42.376 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (start): bytearray(b'\xdd\x04\x00\x08\r\n\r\n\r\n\r\x0b\xff\x9bw')

Normal behavior one before and one after incorrect data:
> 2025-06-10 07:16:11.340 DEBUG (MainThread) [custom_components.bms_ble] DP04S007L4S150A: BMS data update
> 2025-06-10 07:16:11.341 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] BMS already connected
> 2025-06-10 07:16:11.341 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] TX BLE data #1 (WNR): dd a5 03 00 ff fd 77
> 2025-06-10 07:16:11.493 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (start): bytearray(b'\xdd\x03\x00"\x057\x00\x00\xb5\x94\xf5P\x00\x011N\x00\x00\x00\x00')
> 2025-06-10 07:16:11.494 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (cnt.): bytearray(b'\x00\x00f]\x03\x04\x01\x0bq\x00\x00\x00\xc3P\xb5\x94\x00\x00\xf8\xf1')
> 2025-06-10 07:16:11.495 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (cnt.): bytearray(b'w')
> 2025-06-10 07:16:11.496 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] TX BLE data #1 (WNR): dd a5 04 00 ff fc 77
> 2025-06-10 07:16:11.591 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (start): bytearray(b'\xdd\x04\x00\x08\r\n\r\n\r\n\r\x0b\xff\x9bw')
> 2025-06-10 07:16:11.592 DEBUG (MainThread) [custom_components.bms_ble] DP04S007L4S150A: BMS data sample {'temp_sensors': 1, 'voltage': 13.35, 'current': 0.0, 'battery_level': 93, 'cycle_charge': 464.84, 'cycles': 1, 'problem_code': 0, 'temp#0': 19.8, 'cell#0': 3.338, 'cell#1': 3.338, 'cell#2': 3.338, 'cell#3': 3.339, 'delta_voltage': 0.001, 'cycle_capacity': 6205.614, 'power': 0.0, 'battery_charging': False, 'temperature': 19.8, 'problem': False}
> 2025-06-10 07:16:11.592 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching DP04S007L4S150A data in 0.252 seconds (success: True)

> 2025-06-10 07:17:12.345 DEBUG (MainThread) [custom_components.bms_ble] DP04S007L4S150A: BMS data update
> 2025-06-10 07:17:12.347 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] BMS already connected
> 2025-06-10 07:17:12.348 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] TX BLE data #1 (WNR): dd a5 03 00 ff fd 77
> 2025-06-10 07:17:12.620 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (start): bytearray(b'\xdd\x03\x00"\x057\x00\x00\xb5\x94\xf5P\x00\x011N\x00\x00\x00\x00')
> 2025-06-10 07:17:12.622 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (cnt.): bytearray(b'\x00\x00f]\x03\x04\x01\x0bq\x00\x00\x00\xc3P\xb5\x94\x00\x00\xf8\xf1')
> 2025-06-10 07:17:12.626 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (cnt.): bytearray(b'w')
> 2025-06-10 07:17:12.627 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] TX BLE data #1 (WNR): dd a5 04 00 ff fc 77
> 2025-06-10 07:17:12.718 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms::DP04S007L4S150A:A613)] RX BLE data (start): bytearray(b'\xdd\x04\x00\x08\r\n\r\n\r\n\r\x0b\xff\x9bw')
> 2025-06-10 07:17:12.719 DEBUG (MainThread) [custom_components.bms_ble] DP04S007L4S150A: BMS data sample {'temp_sensors': 1, 'voltage': 13.35, 'current': 0.0, 'battery_level': 93, 'cycle_charge': 464.84, 'cycles': 1, 'problem_code': 0, 'temp#0': 19.8, 'cell#0': 3.338, 'cell#1': 3.338, 'cell#2': 3.338, 'cell#3': 3.339, 'delta_voltage': 0.001, 'cycle_capacity': 6205.614, 'power': 0.0, 'battery_charging': False, 'temperature': 19.8, 'problem': False}
> 2025-06-10 07:17:12.720 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching DP04S007L4S150A data in 0.374 seconds (success: True)

4mr, 2025-06-10:
After analyzing the code in the file [basebms.py](https://github.com/patman15/BMS_BLE-HA/blob/main/custom_components/bms_ble/plugins/basebms.py#L386), I noticed an error: on the first attempt, the variable attempt will be 0, and after calculations, the timeout always ends up being BLEAK_TRANSIENT_BACKOFF_TIME (0.25 seconds), which is clearly too short for a response.

Look into the code:
```python
for attempt in range(BaseBMS.MAX_RETRY):
    self._data_event.clear()  # clear event before requesting new data
    await self._send_msg(
        data, max_size, char or self.uuid_tx(), attempt, inv_wr_mode
    )
    try:
        if wait_for_notify:
            await asyncio.wait_for(
                self._wait_event(),
                BLEAK_TRANSIENT_BACKOFF_TIME
                * min(2**attempt, BaseBMS._MAX_TIMEOUT_FACTOR),
            )
```

patman15, 2025-06-10:
Well I did this on purpose, also the timeout seems to be working most of the time. The problem actually is that the duplicate answer is not ignored. I need to check that.
But yes, you can simply increase the timeout for a quick solution.

4mr, 2025-06-10:
> The problem actually is that the duplicate answer is not ignored. I need to check that.

I found the reason: after receiving the first final packet, `data = BMS._decode_data(self._data_final)` is executed, after which a second request is made (`self._await_reply(BMS._cmd(b"\x04"))`) to get the `_cell_voltages` values. However, there was another request for the battery data (because the timeout for attempt 0 is very short, only 0.25 seconds), and the plugin receives the full battery information instead of the cell values. So, in the `_cell_voltages` function, we need to check which response was received - 03 (battery status) or 04 (cell status). Apparently, we also need to check the data in the `_decode_data` function to ensure it's battery status data.

```python
async def _async_update(self) -> BMSsample:
    """Update battery status information."""
    data: BMSsample = {}
    await self._await_reply(BMS._cmd(b"\x03"))
    data = BMS._decode_data(self._data_final)
    data["temp_values"] = BMS._temp_sensors(
        self._data_final, int(data.get("temp_sensors", 0))
    )

    await self._await_reply(BMS._cmd(b"\x04"))
    data["cell_voltages"] = BMS._cell_voltages(self._data_final)
```

patman15, 2025-06-10:
When I'm back to my development system, I'll change the code to do all queries and build up an array of answers like for other BMSs.
The goal is not to have a long first timeout, but to repeat in case the BMS fails. This is due to a large number of supported BMS.
The TIMEOUT variable is used in some BMS, not for yours.
I'll send you a ping when I'm done.
Thanks for your analysis!

patman15, 2025-06-29:
@4mr could you please check the [linked branch](https://github.com/patman15/BMS_BLE-HA/tree/362-strange-behavior-of-the-integration-when-reading-from-jbd-bms) via [manual install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#installation)? I think it should fix the issue.