patman15/BMS_BLE-HA #365: Support for latest JK BMS v19, fw 19.05

Macrisu, 2025-06-11:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue%20label%3Aenhancement%20OR%20label%3A%22new%20device%22).
- [x] I have checked out the latest <a href="https://github.com/patman15/BMS_BLE-HA/releases">[image]</a>

### Please describe the enhancement you would like to have.

The latest JK BMS hw 19 does not work, fw 19.05
I've tried the stable and the newest beta.

Logs [here](https://github.com/patman15/BMS_BLE-HA/issues/90#issuecomment-2956909729)

patman15, 2025-06-11:
Thanks for the logs. As mentioned I'm pretty sure it would work with a performance BT adapter (~7€), but I think I can fix the mess the Raspberry PI internal adapter produces.
I'll give it a try when I'm back at my development system.
Best regards,
Patrick

Macrisu, 2025-06-14:
I had an Asus BT-500 USB and I've insertit into rPi 5.
I've disabled the internal BT and Using the Asus BT dongle I still have the same problems.

Log with 1.18.0-beta.3

[home-assistant_bms_ble_2025-06-14T20-02-07.142Z.log](https://github.com/user-attachments/files/20741069/home-assistant_bms_ble_2025-06-14T20-02-07.142Z.log)

patman15, 2025-06-15:
@Macrisu thanks for proofing me wrong! In that case they really changed the behaviour of the BMS in version 19. I'm pretty sure it was working before. Anyhow I'll fix it!
Thanks a lot for the logs I'll use them to avoid regressions and implement tests accordingly.

sag007, 2025-06-15:
Hey Guys,

I have just upgraded my batteries to use the new JK BMS ver19 and found the same issue. I am unable to add the bms's to the bluetooth proxy, they proxy doesn't even show the names that I setup in it.

patman15, 2025-06-16:
@sag007 I would say, you have a different problem. Most likely with your proxy. This issue is about the integration being stuck in the initialisation phase of the BMS.
Cannot say more since you did not provide any logs.

sag007, 2025-06-17:
Hey Patman15,

Yeah sorry for not including the logs, I wasn't sure how to obtain them, please see logs below. If you require more please let me know and Ill see how i go.

Thanks again.

Nathan

2025-06-16 19:48:40.013 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JXVZMC3QDGVCA7W4NWK0N81T version=1 domain=bms_ble title=JK_Batt_1-02 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=C8:47:80:28:9A:13>
2025-06-16 19:48:40.013 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JXVZPBKME8TP4KB5Y33Y4XTF version=1 domain=bms_ble title=JK_Batt_2-03 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=C8:47:80:28:99:F6>
2025-06-16 19:48:40.013 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JXVZR9Q4DBXAPWAEFP2KHRN6 version=1 domain=bms_ble title=JK_Batt_3-04 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=C8:47:80:28:BB:EC>
2025-06-16 19:48:40.051 DEBUG (MainThread) [custom_components.bms_ble.jikong_bms::JK_Batt_1-02:9A13)] initializing Jikong Smart BMS, BT address: C8:47:80:28:9A:13
2025-06-16 19:48:40.051 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for JK_Batt_1-02 (C8:47:80:28:9A:13) as Jikong Smart BMS
2025-06-16 19:48:40.051 DEBUG (MainThread) [custom_components.bms_ble] JK_Batt_1-02: advertisement: {'name': 'JK_Batt_1-02', 'address': 'C8:47:80:28:9A:13', 'rssi': -127, 'manufacturer_data': {2917: b'\x88\xa0\xc8G\x80(\x9a\x13'}, 'service_data': {}, 'service_uuids': ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000fee7-0000-1000-8000-00805f9b34fb', '0000ffe0-0000-1000-8000-00805f9b34fb', 'f000ffc0-0451-4000-b000-000000000000'], 'source': 'D8:3B:BF:4E:97:FA', 'advertisement': AdvertisementData(local_name='JK_Batt_1-02', manufacturer_data={2917: b'\x88\xa0\xc8G\x80(\x9a\x13'}, service_uuids=['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000fee7-0000-1000-8000-00805f9b34fb', '0000ffe0-0000-1000-8000-00805f9b34fb', 'f000ffc0-0451-4000-b000-000000000000'], rssi=-127), 'device': BLEDevice(C8:47:80:28:9A:13, JK_Batt_1-02), 'connectable': True, 'time': 4421997.745705816, 'tx_power': None, 'raw': None}
2025-06-16 19:48:40.051 DEBUG (MainThread) [custom_components.bms_ble] JK_Batt_1-02: BMS data update
2025-06-16 19:48:40.051 DEBUG (MainThread) [custom_components.bms_ble.jikong_bms::JK_Batt_1-02:9A13)] connecting BMS
2025-06-16 19:48:40.051 DEBUG (MainThread) [bleak_retry_connector] C8:47:80:28:9A:13 - C8:47:80:28:9A:13: Connection attempt: 1
2025-06-16 19:48:40.051 DEBUG (MainThread) [bleak_retry_connector] C8:47:80:28:9A:13 - C8:47:80:28:9A:13: Failed to connect: No backend with an available connection slot that can reach address C8:47:80:28:9A:13 was found, device_missing: False, backing off: 4.0 (attempt: 1, last rssi: None)
2025-06-16 19:48:40.051 DEBUG (MainThread) [bleak_retry_connector.bluez] JK_Batt_1-02 - C8:47:80:28:9A:13: Waited 1.4804303646087646e-05 seconds to disconnect
2025-06-16 19:48:40.052 DEBUG (MainThread) [custom_components.bms_ble.jikong_bms::JK_Batt_2-03:99F6)] initializing Jikong Smart BMS, BT address: C8:47:80:28:99:F6
2025-06-16 19:48:40.052 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for JK_Batt_2-03 (C8:47:80:28:99:F6) as Jikong Smart BMS
2025-06-16 19:48:40.052 DEBUG (MainThread) [custom_components.bms_ble] JK_Batt_2-03: advertisement: {'name': 'JK_Batt_2-03', 'address': 'C8:47:80:28:99:F6', 'rssi': -127, 'manufacturer_data': {2917: b'\x88\xa0\xc8G\x80(\x99\xf6'}, 'service_data': {}, 'service_uuids': ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000fee7-0000-1000-8000-00805f9b34fb', '0000ffe0-0000-1000-8000-00805f9b34fb', 'f000ffc0-0451-4000-b000-000000000000'], 'source': 'D8:3B:BF:4E:97:FA', 'advertisement': AdvertisementData(local_name='JK_Batt_2-03', manufacturer_data={2917: b'\x88\xa0\xc8G\x80(\x99\xf6'}, service_uuids=['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000fee7-0000-1000-8000-00805f9b34fb', '0000ffe0-0000-1000-8000-00805f9b34fb', 'f000ffc0-0451-4000-b000-000000000000'], rssi=-127), 'device': BLEDevice(C8:47:80:28:99:F6, JK_Batt_2-03), 'connectable': True, 'time': 4421997.745705816, 'tx_power': None, 'raw': None}
2025-06-16 19:48:40.052 DEBUG (MainThread) [custom_components.bms_ble] JK_Batt_2-03: BMS data update
2025-06-16 19:48:40.052 DEBUG (MainThread) [custom_components.bms_ble.jikong_bms::JK_Batt_2-03:99F6)] connecting BMS
2025-06-16 19:48:40.052 DEBUG (MainThread) [bleak_retry_connector] C8:47:80:28:99:F6 - C8:47:80:28:99:F6: Connection attempt: 1
2025-06-16 19:48:40.053 DEBUG (MainThread) [bleak_retry_connector] C8:47:80:28:99:F6 - C8:47:80:28:99:F6: Failed to connect: No backend with an available connection slot that can reach address C8:47:80:28:99:F6 was found, device_missing: False, backing off: 4.0 (attempt: 1, last rssi: None)
2025-06-16 19:48:40.053 DEBUG (MainThread) [bleak_retry_connector.bluez] JK_Batt_2-03 - C8:47:80:28:99:F6: Waited 1.2770295143127441e-05 seconds to disconnect
[... cut ...]
2025-06-16 19:49:16.069 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching JK_Batt_3-04 data in 36.016 seconds (success: False)
2025-06-16 19:49:16.069 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS (JK_Batt_3-04)
2025-06-16 19:49:16.168 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ (states('sensor.generated_power_total') | int + states('sensor.a_power') | int + states('sensor.b_power') | int + states('sensor.c_power') | int) - states('sensor.critical_loads') | int  }}) renders=2>
Traceback (most recent call last):

sag007, 2025-06-17:
As mentioned also, I have both the NUC internal Bluetooth adaptor and a Bluetooth proxy with slots on both for these devices.

sag007, 2025-06-17:
Looking at the Addon BLE error;

JK_Batt_1-02
No devices or entities
Failed setup, will retry: Could not find BMS (**C8:47:80:28:9A:13**) via Bluetooth.

It doesn't match the mac that the Bluetooth advertise shows that the BMS actually has **C8:47:80:3A:27:CD**.

Not sure if this helps.

sag007, 2025-06-17:
Is there a file I can edit to change the mac address to the one shown in the Bluetooth advertisement log?
Perhaps this might help things.

patman15, 2025-06-19:
@sag007 then you probably ran into the issue this topic was created for. Do you have a debug log to check?

sag007, 2025-06-19:
[home-assistant_bms_ble_2025-06-19T21-19-52.144Z.log](https://github.com/user-attachments/files/20825776/home-assistant_bms_ble_2025-06-19T21-19-52.144Z.log)

patman15, 2025-07-01:
@Macrisu thanks for your patience, I now made a fix in the [linked branch](https://github.com/patman15/BMS_BLE-HA/tree/365-support-jk-bms-fw-19-05). Can you verify it works for you, using [manual installation](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#installation)? (don't forget to reboot)
A debug log would be great for verification. Thanks!

Macrisu, 2025-07-02:
I've instaled the branch

Failed setup, will retry: BMS communication failed, check signal strength (-127 dBm): C8:47:80:3A:61:59 - C8:47:80:3A:61:59: Failed to connect after 9 attempt(s): No backend with an available connection slot that can reach address C8:47:80:3A:61:59 was found: The proxy/adapter is out of connection slots or the device is no longer reachable; Add additional proxies (https://esphome.github.io/bluetooth-proxies/) near this device (BleakOutOfConnectionSlotsError)

[home-assistant_bms_ble_2025-07-02T18-45-21.020Z.log](https://github.com/user-attachments/files/21023771/home-assistant_bms_ble_2025-07-02T18-45-21.020Z.log)

patman15, 2025-07-02:
Looks like your proxy is not connected. Bluetooth fails. This is not due to the integration. Maybe wait a bit?

closed 2025-07-02 by patman15