patman15/BMS_BLE-HA #366: Integration fails to use esphome bluetooth proxy adapter

a9e78rauijasdf, 2025-06-12:
### Checklist

- [x] I have tried the latest version of BMS_BLE-HA.
- [x] I have enabled debug logging for my installation.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [x] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

ANT BMS

### Describe the issue

Hi,

I am trying to have the integration detect my ANT BMS via a esphome bluetooth proxy (which supports active connections). The debug log from the integration logs devices from the "hci0" adapter which is far away from the BMS devices - which is why i have specified the "adapter" parameter for the BMS. Even though I have provided the correct name for the device it has to use, I can see from the logfile that it still uses HCI0.

I have been through the "bat mon" addon, which fails to list any BMS connected through a bluetooth proxy - I did read the issue(s) describing similar behavior, but I think I got confused by the references to comments in other issues - I did not see a clear solution there.

From this repository, I am unable to see that the ANT BMS is supported - but It was supported in the "batmon" addon - was support removed or not migrated to this repo?

Im sorry if I sound confused - but this is confusing to me - but I have at least come to the conclusion that ANT BMS does not seem to be supported with this integration - and I am wondering if there is something I have done wrong, or if there is a bug in this fine repository :-) I have read the guide on how to add support for a BMS; but that is out of my skillset Im afraid :(

### Reproduction steps

```text
1. Install BMS BLE integration
2. Turn on bluetooth on the ANT BMS
3. Verify ESPHome bluetooth proxy is available and listing the BMS
4. BMS should appear when opening the integration
```

### Debug logs

22:15:47 INFO [ant] bms1 normal connect failed (Device with address AA:BB:CC:91:00:01 was not found.), connecting with scanner
22:15:56 ERROR [sampling] AntBt(AA:BB:CC:91:00:01,bms1) device not found, retry in 7 seconds

(yes, that is the correct MAC - also confirmed with NRF connect app)

a9e78rauijasdf, 2025-06-13:
Hi Patman,

Thank you for your swift reply and of course for your work with the integration! I think I have been confused by your comment(s) - for example https://github.com/fl4p/batmon-ha/discussions/213#discussion-6419834 - but when re-reading it is pretty clear now, and I totally understand! :-)

I would absolutely like to supply the needed data for ANT BMS support - I have read the instructions for adding a new BMS type, but its just above my skillset to implement that - but I am pretty sure I can deliver the data needed from bluetoothctl output (or gatttool probably).

Would if be helpful to attach that kind of data here, or how can I help you in the best way?

TIA, and again thanks for the work!

patman15, 2025-06-13:
Hi! I would most likely have all the data I need if you are able to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery. The output of the advertisement monitor (copy to clipboard) will also help.
No rush, I cannot proceed at the moment, but will pick up.

a9e78rauijasdf, 2025-06-15:
I believe the initial problem would become the same problem as I started out with: It seems that the integration doesnt use the specified bluetooth adapter, and can only use hci0 - can you confirm that? I have looked through issues, and it seems that other people have issues with using bluetooth proxies. I have a spare Pi i can install rasbpian on to create the standalone-version of Batmon (the name has a cool ring to it! ;) - so I can avoid that issue, even though I would like to use a bluetooth proxy if I could.

I shall provide a connection log for you - which format/app do you prefer data from?

patman15, 2025-06-16:
@a9e78rauijasdf are you sure you understood that you opened an issue for  the integration `BMS_BLE-HA` and not the add-on `Batmon`?
I cannot help you with the latter you need to ask Fabian in his repository.
My integration does support all BT adapters Home Assistant supports, so also proxies, but not yet the ANT BMS.
Best regards,
Patrick

patman15, 2025-07-07:
@a9e78rauijasdf I could start implementing the ANT BMS for the integration. Let me know if you are still interested or if I should close the issue. Thanks!

patman15, 2025-07-08:
Ok, let's check the basics. Have you [installed](https://github.com/patman15/BMS_BLE-HA/#installation) the integrations? Can you please send me the [advertisement information ](https://my.home-assistant.io/redirect/bluetooth_advertisement_monitor/)of the device?

a9e78rauijasdf, 2025-07-09:
Hi again.

I have investigated the BMS - they are on bluetooth classic, and they expose a serial port connection over bluetooth - so I guess they are not a candidate for your integration...?

a9e78rauijasdf, 2025-07-11:
This is strange - they definitely do both, as I can see a serial port connection profile. But now they also appear as BLE devices! Here are some nice data:

Device: ANT-BLE24C-001 [AA:BB:CC:DD:EE:FF]

Service 0000ffe0-0000-1000-8000-00805f9b34fb (Vendor specific):
  Char 0000ffe2-0000-1000-8000-00805f9b34fb (Vendor specific) [write-without-response,write]
  Char 0000ffe1-0000-1000-8000-00805f9b34fb (Vendor specific) [write-without-response,write,notify]
    Descriptor 00002902-0000-1000-8000-00805f9b34fb (17)
      Value (hex): 0100
      Value (text): ^A^@
Service 00001801-0000-1000-8000-00805f9b34fb (Generic Attribute Profile):
  Char 00002a05-0000-1000-8000-00805f9b34fb (Service Changed) [indicate]
    Descriptor 00002902-0000-1000-8000-00805f9b34fb (13)
      Value (hex): 0200
      Value (text): ^B^@

a9e78rauijasdf, 2025-07-11:
I would really really like to work with you on this. Would it be more efficient to communicate through a different channel instead of cluttering this issue?

a9e78rauijasdf, 2025-07-11:
`{"name":"ANT-BLE24C-001","address":"AA:BB:CC:DD:EE:FF","rssi":-70,"manufacturer_data":{"8979":"88a0aabbcc900001"},"service_data":{},"service_uuids":["0000fee7ffe0-0000-1000-8000-00805f9b34fb"],"source":"35:BA:22:AF:A2:8F","connectable":true,"time":1752262308.3865466,"tx_power":null}`

a9e78rauijasdf, 2025-07-12:
I have the ANT BMS 24S 300A which is not explicitly mentioned on Syssis repo, but I imagine they are very similar. I have noticed the new branch, and I am familiar with the manual install. You are very welcome to reach out as soon as the raining starts and you feel like it - appreciate it!

patman15, 2025-07-19:
@a9e78rauijasdf can you please try the [branch linked to this issue](https://github.com/patman15/BMS_BLE-HA/tree/366-support-ant-bms) via [manual install](https://github.com/patman15/BMS_BLE-HA#installation)? I would need a debug log, because I still need some information to clean up the code.

closed 2025-08-21 by patman15