patman15/BMS_BLE-HA #372: Support for built-in BMS of ECO-LFP1215003 (ECO-WORTHY 12V 300Ah LiFePO4 battery)

JaronrH, 2025-06-27:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue%20label%3Aenhancement%20OR%20label%3A%22new%20device%22).
- [x] I have checked out the latest <a href="https://github.com/patman15/BMS_BLE-HA/releases">[image]</a>

### Please describe the enhancement you would like to have.

I have a ECO-LFP1215003 (ECO-WORTHY 12V 300Ah LiFePO4 battery).  The Readme says it may not be supported based on the service_uuids and name but I also have seam similar service uuids in other requests (issues) that are supported!  Furthermore, it's being found by nRF which, from my limited understanding, is only scanning BLE devices (unless they are broadcasting BLE then connecting with Classic for the data?).   I've included the advertisement data below to see if this is something you want to explore or if this is, indeed, an unsupported device! Thanks!

{"name":"ECO6D42","address":"A2:12:DE:74:6D:42","rssi":-71,"manufacturer_data":{"4770":"de746d42"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb","00000001-0000-1000-8000-00805f9b34fb"],"source":"30:AE:A4:0B:99:7A","connectable":true,"time":1751060707.1500342,"tx_power":null}

### Additional context

I'm using the latest release available in HACS.

JaronrH, 2025-06-28:
Unfortunately I don't have a rooted android device to do the sniffing with.  I'm going to close this and will reopen in the future if I can get a trace log and verify it isn't using Bluetooth Classic.  

Thanks for your help!

closed 2025-06-28 by JaronrH

patman15, 2025-06-29:
JFYI, @JaronrH there should not be a need for a rooted Android device. I also don't have one and could do the recording.

JaronrH, 2025-06-29:
At this point I already went with a Victron shunt for BMS. Universal battery support and documented advertisement specification! I have my existing Shelly device listening for btle and sending the advertisements over mqtt. A small c# service decrypts and parses out the data (posts it back to mqtt). HA can then read what I need from the mqtt server. 

Little hackish but works very well!