patman15/BMS_BLE-HA #373: Vatrer battery detecting 2 of the 4 batteries in stack

jthopwood, 2025-06-27:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I have thought about [providing a debug log](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) and/or [diagnostic data](https://www.home-assistant.io/docs/configuration/troubleshooting/#download-diagnostics).

### Describe the issue

I have 4 Vatrer batteries the signal is better than fair bit never good due to the design the lowest signal is-75 in the monitor. I can see all 4 batteries in the monitor but only 2 will connect. This may be an age thing as 2 batteries are 6 months older.  I'm also able to connect to all batteries with the Vatrer and the Over Kill Solar Apps. 

Va-0001 and Va-0002 will not connect while Va-0003 and Va-0004 work as expected. I can build a debug but don't think anything helpful is in ther as Va-0001 and Va-0002 are not mentioned in the log. 

Connection slot allocation is at 40% on the Bluetooth.

Battery details from monitor:

{"name":"Va-0001","address":"A4:C1:37:03:AA:08","rssi":-74,"manufacturer_data":{"43528":"0337c1a4"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"00:1A:7D:DA:71:09","connectable":true,"time":1751062597.8401532,"tx_power":null}

{"name":"Va-0002","address":"A4:C1:37:44:C0:75","rssi":-66,"manufacturer_data":{"49269":"4437c1a4"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"00:1A:7D:DA:71:09","connectable":true,"time":1751062597.4311616,"tx_power":null}

{"name":"Va-0003","address":"A4:C1:37:55:9C:FE","rssi":-44,"manufacturer_data":{"49572":"37559cfe"},"service_data":{},"service_uuids":["00001800-0000-1000-8000-00805f9b34fb","00001801-0000-1000-8000-00805f9b34fb","0000180a-0000-1000-8000-00805f9b34fb","0000fa00-0000-1000-8000-00805f9b34fb","0000ff00-0000-1000-8000-00805f9b34fb"],"source":"00:1A:7D:DA:71:09","connectable":true,"time":1751061291.6568425,"tx_power":null}

{"name":"Va-0004","address":"A4:C1:37:35:97:57","rssi":-61,"manufacturer_data":{"49572":"37359757"},"service_data":{},"service_uuids":["00001800-0000-1000-8000-00805f9b34fb","00001801-0000-1000-8000-00805f9b34fb","0000180a-0000-1000-8000-00805f9b34fb","0000fa00-0000-1000-8000-00805f9b34fb","0000ff00-0000-1000-8000-00805f9b34fb"],"source":"00:1A:7D:DA:71:09","connectable":true,"time":1751062555.2762685,"tx_power":null}

Any help is appreciated this is a very nice integration
John

jthopwood, 2025-06-28:
They appear to be the same for all 4 batteries. Is there something I can do to get more details? The batteries are in a 19" rack and stacked and very difficult to physically access physically. I don't see anything different in the apps I have used to directly access the, and the battery LCD interfaces are the same.

patman15, 2025-06-29:
@jthopwood I had a look again and can't find a reason, why the first two batteries broadcast different information. I suggest you temporarily rename the batteries to their model ID, i.e. `SP16S020` add them and then you should rename if you want. Seems like a bugbin their manufacturing process ...
Let me know if that would work for you.

jthopwood, 2025-06-29:
Naming the batteries to SP16S020 worked. I was able to rename them back to my original names after connecting them to the HA system with no issues to allow me to easily identify the location in the rack.  All 4 batteries are now connected and working.

Thank you so much for your help and for building this integration. 

John