patman15/BMS_BLE-HA #378: Daly BMS with WiFi/Bluetooth Dongle

Joschka93, 2025-06-29:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue%20label%3Aenhancement%20OR%20label%3A%22new%20device%22).
- [x] I have checked out the latest <a href="https://github.com/patman15/BMS_BLE-HA/releases">[image]</a>

### Please describe the enhancement you would like to have.

Please put support for the new WiFi/bluetooth to the addon.

patman15, 2025-06-29:
I do not own such a device, so I depend on information you can provide. So far there is unfortunately none.

Joschka93, 2025-07-01:
(Please excuse any grammatical errors, sent from my mobile device.)

patman15, 2025-07-01:
@Joschka93 this is from the batmon addon not from this integration. If you have issue with this addon, you need to raise an issue at https://github.com/fl4p/batmon-ha/issues. Wrong repository ...

closed 2025-08-12 by github-actions[bot]