patman15/BMS_BLE-HA #381: 多个 同型号 daly bms 蓝牙 设备 只能启用其中的一个，无法同时启用2个

phayes2844, 2025-07-02:
多个 同型号 daly bms 蓝牙 设备 只能启用其中的一个，无法同时启用2个

phayes2844, 2025-07-03:
[image]
"I have two BMS boards and two sets of batteries, all using the same model of Daly BMS. Essentially, there are two identical Bluetooth devices that need to be added. However, I can only add one of them. If I try to add the other one, it fails as shown in the picture. I have to disable one before I can enable the other. It's impossible to add both BMS boards simultaneously."

patman15, 2025-07-03:
Since there is no red text, I assume this is not an error message (yet). Did you follow the [troubleshooting guide](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#if-your-device-is-not-recognized)? In order to help, I need a debug log of the integration, but I would assume you have an issue with BT being overloaded or disturbances. Do you use a Raspberry Pi? How long did you wait (>1 minute)?

phayes2844, 2025-07-03:
[image]
Yes, I am using the 1.8.0-beta releases version, but the issue still persists.

closed 2025-08-12 by github-actions[bot]