patman15/BMS_BLE-HA #383: TTWEN LifePo4 Battery with BT

horinka-gabor, 2025-07-02:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue%20label%3Aenhancement%20OR%20label%3A%22new%20device%22).
- [x] I have checked out the latest <a href="https://github.com/patman15/BMS_BLE-HA/releases">[image]</a>

### Please describe the enhancement you would like to have.

Hello,
I have Ttwen 12V 100Ah battery from Ali. The integration can not recognize it. Can you help me please?
{"name":"TN01491","address":"A5:C2:37:2D:69:0B","rssi":-60,"manufacturer_data":{"26891":"2d37c2a5"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"3C:61:05:3D:DA:5A","connectable":false,"time":1751492353.39337,"tx_power":null}

Thank You!
Best Regards 
G.H.

### Additional context

App:
https://www.jiabaida.com/xiaoxiang.html

patman15, 2025-07-03:
Please try renaming the battery to `JBD-`* as I would guess it is a JBD BMS. I might modify the autodetection at a later point in time. Currently, the information provided by the battery is too wrong.

patman15, 2025-07-03:
> Is anything what can i do to help to you to solve it?

Nope, but I think I can, please consider [README - Troubleshooting - point 6](https://github.com/patman15/BMS_BLE-HA/tree/1.17.0?tab=readme-ov-file#if-your-device-is-not-recognized) your proxy is not configured for active connections.

patman15, 2025-07-03:
@horinka-gabor so issue solved?

horinka-gabor, 2025-07-03:
The last issue is solved, but the main is not yet. The main issue is, i have to rename the battery to recognize. I trust you to close it or not.

patman15, 2025-07-03:
With the current functionality of homeassistant I cannot do better auto-detection of JBD devices. Their BT announcement is simply total crap. I noted it down in case I will rewrite the detection algorithm myself, e.g. for a version 2.x.
So I'll close this issue, thanks for confirming!

closed 2025-07-03 by patman15