patman15/BMS_BLE-HA #385: My JBD BMS doesnt seem to be detected.

manfromua, 2025-07-04:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue%20label%3Aenhancement%20OR%20label%3A%22new%20device%22).
- [x] I have checked out the latest <a href="https://github.com/patman15/BMS_BLE-HA/releases">[image]</a>

### Please describe the enhancement you would like to have.

{"name":"kon_up","address":"A4:C1:37:41:B6:E8","rssi":-54,"manufacturer_data":{"46824":"4137c1a4"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"78:E3:6D:E3:FC:D8","connectable":true,"time":1751644423.7221422,"tx_power":null}

{"name":"kon_down","address":"A4:C1:37:51:B8:D1","rssi":-75,"manufacturer_data":{"47313":"5137c1a4"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb"],"source":"78:E3:6D:E3:FC:D8","connectable":true,"time":1751644423.7221422,"tx_power":null}

### Additional context

JBD-SP16S020
https://cdn.shopify.com/s/files/1/0606/5199/5298/files/JBD-SP16S020.pdf?v=1668845566

patman15, 2025-07-04:
Please temporarily rename to `JBD-`* and try again.

Tomblarom, 2025-07-11:
Hey pat! Thanks, that solved it and I was able to import my batteries. 😊 Please add `JBD-`* to your known issues list.

As a newbie, I found it quite hard, to follow the instructions. I did not realize, that there is really ZERO config needed and that the device shows up and needs to be added manually. I thought your extension does that automatically, once a BMS is detected by a proxy.

patman15, 2025-07-12:
@Tomblarom 

> I thought your extension does that automatically, once a BMS is detected by a proxy.

Well it does, if you didn't rename your JBD device. Unfortunately, JBD can only be recognized based on the name, as they screwed up all other identification mechanisms. 😠 

Ok, please help me understand, where you expected more info:
"known-issues" says:
> JBD BMS detection unfortunately needs to rely on name patterns. If you renamed your battery it most likely will not be detected. I do appreciate issues being raised for new vendor naming schemes to ease the life of other users. To help, please follow the instructions in the last list item for [non-detected devices](https://github.com/patman15/BMS_BLE-HA#if-your-device-is-not-recognized).

And point 2 of "if your device is not recognized" says:
> If a name detection pattern is listed ("show up as"), make sure your device matches it.

Do you want me to add the "JBD-*" pattern to the first paragraph? I didn't put it there originally, as I wanted to maintain the list of patterns only once, which is in the "supported devices" list.

Happy to hear your thoughts.

Tomblarom, 2025-07-14:
I had to rename them, because I have more than 4 JBD batteries in my house. 😅

Ok at this point I think it was late in the night and I oversaw the major instructions, that would have helped me quicker. 😁 Honestly I rushed over it and in retro perspective no information in your guide is missing.

Thanks for picking up on my comment! Highly appreciate your work! 😉

patman15, 2025-07-14:
@Tomblarom ok great. Still if you find something to make better in the documentation, let me know. For the moment, I'll close the issue, just raise a new one, if you run into troubles. Thanks!