patman15/BMS_BLE-HA #388: [unavavailable] for some time - since 1.18

gkathan, 2025-07-07:
### Checklist

- [x] I have tried the latest version of BMS_BLE-HA.
- [x] I have enabled debug logging for my installation.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [x] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

offgridtect smartbatt (OGT) 

### Describe the issue

hi patrick,
if you remember - we two did the iontegratrion for my offgridtec smartbatt back 1 year ago or so ... cannot remember whether mine is a typa A or type B ...
your integration worked rock solid till i installed 1.18 - from then on i have regular "unavailable" messages for a cycle of 30-60 seconds and then again values 
any ideas ?`
cheerz

### Reproduction steps

```text
1.
2.
3.
...
```

### Debug logs

cannot find option to enable debug logging on your hacs integration

patman15, 2025-07-07:
I have 1.18 running with my type B battery without any issues, which actually doesn't mean much. 😎 I need a debug log to be able to say more.

> cannot find option to enable debug logging on your hacs integration

Why not? Did you follow the [instructions](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics)? Should be there as soon as you have at least 1 battery active.

Tulexcorp, 2025-07-11:
Hi  Pat,

I'll just wait for the fix, I'm likely to mess things up downgrading. 

For logging, I meant across the board. We can't see info for any integrations. I hadn't been to that area in settings since the last few updates, so it was a bit of a shock to not be able to log the way we could in the past.

patman15, 2025-07-12:
@Tulexcorp there are two issues actually. I fixed the first in the [branch linked to this ticket](https://github.com/patman15/BMS_BLE-HA/tree/388-seplos-v2-double-response---since-118). Could you try it via [manual install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#installation) please? I will fix a second issue a soon as you confirm the first is gone. So I would condly ask you for a second test then.

Tulexcorp, 2025-07-12:
Installed. Log attached.

Hopefully these are real issues you found and not caused by something on my end. After sending the previous log, I rebooted HA. While I did still get the occasional unavailable, it was nowhere near as bad.

[home-assistant_bms_ble_2025-07-12T15-31-39.007Z.log](https://github.com/user-attachments/files/21197064/home-assistant_bms_ble_2025-07-12T15-31-39.007Z.log)

patman15, 2025-07-12:
> Hopefully these are real issues you found and not caused by something on my end. 

No these are real issues that surfaced due to a change I need to do for other BMSs. I did another small fix, can you check again? This is very helpful!

Tulexcorp, 2025-07-12:
I let it run for an hour, looks pretty good. I think Seplos has it's own issues with BT, but when I do see unavailable, they don't last as long as they did in the past, seems like about 30 sec now. From my graph over an hour - 

Battery - Number of Unavailable - Signal Strength - Link Quality
1 - 1 - 85 - 99
2 - 0 - 85 - 100
3 - 2 - 86 - 97
4 - 2 - 79 - 98
5 - 0 - 80 - 100
6 - 0 - 89 - 100

patman15, 2025-07-13:
Should I do a new patch release? Without log files I cannot fix something specific for you. I don't think there is a high chance that the fixes so far affect the OGT BMS as well. Just saying ...