patman15/BMS_BLE-HA #394: SOK BMS unavailable after 1.18.0 upgrade

awilliam, 2025-07-11:
### Checklist

- [x] I have tried the latest version of BMS_BLE-HA.
- [x] I have enabled debug logging for my installation.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [x] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

SOK SK12V100H

### Describe the issue

SOK BMS reports unavailable after upgrade to 1.18.0.  Works again after manual downgrade to 1.17.0.

### Reproduction steps

```text
1. Have SOK SK12V100H working with 1.17.0
2. Upgrade to 1.18.0
3.
...
```

### Debug logs

```
2025-07-04 15:23:33.781 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JTVVG82F41FC8RC6NBP4HF5S version=1 domain=bms_ble title=SOK-AA02120 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=D6:6C:0A:61:17:
9B>
2025-07-04 15:23:33.781 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] initializing Chunguang Song ABC BMS, BT address: D6:6C:0A:61:17:9B
2025-07-04 15:23:33.782 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for SOK-AA02120 (D6:6C:0A:61:17:9B) as Chunguang Song ABC BMS
2025-07-04 15:23:33.782 DEBUG (MainThread) [custom_components.bms_ble] SOK-AA02120: advertisement: {'name': 'SOK-AA02120', 'address': 'D6:6C:0A:61:17:9B', 'rssi': -62, 'manufacturer_data': {}, 'service_data': {}, 'service_uuids': ['000018
00-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000ffe0-0000-1000-8000-00805f9b34fb', '0000fff0-0000-1000-8000-00805f9b34fb', 'f000
ffc0-0451-4000-b000-000000000000'], 'source': 'E8:4E:06:A6:CE:F2', 'advertisement': AdvertisementData(local_name='SOK-AA02120', service_uuids=['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-
1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000ffe0-0000-1000-8000-00805f9b34fb', '0000fff0-0000-1000-8000-00805f9b34fb', 'f000ffc0-0451-4000-b000-000000000000'], rssi=-62), 'device': BLEDevice(D6:6C:0A:61:17:9B, SO
K-AA02120), 'connectable': True, 'time': 2534306.530514934, 'tx_power': None, 'raw': None}
2025-07-04 15:23:33.782 DEBUG (MainThread) [custom_components.bms_ble] SOK-AA02120: BMS data update
2025-07-04 15:23:33.782 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] connecting BMS
2025-07-04 15:23:33.783 DEBUG (MainThread) [bleak_retry_connector] D6:6C:0A:61:17:9B - D6:6C:0A:61:17:9B: Connection attempt: 1
2025-07-04 15:23:40.475 DEBUG (MainThread) [bleak_retry_connector] D6:6C:0A:61:17:9B - D6:6C:0A:61:17:9B: Connected after 1 attempts
2025-07-04 15:23:40.691 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] TX BLE req #1 (WNR): ee c4 00 00 00 4f
2025-07-04 15:23:40.945 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] TX BLE request timed out.
2025-07-04 15:23:40.946 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] TX BLE req #2 (WNR): ee c4 00 00 00 4f
2025-07-04 15:23:41.450 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] TX BLE request timed out.
2025-07-04 15:23:41.450 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] TX BLE req #3 (WNR): ee c4 00 00 00 4f
2025-07-04 15:23:42.455 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] TX BLE request timed out.
2025-07-04 15:23:42.455 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] TX BLE req #1 (!WNR): ee c4 00 00 00 4f
2025-07-04 15:23:42.491 WARNING (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] TX BLE request error, retrying connection (BleakDBusError)
2025-07-04 15:23:42.491 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] disconnecting BMS
2025-07-04 15:23:44.740 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] disconnected from BMS
2025-07-04 15:23:44.742 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] connecting BMS
2025-07-04 15:23:44.743 DEBUG (MainThread) [bleak_retry_connector] D6:6C:0A:61:17:9B - D6:6C:0A:61:17:9B: Connection attempt: 1
2025-07-04 15:23:46.081 WARNING (MainThread) [homeassistant.components.websocket_api.http.connection] [281472275363648] from 192.168.8.224 (Mozilla/5.0 (X11; Linux x86_64; rv:137.0) Gecko/20100101 Firefox/137.0): Disconnected: Received close message during auth phase
2025-07-04 15:23:50.094 DEBUG (MainThread) [bleak_retry_connector] D6:6C:0A:61:17:9B - D6:6C:0A:61:17:9B: Connected after 1 attempts
2025-07-04 15:23:50.311 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] TX BLE req #1 (WNR): ee c2 00 00 00 46
2025-07-04 15:23:50.401 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] RX BLE data: bytearray(b'\xcc\xf0\xd13\x00\\\xf1\xff\xa0\x86\x01\x838\x01\x05\x00\x00\x00\x00`')
2025-07-04 15:23:50.408 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] RX BLE data: bytearray(b'\xcc\xf3\x15\n\x1e\x002\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11')
2025-07-04 15:23:50.461 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] RX BLE data: bytearray(b'\xcc\xf4\x01\xe9\x0c\x00\x02\xf2\x0c\x00\x03\xf0\x0c\x00\x04\xea\x0c\x00\x00u')
2025-07-04 15:23:50.462 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] TX BLE req #1 (WNR.): ee c1 00 00 00 ce
2025-07-04 15:23:50.560 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] RX BLE data: bytearray(b'\xcc\xf0\xd03\x00\\\xf1\xff\xa0\x86\x01\x838\x01\x05\x00\x00\x00\x00O')
2025-07-04 15:23:50.591 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] RX BLE data: bytearray(b'\xcc\xf2\x01\x01\x01\x1d\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00D')
2025-07-04 15:23:50.591 ERROR (MainThread) [custom_components.bms_ble] Unexpected error fetching SOK-AA02120 data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 382, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/bms_ble/coordinator.py", line 131, in _async_update_data
    if not (bms_data := await self._device.async_update()):
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/bms_ble/plugins/basebms.py", line 448, in async_update
    data: BMSsample = await self._async_update()
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/bms_ble/plugins/abc_bms.py", line 178, in _async_update
    result: BMSsample = BMS._decode_data(self._data_final)
                        ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/bms_ble/plugins/abc_bms.py", line 160, in _decode_data
    data[cmd][idx : idx + size], byteorder="little", signed=sign
    ~~~~^^^^^
KeyError: 249
2025-07-04 15:23:50.597 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching SOK-AA02120 data in 16.814 seconds (success: False)
2025-07-04 15:23:50.597 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS (SOK-AA02120)
2025-07-04 15:23:50.597 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] disconnecting BMS
2025-07-04 15:23:54.882 DEBUG (MainThread) [custom_components.bms_ble.abc_bms::SOK-AA02120:179B)] disconnected from BMS

```

patman15, 2025-07-12:
@awilliam thanks for reporting, the issues were actual present also before, just revealed by the update. I fixed them in the [branch linked to this ticket](https://github.com/patman15/BMS_BLE-HA/tree/394-sok-bms-unavailable-after-1180-upgrade). Could you please check via [manual install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#installation) and confirm? Thanks!

awilliam, 2025-07-13:
> [@awilliam](https://github.com/awilliam) thanks for reporting, the issues were actual present also before, just revealed by the update. I fixed them in the [branch linked to this [...]

Works for me.  Thanks!

closed 2025-07-13 by patman15