patman15/BMS_BLE-HA #408: Add support for Camelenergy battery (BMS-SMART broadcasting bms)

marican, 2025-07-21:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue%20label%3Aenhancement%20OR%20label%3A%22new%20device%22).
- [x] I have checked out the latest <a href="https://github.com/patman15/BMS_BLE-HA/releases">[image]</a>

### Please describe the enhancement you would like to have.

I realize this is probably only helps me as these batteries are basically unheard of, but ive got 2 of these 24v lifepo batteries with bluetooth app support from a company called camelenergy.  https://www.amazon.com/dp/B0DQSR8K2H?ref_=ppx_hzsearch_conn_dt_b_fed_asin_title_1
They use a very specific android app in order to communicate SOC and other stats.  Every other bluetooth battery monitor app I've tried is unable to recognize them.  The BMS advertises as BMS-SMART_last6ofuuid. 

I'm not well versed in bluetooth data requests, nor github in general, and I've tried to read all available documentation for this kind of feature request, but if I missed something please let me know. 

Thsnks,
Drew

marican, 2025-08-11:
Sorry for the delay on this,  Here's the advertisement monitor information for both batteries.   I'm working to get the bluetooth traffic captured this evening.  Hopefully will update then.

Thanks!

{"name":"BMS-SMART_708BFC","address":"C5:CF:39:70:8B:FC","rssi":-78,"manufacturer_data":{},"service_data":{},"service_uuids":[],"source":"DC:45:46:85:5D:3A","connectable":true,"time":1754947291.43655,"tx_power":null}

{"name":"BMS-SMART_1F8FF7","address":"8A:40:18:1F:8F:F7","rssi":-93,"manufacturer_data":{},"service_data":{},"service_uuids":[],"source":"DC:45:46:85:5D:3A","connectable":true,"time":1754947291.507552,"tx_power":null}

marican, 2025-08-12:
Heres the collected output from wireshark.  One thing of note that the doc doesn't mention is that you need to have an active adb connection established via adb devices -l inorder to get the android bluetooth snoop to show up as a source in wireshark.  Perhaps thats obvious, but took me a while to figure out.  Anyway, let me know if this helps.  Thanks again!

[camelenergybtwireshark.zip](https://github.com/user-attachments/files/21726127/camelenergybtwireshark.zip)

marican, 2025-08-12:
Not sure if this helps, but these are 24v lifepo batteries and the app was showing a SOC of 100% when I snooped on them.

patman15, 2025-08-12:
@marican unfortunately this is not a Bluetooth Low Energy device that supports the GATT protocol. It's legacy BT which is not supported by Home Assistant.

closed 2025-08-24 by patman15