patman15/BMS_BLE-HA #412: Tian Power BMS

grydo, 2025-07-22:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue%20label%3Aenhancement%20OR%20label%3A%22new%20device%22).
- [x] I have checked out the latest <a href="https://github.com/patman15/BMS_BLE-HA/releases">[image]</a>

### Please describe the enhancement you would like to have.

Hi, It would be very interesting if Tian Power BMS could be read by this software, given the various requests from users.

Thank you very much.

patman15, 2025-07-22:
@grydo not sure which users (except for you) have requested it and where, but sure, we can give it a try. Do you have any of the data that was asked for this template available? Any references?

grydo, 2025-07-22:
What data exactly would you like so that we can be of the best help to you?

patman15, 2025-07-23:
In general I need:
> For new BMS, please provide any links to documentation, the BT protocol, link to an Android app, other open source implementations, or the like.

In particular these things help a lot:
> Add any other context (documentation) or screenshots about the feature request here. For new BMS types, please provide the advertisement data of the device. (Go to the [advertisement monitor](https://my.home-assistant.io/redirect/bluetooth_advertisement_monitor/), click the device in question, and provide the information via **`copy to clipboard`**). Optionally, the [service explorer script](https://github.com/hbldh/bleak/blob/develop/examples/service_explorer.py) can help as well. To speed up the integration please see if you are able to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery.

grydo, 2025-07-23:
protocol bms

[tianpower.pdf](https://github.com/user-attachments/files/21391673/tianpower.pdf)

app android

https://play.google.com/store/apps/details?id=com.green.basen.jizhi&pcampaignid=web_share

link code esphome 

https://github.com/syssi/esphome-tianpower-bms

patman15, 2025-07-24:
Hey! A sample from the [advertisement monitor](https://my.home-assistant.io/redirect/bluetooth_advertisement_monitor/) via `copy to clipboard` would be nice but otherwise I think I'm good. Just need to find some time to do the implementation, but there are a number of request still ahead of me.

patman15, 2025-07-26:
@grydo I had some hours to implement the BMS. Could you try the [branch linked to this issue](https://github.com/patman15/BMS_BLE-HA/tree/412-tian-power-bms) via [manual install](https://github.com/patman15/BMS_BLE-HA#installation) and send me a debug log?

grydo, 2025-07-27:
I'll do it as soon as I get home

grydo, 2025-07-29:
[image]
[image]

BMS reading correct, tested on 3 batteries.
The only correction, if possible, is to enter the voltage of the 16 cells.

grydo, 2025-07-29:
@patman15 

log:

[bms_ble-01K1AJK7XDFZ015J2VBBVKEYF3-TP_Batteria 1-b717f4ba2ec931fa6ae4b600b8409683.json](https://github.com/user-attachments/files/21494332/bms_ble-01K1AJK7XDFZ015J2VBBVKEYF3-TP_Batteria.1-b717f4ba2ec931fa6ae4b600b8409683.json)

Did you check the attribute of the delta_voltage sensor by clicking on it? yes correct value.

I meant if it is possible to insert the values of all 16 cells

grydo, 2025-07-29:
@patman15 

[home-assistant_bms_ble_2025-07-29T17-29-05.490Z.log](https://github.com/user-attachments/files/21494539/home-assistant_bms_ble_2025-07-29T17-29-05.490Z.log)

patman15, 2025-07-30:
@grydo the cell voltages are here:
[image]
If you want to have individual sensors, did you check the [FAQ](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#how-do-i-get-the-cell-voltages-as-individual-sensor-for-tracking)?

grydo, 2025-07-30:
Thanks, I fixed it, will the next update integrate this BMS?

patman15, 2025-07-30:
If you confirm you are happy, I'm ok to merge and release it with the next version.

grydo, 2025-07-30:
I confirm, we will make a donation soon, thanks again.