patman15/BMS_BLE-HA #415: Getting detail of 'problem'

haysdb, 2025-07-24:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I have thought about [providing a debug log](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) and/or [diagnostic data](https://www.home-assistant.io/docs/configuration/troubleshooting/#download-diagnostics).

### Describe the issue

Seplos v3

I added a new sensor problem_code (by modifying sensor.py) to return the problem code. I'm trying to decode the bits and not having much success. I'm cross referencing with the Modbus document for Seplos BMS v3 assuming the bluetooth bytes will match the modbus bytes, but they don't seem to match. Can you point me in a direction of where to find out how to decode the problem code?

This isn't elegant but this is what I added to sensor.py:

```    
    BmsEntityDescription(
        key="problem_code",
        name="Problem Code",
        icon="mdi:alert-decagram",
        value_fn=lambda data: data.get("problem_code"),
    ),
```

haysdb, 2025-08-02:
I’m not following. There is no problem_code until I added it. There is only problem, ‘on’ or ‘off’. I have the Modbus document already and said it does not match up with the code I am getting from the BMS.

patman15, 2025-08-02:
> There is no problem_code sensor until I added it. 

Correct, but there is an `attribute` `problem_code` to the sensor `problem`, see here

[image]

> What document?

If you go to this https://github.com/patman15/BMS_BLE-HA/issues/24#issuecomment-2185892838 there is a document called [BMS Modbus-RTU Protocol 蓝牙.docx](https://github.com/user-attachments/files/15951839/BMS.Modbus-RTU.Protocol.docx), then see block EIC on page 14.

haysdb, 2025-08-03:
Thank you. I missed the attribute.

closed 2025-08-29 by patman15