patman15/BMS_BLE-HA #432: Problem after BMS firmware update — "BMS communication timed out"

Tarasenpai, 2025-08-04:
Device/Setup Info:

BMS model: TDT BMS 6032

Firmware updated from: 6032_20016S000_L_41

Firmware updated to: 6032_20016S000_L_65

Integration: Using BMS_BLE-HA dongle

Issue start: After firmware update

Description:
After updating the BMS firmware to version 6035_20016S000_L_65, the BLE dongle attempts to connect to the BMS but fails shortly after with the following repeating error:
Failed setup, will retry: BMS communication timed out
Everything worked fine on firmware 6035_20016S000_L_41. After the update, the system appears to detect the BMS via Bluetooth but cannot complete the communication handshake.

Additional Notes:

No hardware changes were made, only firmware update.

Might be due to changes in BLE protocol or handshake in new firmware

Tarasenpai, 2025-08-04:
[log_fragment.txt](https://github.com/user-attachments/files/21583971/log_fragment.txt)

[home-assistant_2025-08-04T19-47-31.441Z.log](https://github.com/user-attachments/files/21583972/home-assistant_2025-08-04T19-47-31.441Z.log)

patman15, 2025-08-08:
Thanks a lot, will merge it for the next version!

closed 2025-08-08 by patman15