patman15/BMS_BLE-HA #438: Lossigy 48v 100mah battery with bms/bluetooth addition

JustTheDazz, 2025-08-07:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue%20label%3Aenhancement%20OR%20label%3A%22new%20device%22).
- [x] I have checked out the latest <a href="https://github.com/patman15/BMS_BLE-HA/releases">release</a>.

### Please describe the enhancement you would like to have.

I am attempting to add a Lossigy 48v 100mah battery but it doesn't appear to be in the supported list, and I am unable to pair with it as I get the message "bluetoothctl pair 3F:A0:00:07:31:39                 
Attempting to pair with 3F:A0:00:07:31:39
AdvertisementMonitor path registered
hci0 type 6 discovering off
hci0 3F:A0:00:07:31:39 type LE Public connected eir_len 12
[CHG] Device 3F:A0:00:07:31:39 Connected: yes
hci0 3F:A0:00:07:31:39 auth failed with status 0x05 (Authentication Failed)
Failed to pair: org.bluez.Error.AuthenticationFailed"

I have no idea if it's a password issue, or if they even use a password. Is there a way to add this battery? This is where I got it from - 
https://www.amazon.com/dp/B0CZ4GMY5Q/?coliid=I3SX4T8GKO9IB6&colid=6F1QYMQTWDVA&ref_=list_c_wl_lv_ov_lig_dp_it&th=1

### Additional context

I have an app on my iphone and it works fine,  the link for it is - https://apps.apple.com/us/app/bat-bms/id1611741274

{"name":"lossigy","address":"3F:A0:00:07:31:39","rssi":-90,"manufacturer_data":{"12601":"0700a03f"},"service_data":{},"service_uuids":["0000ff00-0000-1000-8000-00805f9b34fb","00000001-0000-1000-8000-00805f9b34fb"],"source":"3C:64:CF:9D:DC:B0","connectable":true,"time":1754558346.9119296,"tx_power":null}

patman15, 2025-08-07:
You can try putting the name here instead of `TP_*`:
https://github.com/patman15/BMS_BLE-HA/blob/49ffe37d990e259ecf186560d72122766475b399/custom_components/bms_ble/plugins/tianpwr_bms.py#L43
I suspect it is a TianPower BMS, but it has no identification mechanisms so I can't modify the auto-detection in general.

JustTheDazz, 2025-08-08:
> Hi! First thing is to get your physical setup so that the RSSI value gets
> safely above -75dBm. Please check the advertisement monitor
> <https://my.home-assistant.io/redirect/bluetooth_advertisement_monitor/>
> and adjust your setup to raise the RSSI level.

Hello. It is at -86dbm( my bt adapter is less than 10ft from the battery)

patman15, 2025-08-08:
Well -86 dBm is incredibly bad. Maybe move further away, check for metal disturbing the transmission, see that WiFi is in good distance (away from BT).
With that bad channel the connection will not work.

patman15, 2025-08-08:
Ok great. Just noticed that I forgot to tell you to change the name also here:
https://github.com/patman15/BMS_BLE-HA/blob/409ebcbf6c81052bebd9515e1dec93e18625c7e3/custom_components/bms_ble/manifest.json#L275
After a restart it should work then. 🤞

JustTheDazz, 2025-08-08:
That changed things , now I see it as a new device , under BLE Battery
Management I see Setup lossigy (:07:31:39) as TIANPWR BMS, but after
clicking add, then setup , it works fora few minutes then I get created
configuration for losing, click finish, then I see Failed setup, will
retry: BMS communication timed out...

JustTheDazz, 2025-08-08:
> Ok, can you get me a debug log?

sure can..

patman15, 2025-08-10:
The log is too short, because the first sequence only shows it trying to find the device and the second is incomplete ...

patman15, 2025-08-11:
Ok, most likely it is a JBD device. So please undo the change in 
https://github.com/patman15/BMS_BLE-HA/blob/49ffe37d990e259ecf186560d72122766475b399/custom_components/bms_ble/plugins/tianpwr_bms.py#L43
and do it here:
https://github.com/patman15/BMS_BLE-HA/blob/0ec8acc147ea1db9bfe3e094c594c405756d9244/custom_components/bms_ble/plugins/jbd_bms.py#L65
i.e. put the name here for detection.

JustTheDazz, 2025-08-11:
Ok, I made the change, however I still get an error"
2025-08-11 06:56:13.532 WARNING (SyncWorker_0) [homeassistant.loader] We
found a custom integration bambu_lab which has not been tested by Home
Assistant. This component might cause stability problems, be sure to
disable it if you experience issues with Home Assistant
2025-08-11 06:56:13.533 WARNING (SyncWorker_0) [homeassistant.loader] We
found a custom integration govee which has not been tested by Home
Assistant. This component might cause stability problems, be sure to
disable it if you experience issues with Home Assistant
2025-08-11 06:56:13.533 WARNING (SyncWorker_0) [homeassistant.loader] We
found a custom integration sonos_cloud which has not been tested by Home
Assistant. This component might cause stability problems, be sure to
disable it if you experience issues with Home Assistant
2025-08-11 06:56:13.534 WARNING (SyncWorker_0) [homeassistant.loader] We
found a custom integration bms_ble which has not been tested by Home
Assistant. This component might cause stability problems, be sure to
disable it if you experience issues with Home Assistant
2025-08-11 06:56:13.534 WARNING (SyncWorker_0) [homeassistant.loader] We
found a custom integration bluetti_bt which has not been tested by Home
Assistant. This component might cause stability problems, be sure to
disable it if you experience issues with Home Assistant
2025-08-11 06:56:13.535 WARNING (SyncWorker_0) [homeassistant.loader] We
found a custom integration emporia_vue which has not been tested by Home
Assistant. This component might cause stability problems, be sure to
disable it if you experience issues with Home Assistant
2025-08-11 06:56:13.535 WARNING (SyncWorker_0) [homeassistant.loader] We
found a custom integration hacs which has not been tested by Home
Assistant. This component might cause stability problems, be sure to
disable it if you experience issues with Home Assistant
2025-08-11 06:56:13.535 WARNING (SyncWorker_0) [homeassistant.loader] We
found a custom integration tapo which has not been tested by Home
Assistant. This component might cause stability problems, be sure to
disable it if you experience issues with Home Assistant
2025-08-11 06:56:18.066 WARNING (MainThread) [homeassistant.helpers.frame]
Detected that custom integration 'tapo' closes the Home Assistant aiohttp
session at custom_components/tapo/hass_tapo.py, line 34: device = await
connect(config=self.config, session=session). Please create a bug report at
https://github.com/petretiandrea/home-assistant-tapo-p100/issues
2025-08-11 06:56:20.697 WARNING (SyncWorker_2) [py.warnings]
/usr/local/lib/python3.13/site-packages/awsiot/mqtt_connection_builder.py:161:
UserWarning: pkg_resources is deprecated as an API. See
https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources
package is slated for removal as early as 2025-11-30. Refrain from using
this package or pin to Setuptools<81.
  import pkg_resources

2025-08-11 06:56:25.951 DEBUG (MainThread)
[custom_components.bms_ble.tianpwr_bms::lossigy:3139)] TX BLE req #1 (WNR):
55 04 83 aa
2025-08-11 06:56:26.669 DEBUG (MainThread)
[custom_components.bms_ble.tianpwr_bms::lossigy:3139)] TX BLE request timed
out.
2025-08-11 06:56:26.669 DEBUG (MainThread)
[custom_components.bms_ble.tianpwr_bms::lossigy:3139)] TX BLE req #2 (WNR):
55 04 83 aa
2025-08-11 06:56:26.696 WARNING (SyncWorker_1)
[homeassistant.helpers.service] Unable to parse services.yaml for the
emporia_vue integration: extra keys not allowed @
data['set_charger_current']['target']['device'][0]['entity']
2025-08-11 06:56:28.101 DEBUG (MainThread)
[custom_components.bms_ble.tianpwr_bms::lossigy:3139)] TX BLE request timed
out.
2025-08-11 06:56:28.101 DEBUG (MainThread)
[custom_components.bms_ble.tianpwr_bms::lossigy:3139)] TX BLE req #3 (WNR):
55 04 83 aa
2025-08-11 06:56:30.960 DEBUG (MainThread)
[custom_components.bms_ble.tianpwr_bms::lossigy:3139)] TX BLE request timed
out.
2025-08-11 06:56:30.961 DEBUG (MainThread)
[custom_components.bms_ble.tianpwr_bms::lossigy:3139)] TX BLE req #1
[... 154 lines cut ...]
[custom_components.bms_ble.tianpwr_bms::lossigy:3139)] TX BLE request timed
out.
2025-08-11 06:57:23.481 WARNING (MainThread)
[custom_components.bms_ble.tianpwr_bms::lossigy:3139)] TX BLE request
error, retrying connection (BleakError)
2025-08-11 06:57:23.481 DEBUG (MainThread)
[custom_components.bms_ble.tianpwr_bms::lossigy:3139)] disconnecting BMS
(False)
2025-08-11 06:57:23.481 DEBUG (MainThread)
[custom_components.bms_ble.tianpwr_bms::lossigy:3139)] connecting BMS
2025-08-11 06:57:23.481 DEBUG (MainThread) [bleak_retry_connector]
3F:A0:00:07:31:39 - 3F:A0:00:07:31:39: Connection attempt: 1
2025-08-11 06:57:26.209 DEBUG (MainThread) [bleak_retry_connector]
3F:A0:00:07:31:39 - 3F:A0:00:07:31:39: Connected after 1 attempts
2025-08-11 06:57:26.479 DEBUG (MainThread) [custom_components.bms_ble]
lossigy: BMS communication timed out
2025-08-11 06:57:26.479 DEBUG (MainThread) [custom_components.bms_ble]
Finished fetching lossigy data in 12.002 seconds (success: False)
2025-08-11 06:57:26.479 DEBUG (MainThread) [custom_components.bms_ble]
Shutting down BMS (lossigy)
2025-08-11 06:57:26.479 DEBUG (MainThread)
[custom_components.bms_ble.tianpwr_bms::lossigy:3139)] disconnecting BMS
(True)
2025-08-11 06:57:27.737 DEBUG (MainThread)
[custom_components.bms_ble.tianpwr_bms::lossigy:3139)] disconnected from BMS

patman15, 2025-08-11:
Great, so ticket can be closed?