patman15/BMS_BLE-HA #441: Add sensors for each cell voltage

chedim, 2025-08-09:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue%20label%3Aenhancement%20OR%20label%3A%22new%20device%22).
- [x] I have checked out the latest <a href="https://github.com/patman15/BMS_BLE-HA/releases">release</a>.

### Please describe the enhancement you would like to have.

Please add a sensor for every cell voltage. The current solution that uses templates is unwieldy to create/recreate with my 24s battery and all that is needed in the add-on is a foreach loop. Thank you!

krahabb, 2025-08-21:
I would too definitely like the addition.
These additional sensors could be made initially disabled so to not clutter the HA state machine by default (only users needing them would then enable those) or, better, add a config option to create or not these entities.

Teunhen, 2025-08-30:
I would love to have all the cell voltages.

patman15, 2025-08-31:
@Teunhen you are aware that this is possible? 

> [!TIP] 
> The [FAQ](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#faq) describes [how to add individual cell voltage sensors](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#how-do-i-get-the-cell-voltages-as-individual-sensor-for-tracking).

The discussion is about whether to add them automatically (initially disabled).

irokezzz, 2025-09-16:
I solved this situation like this:

template:
  - sensor:
      - name: "Battery1 Cell 1 Voltage"
        unit_of_measurement: "V"
        device_class: voltage
        state: >
          {% set cells = state_attr('sensor.Battery1_delta_voltage', 'cell_voltages') %}
          {% if cells %}
            {{ cells.split(',')[0] | float }}
          {% else %}
            unavailable
          {% endif %}

      - name: "Battery1 Cell 2 Voltage"
        unit_of_measurement: "V"
        device_class: voltage
        state: >
          {% set cells = state_attr('sensor.Battery1_delta_voltage', 'cell_voltages') %}
          {% if cells %}
            {{ cells.split(',')[1] | float }}
          {% else %}
            unavailable
          {% endif %}

      - name: "Battery1 Cell 3 Voltage"
        unit_of_measurement: "V"
        device_class: voltage
        state: >
          {% set cells = state_attr('sensorBattery1_delta_voltage', 'cell_voltages') %}
          {% if cells %}
            {{ cells.split(',')[2] | float }}
          {% else %}
            unavailable
          {% endif %}

      # .........

      - name: "Battery1 Cell 16 Voltage"
        unit_of_measurement: "V"
        device_class: voltage
        state: >
          {% set cells = state_attr('sensor.Battery1_delta_voltage', 'cell_voltages') %}
          {% if cells %}
            {{ cells.split(',')[15] | float }}
          {% else %}
            unavailable
          {% endif %}

patman15, 2025-09-19:
As an intermediate solution, the latest beta release [2.0.0-beta.5](https://github.com/patman15/BMS_BLE-HA/releases/tag/2.0.0-beta.5) now contains a maximum cell voltage and minimum cell voltage sensor.

alexdns1, 2025-10-22:
I dont see why thats needed all necessary voltages are already included in prev stable  1.21.0 
Ex. If anyone needs min/max 

`

      - name: "XDZN_001_B871 Max Cell Voltage"
      
        unique_id: "a2c32d55-52e2-42e7-9c20-1e4c7e30c3a1"

        state: >-

          {% if has_value("sensor.xdzn_001_b871_delta_voltage") %}

            {{ state_attr("sensor.xdzn_001_b871_delta_voltage", "cell_voltages") | max }}

          {% else %}

            None

          {% endif %}

        unit_of_measurement: "V"

        device_class: voltage

        state_class: measurement

        availability: >-

          {{ has_value(state_attr('sensor.xdzn_001_b871_delta_voltage', 'cell_voltages')) }}

`

krahabb, 2025-11-12:
I agree, in general, with the idea of exposing every available entity but this in turn would 'overwhelm' a lot of users which are not really interested in these additional entities.
I think the best solution would be to provide a configuration option in ConfigEntry so that everyone could customize the component at configuration time (maybe with re-configure feature too). This way, every user could decide if he wants those cell voltages or not.

Also, technically speaking, having the voltages exposed through an attribute is putting some 'stress' on the recorder which needs to serialize all of the data to a secondary table every time it flushes a new state. Now, this by default happens only once every 30 sec but in my case I've modified the default polling by increasing its frequency and even though I don't see my raspi breaking down for this, I'm kind of inclined to think this is just useless since in practical terms there's no easy way to query attributes history. This in turn brings back the question:
Expose the cell voltages or not?
- If Yes, then it's better to have split sensors which are easily and directly manageable in the UI.
- If No, then it's better to completely discard them (exposing them in an attribute alone is a bit of 'no use' in my opinion)

Teunhen, 2025-11-12:
I vote for expose them, you can always hide them if you don't need/want them. Or like you said in a config.

avg-I, 2025-11-13:
Some HA devices provide _dozens_ of sensors with most of them being disabled by default.
Those who don't need those sensors don't have to do anything, but those who want the sensors can simply enable (some of) them.
I think that it is a good model.

elalemanyo, 2026-01-13:
> I will reconsider this topic in the course of my pull-request for a core integration. Still I wonder what the benefit of this information is (especially with like 32+ cells) that cannot be gathered from the highest/lowest voltage sensor together with the cell number. Both being available.

@patman15 I like having something like this on my dashboard, I can easy see if one cell is not working correctly:

[image]

But I only have 16 😅

randyoo, 2026-02-12:
> Attributes are not recorded.

Humbly, I think you are mistaken:
https://community.home-assistant.io/t/history-of-sensor-attributes/393209/3
https://github.com/cyberjunky/home-assistant-garmin_connect/issues/280

```
MariaDB [homeassistant]> SELECT
    ->     s.state AS voltage_delta,
    ->     JSON_EXTRACT(sa.shared_attrs, '$.cell_voltages') as cell_voltages,
    ->     FROM_UNIXTIME(s.last_updated_ts) as last_updated_at
    -> FROM states s
    -> JOIN states_meta sm ON s.metadata_id = sm.metadata_id
    -> LEFT JOIN state_attributes sa ON s.attributes_id = sa.attributes_id
    -> WHERE sm.entity_id = 'sensor.z121004265_delta_cell_voltage'
    -> ORDER BY s.last_updated_ts DESC
    -> LIMIT 10;
+---------------+------------------------------+----------------------------+
| voltage_delta | cell_voltages                | last_updated_at            |
+---------------+------------------------------+----------------------------+
| 0.003         | [3.38, 3.377, 3.377, 3.379]  | 2026-02-12 12:19:38.088307 |
| 0.004         | [3.379, 3.38, 3.376, 3.38]   | 2026-02-12 12:19:07.107948 |
| 0.004         | [3.376, 3.38, 3.376, 3.38]   | 2026-02-12 12:18:36.231990 |
| 0.003         | [3.38, 3.377, 3.379, 3.379]  | 2026-02-12 12:18:05.912904 |
| 0.004         | [3.377, 3.38, 3.376, 3.38]   | 2026-02-12 12:16:34.268321 |
| 0.004         | [3.379, 3.379, 3.376, 3.38]  | 2026-02-12 12:16:03.900521 |
| 0.003         | [3.376, 3.379, 3.377, 3.377] | 2026-02-12 12:15:33.984063 |
| 0.003         | [3.377, 3.379, 3.379, 3.376] | 2026-02-12 12:15:02.099649 |
| 0.002         | [3.377, 3.379, 3.379, 3.377] | 2026-02-12 12:14:31.288569 |
| 0.004         | [3.38, 3.376, 3.38, 3.376]   | 2026-02-12 12:13:29.066780 |
+---------------+------------------------------+----------------------------+
10 rows in set (0.001 sec)
```

randyoo, 2026-02-12:
Okay, but they are still growing the size of the recorder DB, whereas a disabled-by-default sensor would not.

If enabled, a cell voltage sensor wouldn't store the same value again and again, only when it changes.

Additionally, you create new recorder entries even if the delta V hasn't changed.

The "tip" I shared earlier was straight from HA dev docs. Another option could be [excluding from recording](https://developers.home-assistant.io/blog/2023/09/20/excluding-state-attributes-from-recording/).

patman15, 2026-02-14:
> The "tip" I shared earlier was straight from HA dev docs.

Probably lost in overload. Noted it down for future handling. Thanks!

9812ga98sd, 2026-02-27:
This is a great idea