patman15/BMS_BLE-HA #452: PowerBoozt Battery integration desire

hlathome, 2025-08-15:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue%20label%3Aenhancement%20OR%20label%3A%22new%20device%22).
- [x] I have checked out the latest <a href="https://github.com/patman15/BMS_BLE-HA/releases">release</a>.

### Please describe the enhancement you would like to have.

Hello,

I have a PowerBoozt Battery. The documentation is sparse and even when I asked one of their technical guys I could not figure out which BMS they are using. In HA this is the adv. Info. Any chance I just could play around to figure out if any of the existing BMS would work? Unfortunately I cannot change the name BT-Battery to something else. Many thanks!

{"name":"BT-Battery          ","address":"04:7F:0E:F5:E2:AD","rssi":-94,"manufacturer_data":{"32516":"0ef5e2ad"},"service_data":{},"service_uuids":["0000fff0-0000-1000-8000-00805f9b34fb"],"source":"08:D1:F9:F6:47:C2","connectable":true,"time":1755288287.584273,"tx_power":null}

### Additional context

This is the battery: https://powerboozt.optimabatteries.nl/wp-content/uploads/2025/05/PB.Li_.12.300-EN.pdf

patman15, 2025-08-16:
Hi! Can you please see if you are able to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery.
It looks like a slightly modified E&J BMS so it could be easy to adapt.

patman15, 2025-08-19:
Ok, no hurry. 😊

patman15, 2025-08-19:
I added an initial version that you could try by installing the branch linked to this issue (see column on the right under *development*) via [manual install](https://github.com/patman15/BMS_BLE-HA#installation). If you could provide me a debug log it would be very helpful for further development.
Thanks, Patrick

hlathome, 2025-08-20:
> Could you get me the the information from the [service explorer script](https://github.com/hbldh/bleak/blob/develop/examples/service_explorer.py) ? I'm having troubles mocking a device for the app to check what it does.

Hi Patman,

here's the outcome from service explorer:

`PS C:\Users\Heiko\Downloads> python service_explorer.py --address 04:7F:0E:F5:E2:AD
2025-08-20 19:20:54,946 __main__ INFO: starting scan...
2025-08-20 19:20:55,239 __main__ INFO: connecting to device...
2025-08-20 19:20:56,461 __main__ INFO: connected to BT-Battery           (04:7F:0E:F5:E2:AD)
2025-08-20 19:20:56,462 __main__ INFO: [Service] 00001801-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Attribute Profile
2025-08-20 19:20:56,490 __main__ INFO:   [Characteristic] 00002a05-0000-1000-8000-00805f9b34fb (Handle: 2): Service Changed (write,indicate,read,notify,broadcast,authenticated-signed-writes), Error: Could not read characteristic handle 2: Protocol Error 0x01: Invalid Handle
2025-08-20 19:20:56,491 __main__ INFO: [Service] 00001800-0000-1000-8000-00805f9b34fb (Handle: 4): Generic Access Profile
2025-08-20 19:20:56,519 __main__ INFO:   [Characteristic] 00002a00-0000-1000-8000-00805f9b34fb (Handle: 5): Device Name (read), Value: bytearray(b'BT-Battery          ')
2025-08-20 19:20:56,549 __main__ INFO:   [Characteristic] 00002a01-0000-1000-8000-00805f9b34fb (Handle: 7): Appearance (read), Value: bytearray(b'\xc0\x03')
2025-08-20 19:20:56,580 __main__ INFO:   [Characteristic] 00002a02-0000-1000-8000-00805f9b34fb (Handle: 9): Peripheral Privacy Flag (read), Value: bytearray(b'\x00')
2025-08-20 19:20:56,610 __main__ INFO:   [Characteristic] 00002a04-0000-1000-8000-00805f9b34fb (Handle: 11): Peripheral Preferred Connection Parameters (read), Value: bytearray(b'\x06\x00\x06\x00\x00\x00\x90\x01')
2025-08-20 19:20:56,640 __main__ INFO:   [Characteristic] 00002aa6-0000-1000-8000-00805f9b34fb (Handle: 13): Central Address Resolution (read), Value: bytearray(b'\x00')
2025-08-20 19:20:56,640 __main__ INFO: [Service] 0000fff0-0000-1000-8000-00805f9b34fb (Handle: 15): Vendor specific
2025-08-20 19:20:56,640 __main__ INFO:   [Characteristic] 0000fff1-0000-1000-8000-00805f9b34fb (Handle: 16): Vendor specific (write,notify)
2025-08-20 19:20:56,670 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 18): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2025-08-20 19:20:56,670 __main__ INFO:   [Characteristic] 0000fff2-0000-1000-8000-00805f9b34fb (Handle: 19): Vendor specific (write-without-response,write), Max write w/o rsp size: 514
2025-08-20 19:20:56,671 __main__ INFO:   [Characteristic] 0000fff3-0000-1000-8000-00805f9b34fb (Handle: 21): Vendor specific (write-without-response,write,notify), Max write w/o rsp size: 514
2025-08-20 19:20:56,700 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 23): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2025-08-20 19:20:56,700 __main__ INFO:   [Characteristic] 0000fff4-0000-1000-8000-00805f9b34fb (Handle: 24): Vendor specific (write,notify)
2025-08-20 19:20:56,730 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 26): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2025-08-20 19:20:56,761 __main__ INFO:   [Characteristic] 0000fff5-0000-1000-8000-00805f9b34fb (Handle: 27): Vendor specific (write,read), Value: bytearray(b'')
2025-08-20 19:20:56,790 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 29): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2025-08-20 19:20:56,791 __main__ INFO:   [Characteristic] 0000fff6-0000-1000-8000-00805f9b34fb (Handle: 30): Vendor specific (write,notify)
2025-08-20 19:20:56,820 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 32): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2025-08-20 19:20:56,820 __main__ INFO: disconnecting...
2025-08-20 19:20:56,944 __main__ INFO: disconnected`

hlathome, 2025-08-20:
> You have also updated `manifest.json`? Can you please ensure that the `RSSI` value is better than -75dBm?

yes I transferred the entire folder from the tree/ into the local HA folder as per your directions. I had not been doing that before hence I checked for paranoia all the files. The manifest.json contains the line items with powerboozt.

On the RSSI topic, I understand the values dont look great but the esp32 device is literally 15cm away from the battery. The Victron SmartSolar is probably 75cm away and the Victron integration on the same esp32 receives good data all times. 

[image]

Many thanks for your engagement - please let me know what I can do to help

hlathome, 2025-08-24:
Hi Patman,

I did it now a couple of times with better RSSI values. No success. Could it be in conjunction with the Bluetooth Proxy? I am also very sure I did the right file operations from the issue-tree into the local HA directory but could to be super sure, can I reconfirm that in runtime within HA? Thx

patman15, 2025-08-24:
Did you restart Home Assistant after modifying the files?

hlathome, 2025-08-25:
yes, couple of times. I am a bit clueless now.

patman15, 2025-08-25:
I'll do a beta release that you can install via HACS to be sure both of us have the same code. I'll ping you today.

patman15, 2025-08-25:
@hlathome please check [release 1.20.0-beta.3](https://github.com/patman15/BMS_BLE-HA/releases/tag/1.20.0-beta.3) which can be installed via HACS. If the device is still not detected, please again provide the advertisement data of the device. (Go to the [advertisement monitor](https://my.home-assistant.io/redirect/bluetooth_advertisement_monitor/), click the device in question, and provide the information via **`copy to clipboard`**).

hlathome, 2025-08-25:
Hi Patman, did the update. Still same, device not found. Here's adv. Did you notice the spaces after BT-Battery? I checked with my first post here, the number of spaces seems to be different. A bit odd, maybe the reason why we cannot connect?

{"name":"BT-Battery          ","address":"04:7F:0E:F5:E2:AD","rssi":-72,"manufacturer_data":{"32516":"0ef5e2ad"},"service_data":{},"service_uuids":["0000fff0-0000-1000-8000-00805f9b34fb"],"source":"08:D1:F9:F6:47:C2","connectable":true,"time":1756128114.2095969,"tx_power":null}

hlathome, 2025-08-25:
Additional comment. github seems to cut off spaces! Manually I count 10 spaces between the "y" of Battery and the trailing quote sign.
`{"name":"BT-Battery          ","address":"04:7F:0E:F5:E2:AD","rssi":-72,"manufacturer_data":{"32516":"0ef5e2ad"},"service_data":{},"service_uuids":["0000fff0-0000-1000-8000-00805f9b34fb"],"source":"08:D1:F9:F6:47:C2","connectable":true,"time":1756128114.2095969,"tx_power":null}`

patman15, 2025-08-25:
Oh man, spaces in a name at the end. 🤦 Cool that you spotted it! Please enable debug logging (wait around 1 minutes) and send me a log file attached to a comment. That should give me the next step of information.

patman15, 2025-08-25:
Some good, some bad news. The bad, the response of your BMS in the log above is total mess. The good, I could write a mock battery that works with the vendor app, so decoding is not an issue. Can you please try to grab a longer log (5-10 minutes) and please attach it as file as it otherwise clutters this issue a lot. Thanks!

hlathome, 2025-08-26:
here's the log, hth, from my humble perspective it repeats w/o any new stuff

[home-assistant_bms_ble_2025-08-26T05-49-43.741Z.log](https://github.com/user-attachments/files/21981459/home-assistant_bms_ble_2025-08-26T05-49-43.741Z.log)

patman15, 2025-08-26:
I updated the development branch with a longer timeout to see if the helps. I also fixed the detection pattern. Can you try again using manual install?

hlathome, 2025-08-31:
Hi Patman, did not work. Next thing I am going to do is setup a fresh ESP32 or maybe I will be using a Rapsberry with a HA instance and play a bit with the distances within the Campervan. That will happen only over the next few weeks; suggest we park the topic until then. Would you have any more ideas how I could get to the packet recording though?

patman15, 2025-09-01:
Hi! If it's a transmission issue, I would not recommend a Raspberry Pi to try to fix it.Iwould give better chances to ESPHome Proxy.
Regarding the recording, maybe find someone with a phone that has Android from the last 6 years or so and ask to install the app, do a recording and download it from phone on a PC via USB?
That's the best I can come up with.

closed 2025-10-19 by github-actions[bot]