patman15/BMS_BLE-HA #469: Seplos

grafiq77, 2025-09-01:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I have thought about [providing a debug log](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) and/or [diagnostic data](https://www.home-assistant.io/docs/configuration/troubleshooting/#download-diagnostics).

### Describe the issue

SP112B2409100524
Is there support for this seplos string because it doesn't work for me or is it for a new one?

patman15, 2025-09-03:
Hi! If you could please post the advertisement information from the advertisement monitor, O can add that pattern soon. Seems like Seplos has a lot of different ones.

patman15, 2025-09-07:
Hi!
Can you please install the branch linked to this issue (see column on the right under *development*) via [manual install](https://github.com/patman15/BMS_BLE-HA#installation) and see if it fixes the issue?
Thanks, Patrick

grafiq77, 2025-09-07:
Everything works fine. Integration is possible until the next release. Thank you very much and best regards.

closed 2025-09-07 by grafiq77