patman15/BMS_BLE-HA #474: JBD UP16S015K v1.3 support v5

inhibitz, 2025-09-04:
### Checklist

- [x] I need support with using the integration.
- [x] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [x] I have thought about [providing a debug log](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) and/or [diagnostic data](https://www.home-assistant.io/docs/configuration/troubleshooting/#download-diagnostics).

### Describe the issue

I have 2 SJY 100Ah batteries with BMS JBD UP16S015K v1.3 with v1 firmware v1.E (JBD0 battery) and v5 (JBD1 battery) . The v1.E connection works, but v5 does not. Could you add support for v5.

[image]

Logs:

[image]

patman15, 2025-09-06:
Ok, just learned from the Internet that this is a new protocol. Can you please see if you are able to capture the Bluetooth traffic on [Android](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) or [iOS](https://novelbits.io/debugging-sniffing-secure-ble-ios/) while using the vendor app with the battery?
That would be the easiest for me to modify the code and also have tests for the future.
Thanks, Patrick

inhibitz, 2025-09-06:
> Ok, just learned from the Internet that this is a new protocol. Can you please see if you are able to capture the Bluetooth traffic on [...]

If you offer recommendations on how to do this on android I'll give it a try. My android is not rooted.

patman15, 2025-09-06:
Please just click the link above, there is a guideline, you do not need root.

patman15, 2025-09-08:
> I don't understand the point of collecting logs from an Android phone in that case.

Because I would need to see how the Android app communicates with the BMS in order to adapt the communication protocol used by this integration. This is only possible by recording the Bluetooth traffic that is between the phone (app) and the BMS.

inhibitz, 2025-09-08:
> Because I would need to see how the Android app communicates with the BMS in order to adapt the communication protocol used by this integration. This is only possible by recording the Bluetooth [...]

OK. I'll try.

inhibitz, 2025-09-08:
> Because I would need to see how the Android app communicates with the BMS in order to adapt the communication protocol used by this integration. This is only possible by recording the Bluetooth [...]

I can't download file . Oh.

closed 2025-09-08 by inhibitz

patman15, 2025-09-08:
You can also try to use `adb start-server` and then run Wireshark and select the Android BT device to do live recording.

inhibitz, 2025-09-08:
> You can also try to use `adb start-server` and then run Wireshark and select the Android BT device to do live recording.

I have also compiled a wiresharp cap log. but I can't attach it to the comment yet.

inhibitz, 2025-09-08:
[bugreport](https://drive.google.com/file/d/1gs0s0l2tRTQH-WTuvdzXVWmNtT03UJYh/view?usp=sharing)

patman15, 2025-09-09:
Hi!
Unfortunately your packet logs did not contain any payload data. 😞 Could be a phone issue. Nevertheless, I found some protocol specification and made a small adaptation. Could you please install the branch linked to this issue (see column on the right under *development*) via [manual install](https://github.com/patman15/BMS_BLE-HA#installation) and see if it fixes the issue? If you could provide me a debug log for final check it would be very helpful, no matter it works or does not.
Please make sure your BMS is set to address `0` (master) using the DIP switches.
Thanks, Patrick

inhibitz, 2025-09-09:
> Hi! Unfortunately your packet logs did not contain any payload data. 😞 Could be a phone issue. Nevertheless, I found some protocol specification and made a small adaptation. Could you please install [...]

Ok, Installed new code to custom_components\bms_ble\plugins\jbd_bms.py

    HEAD_RSP: Final[bytes] = bytes([0xDD, 0x00])  # header for responses
    HEAD_CMD: Final[bytes] = bytes([0xDD, 0x00, 0xA5])  # read header for commands

Unfortunately, it didn't help. My battery is a Master.

patman15, 2025-09-09:
>  can you tell me how to get this full debug?

Sure, have a look at the [troubleshooting guide](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#in-case-you-have-troubles-youd-like-to-have-help-with). It links the related [HA articles ](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics)with instructions.

>  take a close look at log bugreport-zircon_global-AP3A.240905.015.A2-2025-09-08-10-03-23.txt.
>
Not sure where that file is, can you point me to it?

> problem with battery: name: JBD1 and MAC: A4:C1:37:35:C3:BA
>
But in your first picture https://github.com/patman15/BMS_BLE-HA/issues/474#issue-3383919397 the ESPHome proxy says that the device with the MAC :33:24 is not connected. So it's a different one. Are you sure it is always the same BMS not connecting?

inhibitz, 2025-09-09:
> > can you tell me how to get this full debug?
> 
> Sure, have a look at the [troubleshooting guide](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#in-case-you-have-troubles-youd-like-to-have-help-with). It links the related [HA articles ](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics)with instructions.
> 
> > take a close look at log bugreport-zircon_global-AP3A.240905.015.A2-2025-09-08-10-03-23.txt.
> 
> Not sure where that file is, can you point me to it?
> 
> > problem with battery: name: JBD1 and MAC: A4:C1:37:35:C3:BA
> 
> But in your first picture [#474 (comment)](https://github.com/patman15/BMS_BLE-HA/issues/474#issue-3383919397) the ESPHome proxy says that the device with the MAC :33:24 is not connected. So it's a different one. Are you sure it is always the same BMS not connecting?

Unfortunately, the HomeAssistant debug log will not show us anything, all connections go between the ESP BLE proxy and the battery client. even through a proxy, I can connect to the same battery with the MAC address : a4:c1:37:15:33:24 but I can't connect to a4:c1:37:35:c3:ba. the file bugreport-zircon_global-AP3A.240905.015.A2-2025-09-08-10-03-23.txt located in the archive https://drive.google.com/file/d/1gs0s0l2tRTQH-WTuvdzXVWmNtT03UJYh/view?usp=sharing (need download).

patman15, 2025-09-10:
I had a look at `bugreport-zircon_global-AP3A.240905.015.A2-2025-09-08-10-03-23.txt` but it is huge and I have no idea what to see there ...
Could you please provide the information from the [service explorer script](https://github.com/hbldh/bleak/blob/develop/examples/service_explorer.py) ? I'm running out of other ideas ...

> Unfortunately, the HomeAssistant debug log will not show us anything, all connections go between the ESP BLE proxy and the battery client.

Well it tells me that the integration connects and that a request is sent to the BMS, but no reply:

> 2025-09-09 15:22:38.392 DEBUG (MainThread) [bleak_retry_connector] A4:C1:37:35:C3:BA - A4:C1:37:35:C3:BA: Connected after 2 attempts
> 2025-09-09 15:22:38.576 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms] JBD1|C3BA: TX BLE req #1 (WNR): dd 00 a5 3f 03 00 ff 1c 77
> 2025-09-09 15:22:39.291 DEBUG (MainThread) [custom_components.bms_ble.jbd_bms] JBD1|C3BA: TX BLE request timed out.

> even through a proxy, I can connect to the same battery with the MAC address : a4:c1:37:15:33:24 but I can't connect to a4:c1:37:35:c3:ba.

Not sure what you mean by that. How do you define "you can connect"? You mean the integration works or something else?